coolsnowwolf/lede #11421: bcm27xx compile error

smallprogram, 2023-08-02:
**Please fix target/linux/bcm27xx/patches-5.15/950-0600-xhci-quirks-add-link-TRB-quirk-for-VL805.patch**

```
Applying /home/zhusir/lede_Pi4_Model_B.config/target/linux/bcm27xx/patches-5.15/950-0600-xhci-quirks-add-link-TRB-quirk-for-VL805.patch using plaintext:
patching file drivers/usb/host/xhci-pci.c
patching file drivers/usb/host/xhci-ring.c
patching file drivers/usb/host/xhci.h
Hunk #1 FAILED at 1906.
1 out of 1 hunk FAILED -- saving rejects to file drivers/usb/host/xhci.h.rej
Patch failed!  Please fix /home/zhusir/lede_Pi4_Model_B.config/target/linux/bcm27xx/patches-5.15/950-0600-xhci-quirks-add-link-TRB-quirk-for-VL805.patch!
make[4]: *** [Makefile:43: /home/zhusir/lede_Pi4_Model_B.config/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.15.122/.prepared_37b85907fe0c0efc41927b867c8ffeea] Error 1
```

buy404, 2023-08-04:
同树莓派3b 4b编译 950-0600-xhci-quirks-add-link-TRB-quirk-for-VL805.patch 补丁报错

Cathgao, 2023-08-04:
我把报错的补丁一个个删了是可以编译过的，暂时不知道对USB有什么影响，但其他功能是一切正常

lzyest, 2023-08-11:
我也是类似情况= =

unk #1 FAILED at 1906.
1 out of 1 hunk FAILED -- saving rejects to file drivers/usb/host/xhci.h.rej
Patch failed!  Please fix /home/lzyest/lede/target/linux/bcm27xx/patches-5.15/950-0600-xhci-quirks-add-link-TRB-quirk-for-VL805.patch!
make[4]: *** [Makefile:43: /home/lzyest/lede/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/linux-5.15.122/.prepared_52f445a3b4336cb6a60594bbbb170fc3] Error 1
make[4]: Leaving directory '/home/lzyest/lede/target/linux/bcm27xx'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/lzyest/lede/target/linux'
time: target/linux/compile#8.59#36.85#37.76
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/lzyest/lede'
make[1]: *** [target/Makefile:23: /home/lzyest/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/lzyest/lede'
make: *** [/home/lzyest/lede/include/toplevel.mk:231：world] 错误 2

closed 2023-08-28 by smallprogram