coolsnowwolf/lede #11424: 无法编译luci-app-cshark，日志如下

ChaingTsung, 2023-08-03:
```shell
Configuring luci-proto-ncm.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libmbedtls12 for libustream-mbedtls20201210
 * pkg_hash_fetch_best_installation_candidate: Packages for libustream-mbedtls20201210 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package libustream-mbedtls20201210.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-cshark:
 *      libmbedtls12
 * opkg_install_cmd: Cannot install package luci-app-cshark.
 * resolve_conffiles: Existing conffile /home/yoki/lede/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/etc/config/socat is different from the conffile in the new package. The new conffile will be placed at /home/yoki/lede/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/etc/config/socat-opkg.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/yoki/lede'
make[1]: *** [package/Makefile:111: /home/yoki/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/yoki/lede'
make: *** [/home/yoki/lede/include/toplevel.mk:231: world] Error 2
```

Jidonk, 2023-09-26:
同问

zxzx1988, 2024-02-28:
luci-app-cshark 依赖libmbedtls12， 所以这里报错很清楚：libmbedtls12没有安装。所以第一步，在make menuconfig 的时候，在Libraries  --->  SSL 里面，选中 libmbedtls。注意前面必须是星号*，不能是M，更不能是空的。
这一步做完luci-app-cshark 找不到libmbedtls12问题就可以解决了。

但是继续往下编的话，有可能还会遇到libustream-mbedtls20201210冲突的问题，如果遇到这个问题，可以在Libraries界面下面，检查一下 libustream-mbedtls和libustream-openssl：
如果libustream-openssl已经被选中， 那么就要确保libustream-mbedtls不要被选中（注意是不要被选中，前面是星号或者M都不可以，因为选中它的话会和其它库冲突，后面还会报错）

然后就可以正常编译、打包了。

coolsnowwolf, 2026-09-05:
该错误来自 2023 年旧依赖链：`libustream-mbedtls20201210` 依赖已废弃的 `libmbedtls12`，并且日志同时包含架构候选不匹配。当前 LuCI 已按上游在 2025 年移除 luci-app-cshark（Lua 应用已弃用），libustream-mbedtls 改为依赖现行 `libmbedtls`，因此不存在对当前树可实施的 cshark 编译修复。请改用受支持的抓包工具/应用；该旧 issue 关闭。

closed 2026-09-05 by coolsnowwolf