coolsnowwolf/lede #11426: 编译报错

Metear, 2023-08-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
按照README编译命令执行到第4步的make V=s -j1报错
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
编译环境ubuntu2020 lts, 4核心， 24G RAM, 500G硬盘
 ERROR: toolchain/gcc/initial failed to build
A：

### (2) 路由器型号和固件版本
master分支默认
A：

### (3) 详细日志
报错部分日志如下：
  cc1-checksum.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a   -L/home/zf/linux/Source/lede/staging_dir/host/lib -L/home/zf/linux/Source/lede/staging_dir/host/lib -L/home/zf/linux/Source/lede/staging_dir/host/lib -lmpc -lmpfr -lgmp -rdynamic -ldl  -L./../zlib -lz -lzstd
/usr/bin/ld: /home/zf/linux/Source/lede/staging_dir/host/lib/libzstd.a(.._.._.._lib_common_pool.c.o): in function `POOL_free.part.0':
pool.c:(.text+0x16f): undefined reference to `pthread_join'
/usr/bin/ld: /home/zf/linux/Source/lede/staging_dir/host/lib/libzstd.a(.._.._.._lib_common_pool.c.o): in function `POOL_create_advanced':
pool.c:(.text+0x333): undefined reference to `pthread_create'
/usr/bin/ld: /home/zf/linux/Source/lede/staging_dir/host/lib/libzstd.a(.._.._.._lib_common_pool.c.o): in function `POOL_resize':
pool.c:(.text+0x599): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
make[5]: *** [/home/zf/linux/Source/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/c/Make-lang.in:87: cc1] Error 1
make[5]: Leaving directory '/home/zf/linux/Source/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/gcc'
make[4]: *** [Makefile:4428: all-gcc] Error 2
make[4]: Leaving directory '/home/zf/linux/Source/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial'
make[3]: *** [Makefile:37: /home/zf/linux/Source/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/.built] Error 2
make[3]: Leaving directory '/home/zf/linux/Source/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#4.05#0.63#4.67
    ERROR: toolchain/gcc/initial failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Error 1
make[2]: Leaving directory '/home/zf/linux/Source/lede'
make[1]: *** [toolchain/Makefile:93: /home/zf/linux/Source/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/zf/linux/Source/lede'
make: *** [/home/zf/linux/Source/lede/include/toplevel.mk:231: world] Error 2

A：

afcafcafc, 2023-08-09:
the same error

WYC-2020, 2023-08-10:
自己加上pthread  依赖

xpc0938, 2023-09-25:
libzstd.a可能是以前编译的临时文件，把/home/zf/linux/Source/lede/staging_dir/host/lib/libzstd.a删除后再编译，应该就可以了。

wanglouxiaozi, 2024-10-18:
修改lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/gcc/Makefile里的“ZSTD_LIB =  -lzstd”，改为“ZSTD_LIB =  -lzstd -lpthread”就可以了。

coolsnowwolf, 2026-09-05:
该日志来自旧版本 GCC/toolchain 的 zstd 线程库链接问题。当前 LEDE 已在 toolchain/gcc/common.mk 中显式使用 `-pthread`（`ac_cv_search_ZSTD_compress=... -pthread`），并且当前 GCC/toolchain 代码已持续更新，原始 2022/2023 构建环境无法在现行 master 复现。请先使用当前 master、全新 staging_dir 和完整日志重试；若仍失败请提交新的可复现日志。现有 issue 按 not planned 关闭。

closed 2026-09-05 by coolsnowwolf