coolsnowwolf/lede #11428: r8125频繁掉线,已关闭aspm

Canary233, 2023-08-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* 有,但并没解决问题

## 2. 详细叙述

### (1) 具体问题

A：ESXi直通8125给openwrt,流量一大就Link Down

### (2) 路由器型号和固件版本

A：x86,最新源码

### (3) 详细日志

A：
```
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.295325] ------------[ cut here ]------------
Fri Aug  4 13:27:07 2023 kern.info kernel: [  218.295810] NETDEV WATCHDOG: eth1 (r8125): transmit queue 0 timed out
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.296305] WARNING: CPU: 0 PID: 0 at dev_watchdog+0x1bf/0x1d0
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.296780] Modules linked in: shortcut_fe_cm pppoe ppp_async i915 amdgpu video pppox ppp_mppe ppp_generic lzo ipt_REJECT bnx2x ax88179_178a aqc111 xt_time xt_tcpudp xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_iprange xt_helper xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_CT wmi via_velocity usbnet tulip ts_fsm ts_bm tcp_bbr slhc rtl8150 r8168 r8152 r8125 pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount mlx5_core mlx4_en mlx4_core macvlan lzo_rle lzo_decompress lzo_compress libcrc32c
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.296830]  iptable_raw iptable_nat iptable_mangle iptable_filter ip_tables igc iavf i40e gpu_sched forcedeth e1000e drm_display_helper drm_buddy crc_ccitt bnx2 asn1_decoder alx 8139too 8139cp evdev i2c_dev cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 bonding ixgbevf ixgbe igbvf e1000 amd_xgbe mdio_devres sit mdio tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag tun nls_utf8 ena shortcut_fe_ipv6 shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_ssse3 sha1_generic authencesn authenc arc4 crypto_acompress
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.301504]  nls_iso8859_1 nls_cp437 sdhci_pltfm igb vfat fat button_hotplug tg3 mii libphy
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.304284] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 6.1.35 #0
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.304554] Hardware name: VMware, Inc. VMware20,1/440BX Desktop Reference Platform, BIOS VMW201.00V.20829224.B64.2211211842 11/21/2022
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.305103] RIP: 0010:dev_watchdog+0x1bf/0x1d0
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.305377] Code: 30 e9 51 ff ff ff 4c 89 ef c6 05 49 8e b5 00 01 e8 96 da fa ff 44 89 f1 4c 89 ee 48 c7 c7 58 b2 07 82 48 89 c2 e8 f1 3e 90 ff <0f> 0b e9 72 ff ff ff 66 2e 0f 1f 84 00 00 00 00 00 48 89 37 c7 47
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.306027] RSP: 0018:ffffc90000003e98 EFLAGS: 00010292
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.306301] RAX: 0000000000000039 RBX: ffff8880084e0478 RCX: 0000000000000027
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.306601] RDX: ffff88803ec1d508 RSI: 0000000000000001 RDI: ffff88803ec1d500
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.306902] RBP: ffffc90000003eb8 R08: 0000000000000000 R09: 00000000ffffefff
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.307203] R10: ffffc90000003d10 R11: ffffffff822b2e68 R12: ffff8880084e03dc
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.307507] R13: ffff8880084e0000 R14: 0000000000000000 R15: ffffffff817b87e0
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.307813] FS:  0000000000000000(0000) GS:ffff88803ec00000(0000) knlGS:0000000000000000
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.308132] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.308419] CR2: 00007f4d3fb6f000 CR3: 0000000009dd4000 CR4: 0000000000350ef0
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.308741] Call Trace:
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.308990]  <IRQ>
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.309229]  ? show_regs.part.0+0x1e/0x20
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.309500]  ? show_regs.cold+0x8/0xd
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.309761]  ? __warn+0x6e/0xc0
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.310014]  ? dev_watchdog+0x1bf/0x1d0
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.310272]  ? report_bug+0xed/0x140
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.310527]  ? handle_bug+0x46/0x80
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.310784]  ? exc_invalid_op+0x19/0x70
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.311038]  ? asm_exc_invalid_op+0x1b/0x20
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.311300]  ? pfifo_fast_init+0x120/0x120
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.311561]  ? dev_watchdog+0x1bf/0x1d0
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.311821]  ? dev_watchdog+0x1bf/0x1d0
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.312075]  ? pfifo_fast_init+0x120/0x120
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.312331]  call_timer_fn.constprop.0+0x16/0x80
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.312594]  run_timer_softirq+0x271/0x2c0
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.312849]  ? lapic_next_event+0x1b/0x20
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.313095]  ? clockevents_program_event+0x8f/0xe0
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.313347]  __do_softirq+0xc0/0x1e8
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.313583]  __irq_exit_rcu+0x59/0x80
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.313811]  irq_exit_rcu+0x9/0x10
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.314034]  sysvec_apic_timer_interrupt+0x74/0x80
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.314271]  </IRQ>
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.314466]  <TASK>
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.314655]  asm_sysvec_apic_timer_interrupt+0x1b/0x20
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.314882] RIP: 0010:native_safe_halt+0x13/0x20
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.315101] Code: 07 0f 00 2d ff 98 48 00 f4 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 8b 05 da 19 ae 00 85 c0 7e 07 0f 00 2d df 98 48 00 fb f4 <c3> cc cc cc cc cc cc cc cc cc cc cc cc 55 48 89 e5 53 e8 26 80 ff
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.315676] RSP: 0018:ffffffff82203da8 EFLAGS: 00000246
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.315908] RAX: 0000000000000000 RBX: ffff888001eba400 RCX: 4000000000000000
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.316153] RDX: ffff88803ec00000 RSI: ffff888001a1b800 RDI: 0000000000000001
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.316394] RBP: ffffffff82203db8 R08: ffff888001eba400 R09: 0000000000000254
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.316632] R10: 0000000000000000 R11: 000000000000024f R12: ffff888001a1b864
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.316869] R13: ffff888001a1b864 R14: ffffffff822d6bc0 R15: ffffffff822d6ba8
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.317104]  ? acpi_idle_do_entry+0x68/0x90
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.317297]  acpi_idle_enter+0x8a/0xd0
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.317484]  cpuidle_enter_state+0x81/0x2c0
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.317676]  cpuidle_enter+0x2f/0x40
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.317851]  call_cpuidle+0x1e/0x40
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.318023]  do_idle+0x155/0x170
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.318188]  cpu_startup_entry+0x1b/0x20
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.318362]  rest_init+0xc9/0xd0
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.318525]  arch_call_rest_init+0x9/0xa
Fri Aug  4 13:27:07 2023 kern.warn kernel: [  218.318698]  start_kernel+0x672/0x684
[... cut ...]
Fri Aug  4 13:27:07 2023 daemon.notice netifd: Network device 'eth1' link is down
Fri Aug  4 13:27:07 2023 kern.info kernel: [  218.391311] br-lan: port 2(eth1) entered disabled state
Fri Aug  4 13:27:49 2023 authpriv.info dropbear[10932]: Child connection from 10.0.0.171:8156
Fri Aug  4 13:27:51 2023 authpriv.notice dropbear[10932]: Password auth succeeded for 'root' from 10.0.0.171:8156
Fri Aug  4 13:28:14 2023 daemon.warn odhcpd[2136]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
```

hcym, 2023-08-04:
这个不应该了，去年11月以后的驱动没问题了
我的pn41用的很好，bios还是出场的旧版本，内核版本最新的6.1 你说的那个参数我从来没看过，倒是Intel的v217和v219搞的我死去活来

ostrichii, 2023-08-23:
流量大的时候，看看是不是软中断把一个核心打满了，如果是，手动把流量大的网卡rx tx绑到不同核心上