coolsnowwolf/lede #11437: 最新源码编译报错

pk49800, 2023-08-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
编译报错
### (1) 具体问题

A：
 ERROR: package/kernel/mac80211 failed to build (build variant: regular).
### (2) 路由器型号和固件版本

主机型号 | Netgear Nighthawk X4S R7800

固件版本 | OpenWrt R23.7.7 / LuCI Master (git-23.198.59743-991daf5)

### (3) 详细日志

```bash
make[3]: Leaving directory '/home/pk49800/lede/feeds/packages/utils/dockerd'
time: package/feeds/packages/dockerd/compile#0.21#0.16#1.17
make[3]: Entering directory '/home/pk49800/lede/feeds/luci/applications/luci-app-docker'
In file included from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/net/wireless/util.c:12:
/home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/backport-include/linux/etherdevice.h:50:20: error: redefinition of 'eth_hw_addr_set'
   50 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/backport-include/linux/etherdevice.h:3,
                 from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/net/wireless/util.c:12:
./include/linux/etherdevice.h:298:20: note: previous definition of 'eth_hw_addr_set' with type 'void(struct net_device *, const u8 *)' {aka 'void(struct net_device *, const unsigned char *)'}
  298 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/include/linux/ieee80211.h:20,
                 from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/include/net/cfg80211.h:27,
                 from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/net/wireless/radiotap.c:19:
/home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/backport-include/linux/etherdevice.h:50:20: error: redefinition of 'eth_hw_addr_set'
   50 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/backport-include/linux/etherdevice.h:3,
                 from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/include/linux/ieee80211.h:20,
                 from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/include/net/cfg80211.h:27,
                 from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/net/wireless/radiotap.c:19:
./include/linux/etherdevice.h:298:20: note: previous definition of 'eth_hw_addr_set' with type 'void(struct net_device *, const u8 *)' {aka 'void(struct net_device *, const unsigned char *)'}
  298 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/include/linux/ieee80211.h:20,
                 from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/include/net/cfg80211.h:27,
                 from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/net/wireless/sysfs.c:16:
/home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/backport-include/linux/etherdevice.h:50:20: error: redefinition of 'eth_hw_addr_set'
   50 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/backport-include/linux/etherdevice.h:3,
                 from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/include/linux/ieee80211.h:20,
                 from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/include/net/cfg80211.h:27,
                 from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/net/wireless/sysfs.c:16:
./include/linux/etherdevice.h:298:20: note: previous definition of 'eth_hw_addr_set' with type 'void(struct net_device *, const u8 *)' {aka 'void(struct net_device *, const unsigned char *)'}
  298 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/include/linux/ieee80211.h:20,
                 from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/net/mac80211/wpa.c:13:
/home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/backport-include/linux/etherdevice.h:50:20: error: redefinition of 'eth_hw_addr_set'
   50 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/backport-include/linux/etherdevice.h:3,
                 from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/include/linux/ieee80211.h:20,
                 from /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/net/mac80211/wpa.c:13:
./include/linux/etherdevice.h:298:20: note: previous definition of 'eth_hw_addr_set' with type 'void(struct net_device *, const u8 *)' {aka 'void(struct net_device *, const unsigned char *)'}
  298 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
[... 52 lines cut ...]
mkdir -p /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp
SHELL= flock /home/pk49800/lede/tmp/.root-copy.flock -c 'cp -fpR /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-app-docker/.pkgdir/luci-i18n-docker-zh-cn/. /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/'
touch /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.luci-i18n-docker-zh-cn_installed
echo "luci-i18n-docker-zh-cn" >> /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/luci-app-docker.default.install
make[3]: Leaving directory '/home/pk49800/lede/feeds/luci/applications/luci-app-docker'
time: package/feeds/luci/luci-app-docker/compile#0.11#0.00#0.12
make[9]: *** [scripts/Makefile.build:262: /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/net/wireless/reg.o] Error 1
make[9]: *** [scripts/Makefile.build:262: /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/net/mac80211/offchannel.o] Error 1
make[9]: *** [scripts/Makefile.build:262: /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/net/mac80211/scan.o] Error 1
make[8]: *** [scripts/Makefile.build:497: /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/net/mac80211] Error 2
make[9]: *** [scripts/Makefile.build:262: /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/net/wireless/scan.o] Error 1
[... cut ...]
make[4]: Leaving directory '/home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24'
make[3]: *** [Makefile:401: /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1.24/.built] Error 2
make[3]: Leaving directory '/home/pk49800/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#8.84#0.86#1.62
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
```

chenhaowen01, 2023-08-09:
我在编译CR660X固件时遇到了一样的问题，源码版本是3441bc5588a1a208f79a7bbc6fbeb90e987433a4。
出错的log：
```shell
make[4]: Entering directory '/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24'
make[5]: 'Kconfig.versions' is up to date.
make[7]: 'Kconfig.versions' is up to date.
make[8]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (                \
echo >&2;                                                       \
echo >&2 "  ERROR: Kernel configuration is invalid.";           \
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";      \
echo >&2 ;                                                      \
/bin/false)
/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/scripts/make -f ./scripts/Makefile.build obj=/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24 \
single-build= \
need-builtin=1 need-modorder=1
/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/scripts/make -f ./scripts/Makefile.build obj=/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/compat \
 \
need-builtin=1 \
need-modorder=1
{   echo /home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/compat/compat.ko; :; } \
| awk '!x[$0]++' - > /home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/compat/modules.order
/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/scripts/make -f ./scripts/Makefile.build obj=/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/drivers/net/wireless \
 \
need-builtin=1 \
need-modorder=1
/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/scripts/make -f ./scripts/Makefile.build obj=/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/drivers/net/wireless/admtek \
 \
need-builtin=1 \
need-modorder=1
{  :; } \
| awk '!x[$0]++' - > /home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/drivers/net/wireless/admtek/modules.order
/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/scripts/make -f ./scripts/Makefile.build obj=/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/drivers/net/wireless/ath \
 \
need-builtin=1 \
need-modorder=1
{  :; } \
| awk '!x[$0]++' - > /home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/drivers/net/wireless/ath/modules.order
/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/scripts/make -f ./scripts/Makefile.build obj=/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/drivers/net/wireless/atmel \
 \
need-builtin=1 \
need-modorder=1
{  :; } \
| awk '!x[$0]++' - > /home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/drivers/net/wireless/atmel/modules.order
/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/scripts/make -f ./scripts/Makefile.build obj=/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/drivers/net/wireless/broadcom \
 \
need-builtin=1 \
need-modorder=1
{  :; } \
| awk '!x[$0]++' - > /home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/drivers/net/wireless/broadcom/modules.order
/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/scripts/make -f ./scripts/Makefile.build obj=/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/drivers/net/wireless/intel \
 \
need-builtin=1 \
need-modorder=1
{  :; } \
| awk '!x[$0]++' - > /home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/drivers/net/wireless/intel/modules.order
/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/scripts/make -f ./scripts/Makefile.build obj=/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/drivers/net/wireless/intersil \
 \
need-builtin=1 \
need-modorder=1
{  :; } \
| awk '!x[$0]++' - > /home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/drivers/net/wireless/intersil/modules.order
/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/scripts/make -f ./scripts/Makefile.build obj=/home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/drivers/net/wireless/marvell \
 \
[... 64 lines cut ...]
                    ^~~~~~~~~~~~~~~
In file included from /home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/backport-include/linux/etherdevice.h:3,
                 from /home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/include/linux/ieee80211.h:20,
                 from /home/chw/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1.24/include/net/mac80211.h:20,
[... cut ...]
make[3]: Leaving directory '/home/chw/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#1.72#0.41#2.00
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/chw/lede'
make[1]: *** [package/Makefile:110: /home/chw/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/chw/lede'
make: *** [/home/chw/lede/include/toplevel.mk:231: world] Error 2
```

WYC-2020, 2023-08-10:
就是升级了linux 内核 里面别人定义了这个函数了，所以 有两种修改方法 一种去掉linux 内部的定义 一种就是修复这个库 吧这个定义去掉

coolsnowwolf, 2023-08-10:
fixed #38bca36d20b573daf6db3a848dab5100a608dd2d

closed 2023-08-10 by pk49800

WYC-2020, 2023-08-11:
> fixed #38bca36d20b573daf6db3a848dab5100a608dd2d

这个修复方法 5.10 编译会出错 需要升级下5.10内核，因为5.10目前还没有这个定义，升级到189就行了，我试过了

ywt114, 2023-08-13:
r8152基于5.4内核是修好了，不过基于5.10的挂了