coolsnowwolf/lede #11438: lede使用i350 sriov vf网卡做旁路由 ipv6无法访问外网

jxdqwer, 2023-08-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用ikuai做主路由拨号，ip:192.168.11.2  lede使用i350 sriov vf网卡做旁路由lan网卡 ip: 192.168.11.2
lede新建dhcpv6客户端可以获取24开头的ipv6地址，但是无法ping通其他ipv6地址。
root@OpenWrt:~# ping6 ipw.cn
PING ipw.cn (2408:874c:1ff:52:2d::): 56 data bytes

卡在这里无法ping通，内网里的其他使用vf网卡的设备也无法ping通lede获取的ipv6地址，但是可以ping通外网，也就是说可以正常使用ipv6。

### (2) 路由器型号和固件版本

A：QEMU Standard PC (Q35 + ICH9 - Intel(R) Xeon(R) CPU D-1581 @ 1.80GHz : 4C4T OpenWrt R23.7.7 / LuCI Master (git-23.198.59743-991daf5)

### (3) 详细日志
[    0.000000] Linux version 6.1.44 (dean@d1581) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6131-3441bc558) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT_DYNAMIC Wed Aug  9 08:25:17 2023
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=47b82509-3f9e-fe36-6362-176beabfbf02 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007ffddfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffde000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b0000000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (Q35 + ICH9, 2009), BIOS rel-1.16.2-0-gea1b7a073390-prebuilt.qemu.org 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000001] kvm-clock: using sched offset of 5434892724 cycles
[    0.000003] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000006] tsc: Detected 1799.998 MHz processor
[    0.000685] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000689] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000693] last_pfn = 0x7ffde max_arch_pfn = 0x400000000
[    0.000735] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.010146] found SMP MP-table at [mem 0x000f5b50-0x000f5b5f]
[    0.010166] Using GB pages for direct mapping
[    0.010360] ACPI: Early table checksum verification disabled
[    0.010363] ACPI: RSDP 0x00000000000F5980 000014 (v00 BOCHS )
[    0.010373] ACPI: RSDT 0x000000007FFE53D2 00003C (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.010380] ACPI: FACP 0x000000007FFE50E8 0000F4 (v03 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.010388] ACPI: DSDT 0x000000007FFDF040 0060A8 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.010393] ACPI: FACS 0x000000007FFDF000 000040
[    0.010396] ACPI: APIC 0x000000007FFE51DC 000090 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.010400] ACPI: SSDT 0x000000007FFE526C 0000CA (v01 BOCHS  VMGENID  00000001 BXPC 00000001)
[    0.010404] ACPI: HPET 0x000000007FFE5336 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.010408] ACPI: MCFG 0x000000007FFE536E 00003C (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.010411] ACPI: WAET 0x000000007FFE53AA 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.010414] ACPI: Reserving FACP table memory at [mem 0x7ffe50e8-0x7ffe51db]
[    0.010416] ACPI: Reserving DSDT table memory at [mem 0x7ffdf040-0x7ffe50e7]
[    0.010417] ACPI: Reserving FACS table memory at [mem 0x7ffdf000-0x7ffdf03f]
[    0.010417] ACPI: Reserving APIC table memory at [mem 0x7ffe51dc-0x7ffe526b]
[    0.010418] ACPI: Reserving SSDT table memory at [mem 0x7ffe526c-0x7ffe5335]
[    0.010419] ACPI: Reserving HPET table memory at [mem 0x7ffe5336-0x7ffe536d]
[    0.010420] ACPI: Reserving MCFG table memory at [mem 0x7ffe536e-0x7ffe53a9]
[    0.010420] ACPI: Reserving WAET table memory at [mem 0x7ffe53aa-0x7ffe53d1]
[    0.010469] Zone ranges:
[    0.010471]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[... 802 lines cut ...]
[   15.172608] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[   15.175684] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[   15.547154] [drm] amdgpu kernel modesetting enabled.
[   15.630302] kmodloader: done loading kernel modules from /etc/modules.d/*
[   19.363947] 8021q: adding VLAN 0 to HW filter on device eth0
[   19.368626] br-lan: port 1(eth0) entered blocking state
[   19.370774] br-lan: port 1(eth0) entered disabled state
[   19.373984] device eth0 entered promiscuous mode
[   19.448778] igbvf 0000:01:00.0: Link is Up 1000 Mbps Full Duplex
[   19.450960] br-lan: port 1(eth0) entered blocking state
[   19.451979] br-lan: port 1(eth0) entered forwarding state
[   19.454041] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   44.324760] squashfs: Unknown parameter 'discard'
[   44.330162] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x89e45589)
[   44.332903] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
[   44.336083] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x2e48937b)
[   44.339356] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
[   44.344848] exFAT-fs (sda128): invalid boot record signature
[   44.347307] exFAT-fs (sda128): failed to read boot sector
[   44.349604] exFAT-fs (sda128): failed to recognize exfat type
[   74.579472] /dev/sda1: Can't open blockdev
[   74.583034] /dev/sda1: Can't open blockdev
[   74.586066] /dev/sda1: Can't open blockdev
[   74.589238] squashfs: Unknown parameter 'discard'
[   74.593508] /dev/sda1: Can't open blockdev

jxdqwer, 2023-08-10:
补充一下，如果lan不使用i350 vf网卡，使用PVE的virio网卡，lede获取到ipv6地址之后，可以ping通其他ipv6地址，正常使用ipv6。

jxdqwer, 2023-08-10:
补充一下，我已经尝试了5.4 5.10 5.15 6.1内核的openwrt lede immwrt，此场景下均无法正常使用ipv6。但是如果使用pve创建的virio网卡，ipv6一切正常。

jxdqwer, 2023-08-11:
补充，
使用immortalwrt[v23.05.0-rc3]编译固件，此场景下ipv6正常工作。
immortalwrt[v21.02.6]编译固件，此场景下ipv6可以获取ipv6地址但不能正常工作。
lede最新源码编译固件，此场景下ipv6可以获取ipv6地址但不能正常工作。

jxdqwer, 2023-08-15:
update，immortalwrt[v23.05.0-rc3]用了一段时间也不行了，qemu删除vf网卡，改成virio网卡，又可以了。。。真怪

Canary233, 2025-03-09:
> update，immortalwrt[v23.05.0-rc3]用了一段时间也不行了，qemu删除vf网卡，改成virio网卡，又可以。。。真怪，vf网卡怎么就不行呢。

解决了吗，博通卡也遇到这个问题了

Mrkelo, 2025-06-13:
同问，楼主解决了么，X710 在用 pve做sriov，Ikuai和op在同一个PF的VF下，也是ipv4通但是IPV6不通，可以获取到24开头的公网地址，但无法和ikuai通讯导致出不了公网

Mrkelo, 2025-06-18:
> > 同问，楼主解决了么，X710 在用 pve做sriov，Ikuai和op在同一个PF的VF下，也是ipv4通但是IPV6不通，可以获取到24开头的公网地址，但无法和ikuai通讯导致出不了公网
> 
> 没解决，你的设备不走op能用v6吗

不走OP是可以用的，包括用其他系统，debian、fedora、ikuai之类的都可以使用VF互相通信，就唯独OP不行。

coolsnowwolf, 2026-09-05:
已核对当前 LEDE 与 OpenWrt 主线：igbvf/iavf 均使用 Linux 原生驱动，没有针对 IPv6 的 LEDE 特殊补丁；原日志还显示两块 I350 VF 已被 igbvf 正常识别并建立 1Gbps 链路。该问题限定在 QEMU/KVM、iKuai 与 LEDE 共用 PF/VF、桥接及 IPv6 邻居发现的组合环境，缺少 tcpdump 邻居发现报文、ip -6 neigh、VF trust/spoof-check、桥接和 RA 配置，无法证明是源码缺陷。后续若仍可复现，请补充上述信息后按具体拓扑重新提交；本 issue 关闭为环境/配置问题。

closed 2026-09-05 by coolsnowwolf