coolsnowwolf/lede #11439: 2023新源码编译的x86-64固件启动不了，ehci-pci: irq 23,io mem 0xdff05000

ellermister, 2023-08-10:
## 1.关于你要提交的问题
2023新源码编译的x86-64固件启动不了，ehci-pci: irq 23,io mem 0xdff05000
曾经2022年编译的可以启动

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
a. 用refus把openwrt-x86-64-generic-squashfs-combined-efi.img刷入U盘
b. 插入到路由，进bios从U盘开机启动
c.日志刷3秒就停止，最后一个内容是：ehci-pci: irq 23,io mem 0xdff05000

### (1) 具体问题

A：卡死无法开机，尝试使用别人的新固件也跑不起来。
2022年自己的编译的固件就可以运行起来，明显区别就是老的固件可能只有180M的img文件，而新的都是540M起的。

### (2) 路由器型号和固件版本

A：n2600, x86-64

### (3) 详细日志

A：
[image]

coolsnowwolf, 2023-08-10:
不再支持 N2600

ellermister, 2023-08-10:
有没有最后一个版本可以用的？是openwrt官方源也不支持n2600了还是仅lede分支不支持了，我想重新编译留一个最后最新的养老。

ellermister, 2023-08-10:
> 不再支持 N2600

有没有最后一个版本可以用的？是openwrt官方源也不支持n2600了还是仅lede分支不支持了，我想重新编译留一个最后最新的养老。@coolsnowwolf

WYC-2020, 2023-08-11:
c83233b0525e73c1e0d4525686388c3eee027267  回到这个版本 应该可以自己去试试
你看下 commit 就知道应该回到那个

WYC-2020, 2023-08-11:
或者自己去 参考80ce76a3650e6eb025a88a2debd904b4de1f7b3b 这个提交 看能否自己 吧最新代码加入支持

ellermister, 2023-08-11:
> 或者自己去 参考80ce76a3650e6eb025a88a2debd904b4de1f7b3b 这个提交 看能否自己 吧最新代码加入支持

好的，感谢，我先放弃了，还是用曾经的稳定版，不懂OP源码不折腾了😁

Her0R0cket, 2023-08-25:
刷个BIOS就能,就是开机的时候会在不同阶段有不同的滴滴叫,
我也不知道具体是哪个了,只能你自己试试了.

[26026MOD.zip](https://github.com/coolsnowwolf/lede/files/12434642/26026MOD.zip)
[SV3_26026_3850bios.zip](https://github.com/coolsnowwolf/lede/files/12434643/SV3_26026_3850bios.zip)
[26026bios.zip](https://github.com/coolsnowwolf/lede/files/12434641/26026bios.zip)

[image]

Her0R0cket, 2023-08-25:
我也是在去年的时候遇到的这个问题,大概有2个月这样没有跟上主线版本,之后编译了就挂了,尝试一个个倒回去,发现好累,后来想到了BIOS还好找到了,不过我现在主路由是ikuai了,op就在nas里开个虚拟机混混了,这样家里网络也少好多问题

Her0R0cket, 2024-05-05:
> @Her0R0cket 你好，我想问下你这三个BIOS，哪个是可以刷最新的openwrt，我目前是默认BIOS，开机有社区金融的字样那个。有刷的顺序要求吗？

我刷的时候就随便刷的,没注意顺序,这么久了,真不记得是哪个了

coolsnowwolf, 2026-09-05:
`ehci-pci: irq 23, io mem ...` 是 EHCI 初始化信息，并非错误或启动停止原因。N2600 属于早期 x86 平台，2023 后镜像体积/EFI 与分区布局变化可能触发 BIOS/USB 启动兼容性；仅凭最后一行无法定位内核故障。当前 x86 主线没有针对 N2600 EHCI 的回归补丁，且该报告无完整串口日志、启动模式及镜像校验信息。建议使用 legacy BIOS/combined 镜像并提供完整 boot log；本旧 issue 现关闭。

closed 2026-09-05 by coolsnowwolf