coolsnowwolf/lede #11444: 拉取最新源码编译5.4内核x86固件报错

bi4tms, 2023-08-11:
拉取最新源码编译5.4内核x86固件报错，具体日志如下：
make[4]: Entering directory '/home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.252'
  CC [M]  /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/src/r8152.o
In file included from /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/src/r8152.c:32:
/home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/src/compatibility.h:611:28: error: redefinition of 'eth_hw_addr_set'
  611 |         static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                            ^~~~~~~~~~~~~~~
In file included from /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/src/r8152.c:17:
./include/linux/etherdevice.h:298:20: note: previous definition of 'eth_hw_addr_set' with type 'void(struct net_device *, const u8 *)' {aka 'void(struct net_device *, const unsigned char *)'}
  298 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
make[5]: *** [scripts/Makefile.build:262: /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/src/r8152.o] Error 1
make[4]: *** [Makefile:1750: /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/src] Error 2
make[4]: Leaving directory '/home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.252'
make[3]: *** [Makefile:43: /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/.built] Error 2
make[3]: Leaving directory '/home/x86.5.4/lede/package/lean/r8152'
time: package/lean/r8152/compile#2.20#0.76#2.97
    ERROR: package/lean/r8152 failed to build.
make[2]: *** [package/Makefile:116: package/lean/r8152/compile] Error 1
make[2]: Leaving directory '/home/x86.5.4/lede'
make[1]: *** [package/Makefile:110: /home/x86.5.4/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/x86.5.4/lede'
make: *** [/home/x86.5.4/lede/include/toplevel.mk:231：world] 错误 2

gsblay, 2023-08-12:
选5.4内核同样报错，选其他的内核编译都没问题。

WYC-2020, 2023-08-12:
这个我修过了，等会给你

ywt114, 2023-08-12:
确实是 暂时通过主目录下执行回滚git reset --hard 3441bc(补全)5588a1a208f79a7bbc6fbeb90e987433a4^暂时使用

WYC-2020, 2023-08-12:
[0001-rework-eth-hw-addr-set-for-kernel-5.4.patch](https://github.com/coolsnowwolf/lede/files/12326604/0001-rework-eth-hw-addr-set-for-kernel-5.4.patch)
放这里
[image]

gsblay, 2023-08-12:
这个patch放上以后，5.4内核编译通过。

ywt114, 2023-08-12:
> 这个我修过了，等会给你

你可以提交pull requests啦

xiangfeidexiaohuo, 2023-08-12:
r8152  哈哈  5.4内核 报错

closed 2023-08-12 by bi4tms

ywt114, 2023-08-13:
r8152基于5.4内核是修好了，不过基于5.10的挂了