coolsnowwolf/lede #11445: 兆能M2 ipq60xx 编译固件失败

zui-lang, 2023-08-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：兆能M2编译固件失败、在config的配置中只选择了芯片和机型，其他默认，保存到`.config`后直接编译，报错

### (2) 路由器型号和固件版本

A：兆能M2 ipq60xx 、内存魔改1G、拉去的是最新的源码，所以内核版本应该是5.4

### (3) 详细日志

A：
```
make[2]: Entering directory '/home/Could0323/lede/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/Could0323/lede/scripts/config'
make[1]: Entering directory '/home/Could0323/lede'
make[2]: Entering directory '/home/Could0323/lede'
+ mkdir -p /home/Could0323/lede/staging_dir/target-aarch64_cortex-a53_musl
+ cd /home/Could0323/lede/staging_dir/target-aarch64_cortex-a53_musl
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /home/Could0323/lede/build_dir/target-aarch64_cortex-a53_musl/stamp
touch /home/Could0323/lede/staging_dir/target-aarch64_cortex-a53_musl/.prepared
+ mkdir -p /home/Could0323/lede/staging_dir/host
+ cd /home/Could0323/lede/staging_dir/host
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /home/Could0323/lede/build_dir/host/stamp /home/Could0323/lede/staging_dir/host/include/sys
install -m0644 /home/Could0323/lede/tools/include/*.h /home/Could0323/lede/staging_dir/host/include/
install -m0644 /home/Could0323/lede/tools/include/sys/*.h /home/Could0323/lede/staging_dir/host/include/sys/
ln -snf lib /home/Could0323/lede/staging_dir/host/lib64
touch /home/Could0323/lede/staging_dir/host/.prepared
make[3]: Entering directory '/home/Could0323/lede/tools/flock'
make[3]: Leaving directory '/home/Could0323/lede/tools/flock'
time: tools/flock/compile#0.04#0.01#0.08
make[3]: Entering directory '/home/Could0323/lede/tools/xz'
make[3]: Leaving directory '/home/Could0323/lede/tools/xz'
time: tools/xz/compile#0.06#0.01#0.10
make[3]: Entering directory '/home/Could0323/lede/tools/sed'
make[3]: Leaving directory '/home/Could0323/lede/tools/sed'
time: tools/sed/compile#0.05#0.02#0.09
make[3]: Entering directory '/home/Could0323/lede/tools/libdeflate'
make[3]: Leaving directory '/home/Could0323/lede/tools/libdeflate'
time: tools/libdeflate/compile#0.04#0.01#0.07
make[3]: Entering directory '/home/Could0323/lede/tools/patch'
make[3]: Leaving directory '/home/Could0323/lede/tools/patch'
time: tools/patch/compile#0.05#0.01#0.07
make[3]: Entering directory '/home/Could0323/lede/tools/tar'
make[3]: Leaving directory '/home/Could0323/lede/tools/tar'
time: tools/tar/compile#0.04#0.03#0.09
make[3]: Entering directory '/home/Could0323/lede/tools/m4'
make[3]: Leaving directory '/home/Could0323/lede/tools/m4'
time: tools/m4/compile#0.05#0.03#0.11
make[3]: Entering directory '/home/Could0323/lede/tools/autoconf'
make[3]: Leaving directory '/home/Could0323/lede/tools/autoconf'
time: tools/autoconf/compile#0.05#0.01#0.08
make[3]: Entering directory '/home/Could0323/lede/tools/autoconf-archive'
make[3]: Leaving directory '/home/Could0323/lede/tools/autoconf-archive'
time: tools/autoconf-archive/compile#0.05#0.01#0.08
make[3]: Entering directory '/home/Could0323/lede/tools/ninja'
make[3]: Leaving directory '/home/Could0323/lede/tools/ninja'
time: tools/ninja/compile#0.04#0.01#0.07
make[3]: Entering directory '/home/Could0323/lede/tools/meson'
[... 49 lines cut ...]
[2/210] g++ -O2 -I/home/Could0323/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1 -DCMake_HAVE_CXX_FILESYSTEM=1   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Bootstrap.cmk   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Source   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Source/LexerParser   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Utilities/std   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Utilities  -c /home/Could0323/lede/build_dir/host/cmake-3.25.2/Source/cmFileTimeCache.cxx -o cmFileTimeCache.o
[3/210] g++ -O2 -I/home/Could0323/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1 -DCMake_HAVE_CXX_FILESYSTEM=1   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Bootstrap.cmk   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Source   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Source/LexerParser   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Utilities/std   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Utilities  -c /home/Could0323/lede/build_dir/host/cmake-3.25.2/Source/cmFileTimes.cxx -o cmFileTimes.o
[4/210] g++ -O2 -I/home/Could0323/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1 -DCMake_HAVE_CXX_FILESYSTEM=1   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Bootstrap.cmk   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Source   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Source/LexerParser   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Utilities/std   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Utilities  -c /home/Could0323/lede/build_dir/host/cmake-3.25.2/Source/cmFileCommand.cxx -o cmFileCommand.o
FAILED: cmFileCommand.o
g++ -O2 -I/home/Could0323/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1 -DCMake_HAVE_CXX_FILESYSTEM=1   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Bootstrap.cmk   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Source   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Source/LexerParser   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Utilities/std   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Utilities  -c /home/Could0323/lede/build_dir/host/cmake-3.25.2/Source/cmFileCommand.cxx -o cmFileCommand.o
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
[5/210] g++ -O2 -I/home/Could0323/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1 -DCMake_HAVE_CXX_FILESYSTEM=1   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Bootstrap.cmk   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Source   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Source/LexerParser   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Utilities/std   -I/home/Could0323/lede/build_dir/host/cmake-3.25.2/Utilities  -c /home/Could0323/lede/build_dir/host/cmake-3.25.2/Source/cmFindBase.cxx -o cmFindBase.o
ninja: build stopped: subcommand failed.
---------------------------------------------
Error when bootstrapping CMake:
Problem while running /home/Could0323/lede/staging_dir/host/bin/ninja
---------------------------------------------
Log of errors: /home/Could0323/lede/build_dir/host/cmake-3.25.2/Bootstrap.cmk/cmake_bootstrap.log
---------------------------------------------
make[3]: *** [Makefile:60: /home/Could0323/lede/build_dir/host/cmake-3.25.2/.configured] Error 9
make[3]: Leaving directory '/home/Could0323/lede/tools/cmake'
time: tools/cmake/compile#14.18#70.77#7857.92
    ERROR: tools/cmake failed to build.
make[2]: *** [tools/Makefile:219: tools/cmake/compile] Error 1
make[2]: Leaving directory '/home/Could0323/lede'
make[1]: *** [tools/Makefile:215: /home/Could0323/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynnynyynnynnyyyyyyyyyyyyyyyyynynnyyyyyyy] Error 2
make[1]: Leaving directory '/home/Could0323/lede'
make: *** [/home/Could0323/lede/include/toplevel.mk:231: world] Error 2
```

yuhongwei380, 2023-08-15:
兆能M2 请用https://github.com/coolsnowwolf/openwrt-gl-ax1800 这个项目去编译

zui-lang, 2023-08-15:
> 兆能M2 请用https://github.com/coolsnowwolf/openwrt-gl-ax1800 这个项目去编译

这个项目我是编译成功了，我看Linux内核是4.6，主要是想问一下ipq60xx的内核是不能升到5.x吗

AnClark, 2023-10-07:
看样子是在构建工具链期间出错，CMake这款工具没有构建成功。我这里也出过错误，后来解决了。

先拉取本仓库的最新提交（`git pull`），然后运行以下命令，清掉缓存：

```
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a
make distclean
```

最后重新编译试试。

---

另：我用这个仓库编译M2的固件，编译成功，但是无法启动。

closed 2023-12-22 by Akimio521