coolsnowwolf/lede #11451: 补丁错误

bigbugcc, 2023-08-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
Patch failed!  Please fix /workdir/openwrt/target/linux/bcm27xx/patches-5.15/950-0600-xhci-quirks-add-link-TRB-quirk-for-VL805.patch!
### (1) 具体问题
A：编译时报错，疑似补丁引起的

### (2) 路由器型号和固件版本

A：Raspberry Pi4/3

### (3) 详细日志
Applying /workdir/openwrt/target/linux/bcm27xx/patches-5.15/950-0600-xhci-quirks-add-link-TRB-quirk-for-VL805.patch using plaintext: 
patching file drivers/usb/host/xhci-pci.c
patching file drivers/usb/host/xhci-ring.c
patching file drivers/usb/host/xhci.h
Hunk #1 FAILED at 1906.
1 out of 1 hunk FAILED -- saving rejects to file drivers/usb/host/xhci.h.rej
Patch failed!  Please fix /workdir/openwrt/target/linux/bcm27xx/patches-5.15/950-0600-xhci-quirks-add-link-TRB-quirk-for-VL805.patch!
make[4]: *** [Makefile:43: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/linux-5.15.122/.prepared_9d5f38fd1f2353ef1c495db30d268a26] Error 1
make[4]: Leaving directory '/workdir/openwrt/target/linux/bcm27xx'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#14.46#6.76#18.11
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

ecrasy, 2023-08-20:
删掉这些出错的补丁文件可以修复编译错误
但是否会影响到程序运行则未知：
```
PATCH_PATH="target/linux/bcm27xx/patches-5.15"
rm -rf ${PATCH_PATH}/950-0600-xhci-quirks-add-link-TRB-quirk-for-VL805.patch
rm -rf ${PATCH_PATH}/950-0606-usb-xhci-add-VLI_TRB_CACHE_BUG-quirk.patch
rm -rf ${PATCH_PATH}/950-0717-usb-xhci-add-a-quirk-for-Superspeed-bulk-OUT-transfe.patch
rm -rf ${PATCH_PATH}/950-0747-usb-xhci-rework-XHCI_VLI_SS_BULK_OUT_BUG-quirk.patch
```

bigbugcc, 2023-08-20:
暂时回滚到补丁修改前了，稳定为主。

ecrasy, 2023-08-22:
官方的树莓派源码在5.15.120之后就切到6.1内核
我对比了LEDE和官方源码的target/linux/bcm27xx/patches-5.15
他们在5.15.120的commit都是一致的
之后分道扬镳
LEDE更新至5.15.122，官方切换至6.1
此时target/linux/bcm27xx/patches-5.15里面有些文件就出现了不适配的问题

closed 2023-08-25 by aiamadeus

bigbugcc, 2023-08-26:
了解，蟹蟹😉