coolsnowwolf/lede #11454: banana pi r3 dts is error

ChaingTsung, 2023-08-15:
banana pi r3 dts is error,for example,pwm-fan

```[    0.049374] pwm-mediatek 10048000.pwm: clock: top fail: -517```

---
###  offical 
some suggestion can be seen at https://forum.banana-pi.org/t/banana-pi-r3-fan-with-pwm/14406/84
 ### OFFICIAL  updates history
https://github.com/frank-w/BPI-Router-Linux/commits/6.5-rc/arch/arm64/boot/dts/mediatek

aiamadeus, 2023-08-16:
是 pwm 风扇无法使用？

ChaingTsung, 2023-08-16:
> pwm 风扇无法使用吗？

Yes

ChaingTsung, 2023-08-16:
It seems this bug had been fixed up on kernel 6.1 by official.(I didn't test.) And My kernel is 5.15.122.

ChaingTsung, 2023-08-16:
> pwm 风扇无法使用吗？

PWM fan run at the first 5s and then IT STOP (even env. is too high) working.

ChaingTsung, 2023-08-16:
> pwm 风扇无法使用吗？

Now it works! Use `&infracfg CLK_INFRA_PWM_HCK` to replace  `&infracfg CLK_INFRA_PWM_STA` in line 249.(File: target/linux/mediatek/files-5.15/arch/arm64/boot/dts/mediatek/mt7986a.dtsi) That's fine.

aiamadeus, 2023-08-17:
https://github.com/coolsnowwolf/lede/blob/c5efeaff691ebccbf9419a92896404ab67c97bf4/target/linux/mediatek/files-5.15/arch/arm64/boot/dts/mediatek/mt7986a.dtsi#L249-L250

I guess you deleted the top clock?

ChaingTsung, 2023-08-17:
> 

NO.Here is the new one.
```config
 243			pwm: pwm@10048000 {
   244				compatible = "mediatek,mt7986-pwm";
   245				reg = <0 0x10048000 0 0x1000>;
   246				#clock-cells = <1>;
   247				#pwm-cells = <2>;
   248				interrupts = <GIC_SPI 137 IRQ_TYPE_LEVEL_HIGH>;
   249				clocks = <&topckgen CLK_TOP_PWM_SEL>,
   250					 <&infracfg CLK_INFRA_PWM_HCK>,				
   251					 <&infracfg CLK_INFRA_PWM1_CK>,
   252					 <&infracfg CLK_INFRA_PWM2_CK>;
   253				clock-names = "top", "main", "pwm1", "pwm2";
   254				status = "disabled";
   255			};

```

closed 2023-08-30 by ChaingTsung