coolsnowwolf/lede #11456: DG1显卡驱动打不上

achinese888, 2023-08-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：DG1显卡无驱动 i915不支持 怎么搞
      日志最重要的几行：[   15.365318] i915 0000:03:00.0: Your graphics device 4908 is not properly supported by i915 in this
[   15.365318] kernel version. To force driver probe anyway, use i915.force_probe=4908
[   15.365318] module parameter or CONFIG_DRM_I915_FORCE_PROBE=4908 configuration option,
[   15.365318] or (recommended) check for kernel updates.

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：[    0.000000] Linux version 6.1.44 (zym@zym-virtual-machine) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6134-961ee0e81) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT_DYNAMIC Tue Aug 15 06:32:55 2023
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=2ef03856-da35-1a9d-b0ef-533a32ce7902 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009afefff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009aff000-0x0000000009ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000a000000-0x000000000a1fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000a200000-0x000000000a20afff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000a20b000-0x000000000affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000b000000-0x000000000b01ffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000b020000-0x000000007df25fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007df26000-0x000000007df26fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007df27000-0x000000007df3cfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007df3d000-0x000000007df3dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007df3e000-0x00000000bcbfdfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bcbfe000-0x00000000bcedcfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bcedd000-0x00000000bcf19fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000bcf1a000-0x00000000bd202fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bd203000-0x00000000bdf61fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bdf62000-0x00000000bdffefff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000bdfff000-0x00000000beffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bf000000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fd100000-0x00000000fd1fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000fea0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb80000-0x00000000fec01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec30000-0x00000000fec30fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedc2000-0x00000000fedcffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedd4000-0x00000000fedd5fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000001bf33ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000001bf340000-0x00000001bfffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by American Megatrends
[    0.000000] efi: ACPI=0xbd1ec000 ACPI 2.0=0xbd1ec014 TPMFinalLog=0xbd1b6000 SMBIOS=0xbde07000 SMBIOS 3.0=0xbde06000 MEMATTR=0xb9ace018 ESRT=0xb9dedf98 
[    0.000000] SMBIOS 3.3.0 present.
[    0.000000] DMI: BIOSTAR Group A320MH/A320MH, BIOS 5.17 05/05/2022
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 3194.271 MHz processor
[    0.000164] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000167] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000180] last_pfn = 0x1bf340 max_arch_pfn = 0x400000000
[... 829 lines cut ...]
[   17.125790] EXT4-fs (sdc1): recovery complete
[   17.146630] EXT4-fs (sdc1): mounted filesystem with ordered data mode. Quota mode: disabled.
[   17.278280] EXT4-fs (sdd1): recovery complete
[   17.283212] EXT4-fs (sdd1): mounted filesystem with ordered data mode. Quota mode: disabled.
[   19.069435] eth1: 0xffffc90000095000, 00:e0:4c:13:04:33, IRQ 55
[   19.244370] eth0: 0xffffc90000075000, f4:b5:20:1a:90:1a, IRQ 53
[   22.388895] r8168: eth1: link up
[   22.392742] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   22.477195] r8168: eth0: link up
[   22.493299] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   23.128712] pppoe-wan: renamed from ppp0
[   28.766386] traps: check[8643] general protection fault ip:8049007 sp:ffbdaa90 error:402 in check[8049000+1000]
[   31.747182] r8168: eth1: link down
[   34.948869] r8168: eth1: link up
[   44.115902] squashfs: Unknown parameter 'discard'
[   75.899735] snd_hda_intel 0000:04:00.0: couldn't bind with audio component
[   75.907321] snd_hda_intel 0000:04:00.0: HSW/BDW HD-audio HDMI/DP requires binding with gfx driver
[  104.664656] squashfs: Unknown parameter 'discard'
[  104.670109] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x89e45589)
[  104.678258] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
[  104.686375] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x2e48937b)
[  104.694552] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
[  104.703343] exFAT-fs (sda128): invalid boot record signature
[  104.709560] exFAT-fs (sda128): failed to read boot sector
[  104.715499] exFAT-fs (sda128): failed to recognize exfat type

WYC-2020, 2023-08-17:
[image]

试试再这里面package/firmware/linux-firmware/intel.mk   吧上面途中这几个加进去看看，如果已经有了就不用加了，没有的加上瞧瞧，再下图这些区域加就行了
[image]

achinese888, 2023-08-17:
不行啊老哥 
[image]
[image]
[image]
[image]

achinese888, 2023-08-17:
l大应该已经加过支持了 但是编译出来说太旧不支持 
[image]

achinese888, 2023-08-17:
应该是要编译intel 的kmd驱动 但是怎么搞 intel官方只有ubuntu的安装方法

achinese888, 2023-08-17:
> [image]
> 
> 试试再这里面package/firmware/linux-firmware/intel.mk 吧上面途中这几个加进去看看，如果已经有了就不用加了，没有的加上瞧瞧，再下图这些区域加就行了 [image]

不行啊老哥

achinese888, 2023-08-17:
[image]
不支持
应该是要编译intel 的kmd驱动 但是怎么搞 intel官方只有ubuntu的安装方法
l大添加过支持 
[image]

Hyy2001X, 2023-09-13:
主线DG1的编解码（转码）功能用不了，驱动打上了也用处不大，DG1的媒体功能只在Windows下面是正常的

Vinke2, 2024-03-12:
> 主线DG1的编解码（转码）功能用不了，驱动打上了也用处不大，DG1的媒体功能只在Windows下面是正常的

rx580呢

Hyy2001X, 2024-03-12:
> > 主线DG1的编解码（转码）功能用不了，驱动打上了也用处不大，DG1的媒体功能只在Windows下面是正常的
> 
> rx580呢

rx580各种格式不支持，同价位推荐1050或者p4

iori2333, 2024-08-20:
DG1可以在Linux下通过[Backport](https://github.com/intel-gpu/intel-gpu-i915-backports)驱动开启编解码功能，但我安装后进入系统后就没有显示输出了，只能通过SSH控制

coolsnowwolf, 2026-09-05:
日志已明确说明：Linux 6.1.44 的 i915 主动拒绝 PCI ID 4908（DG1），并给出 `i915.force_probe=4908` 仅用于强制试探；正确解决路径是升级到包含该设备支持的更新内核/驱动，而不是加入固件文件。当前 LEDE 的 i915 配置仍保持 `CONFIG_DRM_I915_FORCE_PROBE=""`，不会对未知设备强制加载；评论也确认 DG1 的媒体能力在当时主线并不完整，第三方 backport 还会导致显示输出异常。没有可安全回移的 LEDE 包补丁，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf