coolsnowwolf/lede #11457: xray-core编译出错

ChaingTsung, 2023-08-16:
find /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/pypi/xmltodict-0.12.0/ipkg-aarch64_cortex-a53/python3-xmltodict -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="aarch64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="aarch64-openwrt-linux-musl-nm" STRIP="/home/yoki/openwrt/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/yoki/openwrt/scripts/strip-kmod.sh" PATCHELF="/home/yoki/openwrt/staging_dir/host/bin/patchelf" /home/yoki/openwrt/scripts/rstrip.sh /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/pypi/xmltodict-0.12.0/ipkg-aarch64_cortex-a53/python3-xmltodict
(cd /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/pypi/xmltodict-0.12.0/ipkg-aarch64_cortex-a53/python3-xmltodict/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/yoki/openwrt/bin/packages/aarch64_cortex-a53/packages
/home/yoki/openwrt/staging_dir/host/bin/fakeroot /home/yoki/openwrt/staging_dir/host/bin/bash /home/yoki/openwrt/scripts/ipkg-build -m "" /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/pypi/xmltodict-0.12.0/ipkg-aarch64_cortex-a53/python3-xmltodict /home/yoki/openwrt/bin/packages/aarch64_cortex-a53/packages
Packaged contents of /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/pypi/xmltodict-0.12.0/ipkg-aarch64_cortex-a53/python3-xmltodict into /home/yoki/openwrt/bin/packages/aarch64_cortex-a53/packages/python3-xmltodict_0.12.0-2_aarch64_cortex-a53.ipk
echo "python3-xmltodict" >> /home/yoki/openwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/xmltodict.default.install
make[3]: Leaving directory '/home/yoki/openwrt/feeds/packages/lang/python/xmltodict'
time: package/feeds/packages/xmltodict/compile#5.03#0.96#5.60
make[3]: Entering directory '/home/yoki/openwrt/feeds/packages/net/xray-core'
rm -f /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/Xray-core-1.8.3/.built
touch /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/Xray-core-1.8.3/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/yoki/openwrt/tmp/go-build" GO_MOD_CACHE_DIR="/home/yoki/openwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/xtls/xray-core" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/xtls/xray-core/main" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/Xray-core-1.8.3" GO_BUILD_DIR="/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/Xray-core-1.8.3/.go_work/build" GO_BUILD_BIN_DIR="/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/Xray-core-1.8.3/.go_work/build/bin/linux_arm64" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/yoki/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/share/gocode/src" GOOS="linux" GOARCH="arm64" GO386="" GOAMD64="" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/Xray-core-1.8.3=Xray-core-1.8.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/Xray-core-1.8.3=Xray-core-1.8.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -fuse-ld=bfd -znow -zrelro" GOPATH="/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/Xray-core-1.8.3/.go_work/build" GOCACHE="/home/yoki/openwrt/tmp/go-build" GOMODCACHE="/home/yoki/openwrt/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env bash ../../lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1691851547' -linkmode external -extldflags '-L/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -fuse-ld=bfd -Wl,-z,now -Wl,-z,relro'"        -ldflags " -X github.com/xtls/xray-core/core.build=OpenWrt -X github.com/xtls/xray-core/core.version=1.8.3 -buildid '1691851547' -linkmode external -extldflags '-L/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -fuse-ld=bfd -Wl,-z,now -Wl,-z,relro'"  
Finding targets

Building targets
package github.com/xtls/xray-core/main
	imports github.com/xtls/xray-core/main/distro/all
	imports github.com/xtls/xray-core/main/commands/all
	imports github.com/xtls/xray-core/infra/conf
	imports github.com/xtls/xray-core/proxy/wireguard
	imports gvisor.dev/gvisor/pkg/bufferv2
	imports gvisor.dev/gvisor/pkg/atomicbitops
	imports gvisor.dev/gvisor/pkg/cpuid
	imports gvisor.dev/gvisor/pkg/state
	imports gvisor.dev/gvisor/pkg/state/wire
	imports gvisor.dev/gvisor/pkg/gohacks: build constraints exclude all Go files in /home/yoki/openwrt/dl/go-mod-cache/gvisor.dev/gvisor@v0.0.0-20220901235040-6ca97ef2ce1c/pkg/gohacks

make[3]: *** [Makefile:91: /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/Xray-core-1.8.3/.built] Error 1
make[3]: Leaving directory '/home/yoki/openwrt/feeds/packages/net/xray-core'
time: package/feeds/packages/xray-core/compile#1.19#1.85#4.17
    ERROR: package/feeds/packages/xray-core failed to build.
make[2]: *** [package/Makefile:120: package/feeds/packages/xray-core/compile] Error 1
make[2]: Leaving directory '/home/yoki/openwrt'
make[1]: *** [package/Makefile:114: /home/yoki/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yoki/openwrt'
make: *** [/home/yoki/openwrt/include/toplevel.mk:232：world] 错误 2

WYC-2020, 2023-08-17:
算不出来

ChaingTsung, 2023-08-17:
> 算不出来

不知道为什么会这样，取消xray-core就可以了

KevinMX, 2023-08-17:
你的 log 不完整，没有包含具体的报错信息，等于没有日志，只能算命；所以楼上说「算」不出来。

l4326769, 2023-08-17:
我也出现这个问题了

hcym, 2023-08-17:
一般是最后几句，我的都正常，只有树莓4出错是内核515补丁，删掉也编译过了。你们的应该是网络问题，那个go编译对网络要求很高

wuzhuang123, 2023-08-18:
Finding targets

Building targets
这两之间等了很久吧，网络问题，加入export GOPROXY=https://goproxy.io和export GO111MODULE=on试试

vendemmia-cpu, 2023-08-19:
I also have the same problem. any solution was found?

l4326769, 2023-08-19:
> I also have the same problem. any solution was found?

no，i tried to use history version of packages，is no error of golang , but xray-core still cannot be compiled

vendemmia-cpu, 2023-08-19:
> > I also have the same problem. any solution was found?
> 
> no，i tried to use history version of packages，is no error of golang , but xray-core still cannot be compiled

I downgrade golang to 1.20.7 the problem is solved.

l4326769, 2023-08-20:
> > > I also have the same problem. any solution was found?
> > 
> > 
> > no，i tried to use history version of packages，is no error of golang , but xray-core still cannot be compiled
> 
> I downgrade golang to 1.20.7 the problem is solved.

thanks,is a useful way

dd-bpir3, 2023-09-01:
> 
直接输入吗？

dd-bpir3, 2023-09-01:
> Finding targets
> 
> Building targets 这两之间等了很久吧，网络问题，加入export GOPROXY=[https://goproxy.io和export](https://goproxy.io%E5%92%8Cexport) GO111MODULE=on试试

直接输入吗？还是具体咋操作？

closed 2023-11-13 by aiamadeus

tujj99, 2023-12-26:
> > > I also have the same problem. any solution was found?
> > 
> > 
> > no，i tried to use history version of packages，is no error of golang , but xray-core still cannot be compiled
> 
> I downgrade golang to 1.20.7 the problem is solved.

直接在Makefile里面改版本号和sha256吗？