coolsnowwolf/lede #11461: 编译5.4内核的x86固件报错

bi4tms, 2023-08-18:
拉取最新源码全新编译5.4内核的x86固件报错，日志如下：
ouch /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/.built
touch /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/.built_check
make -C /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.253 KCFLAGS="-fmacro-prefix-map=/home/x86.5.4/lede/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/x86.5.4/lede/staging_dir/host/include -I/home/x86.5.4/lede/staging_dir/hostpkg/include -I/home/x86.5.4/lede/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Aug 16 07:46:28 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/x86.5.4/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8125.symvers /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/fullconenat.symvers /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/button-hotplug.symvers /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/shortcut-fe.symvers" KERNELRELEASE=5.4.253 M=/home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/src modules
make[4]: Entering directory '/home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.253'
  CC [M]  /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/src/r8152.o
In file included from /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/src/r8152.c:32:
/home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/src/compatibility.h:612:28: error: redefinition of 'eth_hw_addr_set'
  612 |         static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                            ^~~~~~~~~~~~~~~
In file included from /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/src/r8152.c:17:
./include/linux/etherdevice.h:298:20: note: previous definition of 'eth_hw_addr_set' with type 'void(struct net_device *, const u8 *)' {aka 'void(struct net_device *, const unsigned char *)'}
  298 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
make[5]: *** [scripts/Makefile.build:262: /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/src/r8152.o] Error 1
make[4]: *** [Makefile:1750: /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/src] Error 2
make[4]: Leaving directory '/home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.253'
make[3]: *** [Makefile:43: /home/x86.5.4/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-dkms-2.16.3-4/.built] Error 2
make[3]: Leaving directory '/home/x86.5.4/lede/package/lean/r8152'
time: package/lean/r8152/compile#3.58#0.90#5.36
    ERROR: package/lean/r8152 failed to build.
make[2]: *** [package/Makefile:116: package/lean/r8152/compile] Error 1
make[2]: Leaving directory '/home/x86.5.4/lede'
make[1]: *** [package/Makefile:110: /home/x86.5.4/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/x86.5.4/lede'
make: *** [/home/x86.5.4/lede/include/toplevel.mk:231：world] 错误 2

xdfx1180, 2023-08-18:
同样错误

bi4tms, 2023-08-18:
> 这玩意我记得修过了

拉取的最新源码，全新编译

ywt114, 2023-08-18:
> 这玩意我记得修过了

前几天你修过，不过2天前5.4内核一更新，r8152又挂了

xiangfeidexiaohuo, 2023-08-18:
> 这玩意我记得修过了

是的 修了 还是炸 ，5.4专炸

butinhi, 2023-08-18:
Me too. AX6s

make[3]: Leaving directory '/home/butinhi/lede/feeds/packages/utils/attr'
time: package/feeds/packages/attr/compile#11.27#1.14#13.65
    ERROR: package/feeds/packages/attr failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/attr/compile] Error 1
make[2]: Leaving directory '/home/butinhi/lede'
make[1]: *** [package/Makefile:110: /home/butinhi/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/butinhi/lede'
make: *** [/home/butinhi/lede/include/toplevel.mk:231: world] Error 2
butinhi@KEN-DELL:~/lede$

xdfx1180, 2023-08-20:
> 这个玩意我记得修改过

怎么修改的？指教一下

WYC-2020, 2023-08-20:
> > 这个玩意我记得修改过
> 
> 怎么修改的？指教一下

那不清楚我昨天才编译过没问题，这个补丁我在另外的一个帖子发过，l大合了的，稍微改了下，看了来应该没啥问题的，自己再多排查排查

bi4tms, 2023-08-20:
> > > 这个玩意我记得修改过
> > 
> > 
> > 怎么修改的？指教一下
> 
> 那不清楚我昨天才编译过没问题，这个补丁我在另外的一个帖子发过，l大合了的，稍微改了下，看了来应该没啥问题的，自己再多排查排查

刚刚试了一下，用大雕的编译失败，换成你上传的编译通过

ywt114, 2023-08-21:
> > > 这个玩意我记得修改过
> > 
> > 
> > 怎么修改的？指教一下
> 
> 那不清楚我昨天才编译过没问题，这个补丁我在另外的一个帖子发过，l大合了的，稍微改了下，看了来应该没啥问题的，自己再多排查排查

确实，把l大修改过的patch改回你修改的版本，确实5.4编译通过。

closed 2023-08-21 by bi4tms