coolsnowwolf/lede #11462: 8月19日源码 编译报错

linuxbelletian, 2023-08-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/ubus-2022-06-01-2bebf93c/.pkgdir/ubusd/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.bnx2x-firmware_installed
SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/luci-lib-ip/.pkgdir/luci-lib-ip/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
make[3]: Entering directory '/home/aber/Workspaces/lede/package/boot/grub2'
echo "bnx2x-firmware" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.uci_installed
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp
echo "uci" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/uci.default.install
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libavahi-dbus-support_installed
SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/linux-firmware-20230625/.pkgdir/i915-firmware/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
make[3]: Leaving directory '/home/aber/Workspaces/lede/package/system/uci'
time: package/system/uci/compile#0.26#0.05#0.77
echo "libavahi-dbus-support" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/avahi.dbus.install
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.ubusd_installed
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp
echo "ubusd" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/ubus.default.install
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-lib-ip_installed
SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/coreutils-9.1/.pkgdir/coreutils/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
make[3]: Leaving directory '/home/aber/Workspaces/lede/package/system/ubus'
time: package/system/ubus/compile#0.27#0.07#0.82
SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/avahi-dbus/avahi-0.8/.pkgdir/avahi-dbus-daemon/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
echo "luci-lib-ip" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/luci-lib-ip.default.install
make[3]: Leaving directory '/home/aber/Workspaces/lede/feeds/luci/libs/luci-lib-ip'
time: package/feeds/luci/luci-lib-ip/compile#0.36#0.11#0.45
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.i915-firmware_installed
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.coreutils_installed
echo "i915-firmware" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.installmake[3]: Leaving directory '/home/aber/Workspaces/lede/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#2.29#0.46#3.13
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.avahi-dbus-daemon_installed
echo "avahi-dbus-daemon" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/avahi.dbus.install
[ ! -d ./src/ ] || cp -fpR ./src/. /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/bpftools-lib/bpftools-2022-03-08-04c465fd
make[3]: Leaving directory '/home/aber/Workspaces/lede/feeds/packages/libs/avahi'
echo "coreutils" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/coreutils.default.install
time: package/feeds/packages/avahi/dbus/compile#0.29#0.03#0.60
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp

Applying ./patches/001-cflags.patch using plaintext:
SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/coreutils-9.1/.pkgdir/coreutils-base64/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
patching file libbpf/src/Makefile

Applying ./patches/002-includes.patch using plaintext:
patching file libbpf/include/linux/list.h
patching file src/Makefile
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.coreutils-base64_installed
make[3]: Entering directory '/home/aber/Workspaces/lede/feeds/packages/libs/avahi'
echo "coreutils-base64" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/coreutils.default.install
touch /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/bpftools-lib/bpftools-2022-03-08-04c465fd/.prepared_c0f30707f81c293453c036c875882dd6_6664517399ebbbc92a37c5bb081b5c53
make[3]: Leaving directory '/home/aber/Workspaces/lede/feeds/packages/utils/coreutils'
[... 104 lines cut ...]
Packaged contents of /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/bpftools-lib/bpftools-2022-03-08-04c465fd/ipkg-x86_64/libbpf into /home/aber/Workspaces/lede/bin/packages/x86_64/base/libbpf20220308_2022-03-08-04c465fd-1_x86_64.ipk
echo "libbpf" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/bpftools.lib.install
rm -rf /home/aber/Workspaces/lede/tmp/stage-bpftools
mkdir -p /home/aber/Workspaces/lede/tmp/stage-bpftools/host /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/packages
install -d -m0755 /home/aber/Workspaces/lede/tmp/stage-bpftools/usr/include/bpf
cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/bpftools-lib/bpftools-2022-03-08-04c465fd/ipkg-install/usr/include/bpf/*.h /home/aber/Workspaces/lede/tmp/stage-bpftools/usr/include/bpf/
install -d -m0755 /home/aber/Workspaces/lede/tmp/stage-bpftools/usr/lib
cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/bpftools-lib/bpftools-2022-03-08-04c465fd/ipkg-install/usr/lib/libbpf.{a,so*} /home/aber/Workspaces/lede/tmp/stage-bpftools/usr/lib/
install -d -m0755 /home/aber/Workspaces/lede/tmp/stage-bpftools/usr/lib/pkgconfig
cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/bpftools-lib/bpftools-2022-03-08-04c465fd/ipkg-install/usr/lib/pkgconfig/libbpf.pc /home/aber/Workspaces/lede/tmp/stage-bpftools/usr/lib/pkgconfig/
/home/aber/Workspaces/lede/staging_dir/host/bin/sed -i -e 's,/usr/include,${prefix}/include,g' /home/aber/Workspaces/lede/tmp/stage-bpftools/usr/lib/pkgconfig/libbpf.pc
/home/aber/Workspaces/lede/staging_dir/host/bin/sed -i -e 's,/usr/lib,${exec_prefix}/lib,g' /home/aber/Workspaces/lede/tmp/stage-bpftools/usr/lib/pkgconfig/libbpf.pc
find /home/aber/Workspaces/lede/tmp/stage-bpftools -name '*.la' | xargs -r rm -f;
if [ -f /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/packages/bpftools.lib.list ]; then /home/aber/Workspaces/lede/scripts/clean-package.sh "/home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/packages/bpftools.lib.list" "/home/aber/Workspaces/lede/staging_dir/target-x86_64_musl"; fi
if [ -d /home/aber/Workspaces/lede/tmp/stage-bpftools ]; then (cd /home/aber/Workspaces/lede/tmp/stage-bpftools; find ./ > /home/aber/Workspaces/lede/tmp/stage-bpftools.files);        SHELL= flock /home/aber/Workspaces/lede/tmp/.staging-dir.flock -c ' mv /home/aber/Workspaces/lede/tmp/stage-bpftools.files /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/packages/bpftools.lib.list && cp -fpR /home/aber/Workspaces/lede/tmp/stage-bpftools/* /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/; '; fi
rm -rf /home/aber/Workspaces/lede/tmp/stage-bpftools
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/stamp/.bpftools.lib_installed
make[3]: Leaving directory '/home/aber/Workspaces/lede/package/network/utils/bpftools'
time: package/network/utils/bpftools/lib/compile#18.94#0.78#4.28
make[2]: Leaving directory '/home/aber/Workspaces/lede'
make[1]: *** [package/Makefile:110: /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/aber/Workspaces/lede'
make[1]: Leaving directory '/home/aber/Workspaces/lede'
make: *** [/home/aber/Workspaces/lede/include/toplevel.mk:231: world] Error 2

flyzstu, 2023-08-19:
不使用单线程编译很难发现问题，建议使用make -j1 V=s重新编译

linuxbelletian, 2023-08-20:
补充单线程编译日志:

SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-sit/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-sit_installed
echo "kmod-sit" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-tun/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-tun_installed
echo "kmod-tun" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-slhc/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-slhc_installed
echo "kmod-slhc" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-ppp/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-ppp_installed
echo "kmod-ppp" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-pppox/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-pppox_installed
echo "kmod-pppox" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-pppoe/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-pppoe_installed
echo "kmod-pppoe" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-mppe/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-mppe_installed
echo "kmod-mppe" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-tcp-bbr/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-tcp-bbr_installed
echo "kmod-tcp-bbr" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-mdio/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-mdio_installed
echo "kmod-mdio" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/aber/Workspaces/lede/tmp/.root-copy.flock -c 'cp -fpR /home/aber/Workspaces/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-wireguard/. /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-wireguard_installed
echo "kmod-wireguard" >> /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/root-x86/stamp

[... cut ...]
time: package/utils/util-linux/compile#0.21#0.03#0.23
    ERROR: package/utils/util-linux failed to build.
make[2]: *** [package/Makefile:116: package/utils/util-linux/compile] Error 1
make[2]: Leaving directory '/home/aber/Workspaces/lede'
make[1]: *** [package/Makefile:110: /home/aber/Workspaces/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/aber/Workspaces/lede'
make: *** [/home/aber/Workspaces/lede/include/toplevel.mk:231: world] Error 2

linuxbelletian, 2023-08-20:
看出来了, 应该是 python3.8没有安装

linuxbelletian, 2023-08-20:
ubuntu22.04 lts 现在不单独列 python3.8到 bin下了, 默认直接是 python3 了

closed 2023-08-25 by aiamadeus