coolsnowwolf/lede #11464: NX30Pro编译固件MAC地址不对

OldCoding, 2023-08-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：用最新源码编译出来的固件，在接口中有lan1、lan2、lan3、eth1、eth0、br-lan几个接口，其中eth0和br-lan获取到的mac地址是随机生成的，其余几个接口是固定的，不清楚是不是dsa的原因，6月份lean大佬发布的潘多拉和237大佬发布的闭源驱动Openwrt都是固定MAC，唯独用lede自行编译的就是随机的，一重启MAC地址就变了，我是用作纯AP的，在软路由上分配固定IP的话就会有影响

### (2) 路由器型号和固件版本

A：NX30Pro、最新源码编译固件

### (3) 详细日志

A：
官方固件
br-lan    Link encap:Ethernet  HWaddr 34:CA:81:**:**:3D
          inet addr:192.168.124.1  Bcast:192.168.124.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1560 errors:0 dropped:1 overruns:0 frame:0
          TX packets:759 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:170294 (166.3 KiB)  TX bytes:109990 (107.4 KiB)

eth0      Link encap:Ethernet  HWaddr 34:CA:81:**:**:3C
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1556 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1406 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:190885 (186.4 KiB)  TX bytes:217223 (212.1 KiB)
          Interrupt:75

eth1      Link encap:Ethernet  HWaddr 34:CA:81:**:**:3C
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:75

lan1      Link encap:Ethernet  HWaddr 34:CA:81:**:**:3C
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1042 errors:0 dropped:0 overruns:0 frame:0
          TX packets:849 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:94709 (92.4 KiB)  TX bytes:109730 (107.1 KiB)

lan2      Link encap:Ethernet  HWaddr 34:CA:81:**:**:3C
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lan3      Link encap:Ethernet  HWaddr 34:CA:81:**:**:3C
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:514 errors:0 dropped:0 overruns:0 frame:0
          TX packets:527 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:68168 (66.5 KiB)  TX bytes:90466 (88.3 KiB)

237大佬的
br-lan    Link encap:Ethernet  HWaddr 00:78:32**:**:87
          inet addr:192.168.6.1  Bcast:192.168.6.255  Mask:255.255.255.0
          inet6 addr: fe80::278:32ff:fe26:6087/64 Scope:Link
[... 500 lines cut ...]
[ 3903.545328] device lan2 entered promiscuous mode
[ 3903.555015] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[ 3903.564758] br-lan: port 5(lan3) entered blocking state
[ 3903.570018] br-lan: port 5(lan3) entered disabled state
[ 3903.577154] device lan3 entered promiscuous mode
[ 3903.585043] br-lan: port 6(wlan0) entered blocking state
[ 3903.590421] br-lan: port 6(wlan0) entered disabled state
[ 3903.595955] device wlan0 entered promiscuous mode
[ 3903.602622] br-lan: port 7(wlan1-1) entered blocking state
[ 3903.608163] br-lan: port 7(wlan1-1) entered disabled state
[ 3903.613880] device wlan1-1 entered promiscuous mode
[ 3903.620669] br-lan: port 7(wlan1-1) entered blocking state
[ 3903.626173] br-lan: port 7(wlan1-1) entered forwarding state
[ 3903.631923] br-lan: port 6(wlan0) entered blocking state
[ 3903.637253] br-lan: port 6(wlan0) entered forwarding state
[ 3903.642786] br-lan: port 1(bat0) entered blocking state
[ 3903.648024] br-lan: port 1(bat0) entered forwarding state
[ 3903.756944] batman_adv: bat0: No IGMP Querier present - multicast optimizations disabled
[ 3903.765036] batman_adv: bat0: No MLD Querier present - multicast optimizations disabled
[ 3905.981272] mt7530 mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 3905.988809] br-lan: port 3(lan1) entered blocking state
[ 3905.994029] br-lan: port 3(lan1) entered forwarding state
[ 3906.388916] mt7530 mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control rx/tx
[ 3906.396436] br-lan: port 4(lan2) entered blocking state
[ 3906.401655] br-lan: port 4(lan2) entered forwarding state

coolsnowwolf, 2023-08-21:
[image]
是不是刷错 uboot 丢失了分区？

OldCoding, 2023-08-21:
> [image] 是不是刷错 uboot 丢失了分区？

没有刷错，换回官方固件就正常了，换潘多拉也正常

OldCoding, 2023-08-21:
> lan1、lan2、lan3、eth1都能正常读取到设备MAC，唯独eth0和br-lan是随机的

coolsnowwolf, 2023-08-21:
截个图看看目前的 br-lan 地址？

OldCoding, 2023-08-21:
> 截个图看看目前的 br-lan 地址？

34:CA开头的是真实地址，在pdt_data_1分区里面也能找到
[image]

OldCoding, 2023-08-21:
> 截个图看看目前的 br-lan 地址？

不好意思我找到原因了，是我配置的问题，我把wan口删了，然后把eth1和eth0全部划分到lan里面的

closed 2023-08-21 by OldCoding