coolsnowwolf/lede #11478: fw3 reload 导致 upnp工作异常

ostrichii, 2023-08-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
upnpd在已经有客户端添加了映射规则的情况下，如果动了firewall规则，重启或者重载了firewall之后，upnpd 无法正确处理给客户端返回的已经存在的映射规则，导致客户端重复添加规则或者添加规则失败。

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述
重启或者reload firewall规则之前，使用Upnpc请求已经存在的
[image]

重启或者reload firewall规则之后
[image]

/etc/init.d/miniupnpd restart 重启后恢复

目前在hotplug以及firewall服务中涉及到 fw3 reload的后面添加了/etc/init.d/miniupnpd restart ，firewall 服务重启或者重载之后重启miniupnpd来规避，但是觉得不够优雅，希望屌大的看看能不能解

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：X86，R5S均可复现

### (3) 详细日志

无日志，上面的方法可以稳定复现。

coolsnowwolf, 2023-08-28:
[image]

ostrichii, 2023-08-28:
多谢大雕

ostrichii, 2023-08-28:
> [image]

root@OpenWrt:~# uci get firewall.miniupnpd.family
any
root@OpenWrt:~# uci get firewall.miniupnpd.reload
1
root@OpenWrt:~# uci get firewall.miniupnpd
include
root@OpenWrt:~# uci get firewall.miniupnpd.type
script
root@OpenWrt:~# uci get firewall.miniupnpd.path
/usr/share/miniupnpd/firewall.include
root@OpenWrt:~# uci get firewall.miniupnpd.family
any
root@OpenWrt:~# uci get firewall.miniupnpd.reload
1
root@OpenWrt:~# service firewall reload
=======firewall reload之后看起来还是有问题，restart之后miniupnpd规则会丢失========

reload之后的结果：
ostrich@ubuntu:~$ upnpc -L
Found valid IGD : http://192.168.50.1:5000/ctl/IPConn
Local LAN ip address : 192.168.50.241
 i protocol exPort->inAddr:inPort description remoteHost leaseTime
 0 UDP 34217->0.0.0.0:0     '34217-bilibili' '' 0

restart之后：
ostrich@ubuntu:~$ upnpc -L
upnpc : miniupnpc library test client, version 2.1.
 (c) 2005-2019 Thomas Bernard.
Go to http://miniupnp.free.fr/ or https://miniupnp.tuxfamily.org/
for more information.
List of UPNP devices found on the network :
Found valid IGD : http://192.168.50.1:5000/ctl/IPConn
Local LAN ip address : 192.168.50.241
 i protocol exPort->inAddr:inPort description remoteHost leaseTime

看了下，其实无论是restart还是reload firewall之后，看firewall里面的miniupnpd转发规则都是在的并且是正确的，只是通过客户端upnpc向路由器读取列表的时候有问题，所以判断是处理读取并且返回这部分转发规则的Upnpd服务逻辑哪里有问题，没有读取到firewall的转发规则，返回了0.0.0.0，根据关键字搜索定位到了upnpsoap.c，供大雕方便的时候瞄一眼

ostrichii, 2023-09-06:
> root@OpenWrt:~ # uci get firewall.miniupnpd.family any root@OpenWrt:~ # uci get firewall.miniupnpd.reload 1 root@OpenWrt:~ # uci get firewall.miniupnpd include root@OpenWrt:~ # uci get [...]

跟到了获取规则数据的upnpreplyparse.c文件中的GetValueFromNameValueList函数，然后就找不到这个读的数据是从哪儿写进来了-_-

ostrichii, 2023-09-06:
> 跟到了获取规则数据的upnpreplyparse.c文件中的GetValueFromNameValueList函数，然后就找不到这个读的数据是从哪儿写进来了-_-

附上抓包请求数据

coolsnowwolf, 2026-09-05:
已修复并推送 packages 仓库 commit `4b06b7ea`。iptables 与 nftables 两个 miniupnpd 变体的 `service_triggers()` 现在除监听 `upnpd` 配置外，也监听 `firewall` reload。这样 firewall 重载重建规则链时会自动重启 miniupnpd，重新发布已有映射，避免客户端看到旧状态或重复添加失败。两个 init 脚本均通过 `sh -n` 校验；本地包编译尝试被当前 checkout 的无效/未配置 kernel 目标阻塞，非本次改动错误。

closed 2026-09-05 by coolsnowwolf