coolsnowwolf/lede #11480: 在线编译空间不够导致失败，之前可正常编译

zzy19868, 2023-08-24:
[image]
[P3TERX]大神的脚本，进行到这一步，占用就高达61%了，之前只有30%左右，请问怎么解决？

binge8, 2023-08-25:
最简单的你就参考L大哥怎么清理臃肿的ubuntu的，至少比你现在多出15G左右

VIKINGYFY, 2023-08-26:
https://github.com/VIKINGYFY/OpenWRT-CI

用我的项目即可
编译完成时的空间使用情况
[image]

BenjaminX, 2023-08-27:
https://github.com/BenjaminX/NanoPiR4SOpenWrt/blob/main/.github/workflows/build-openwrt.yml

    - name: Free Disk Space (Ubuntu)
      uses: jlumbroso/free-disk-space@main
      with:
        # this might remove tools that are actually needed,
        # if set to "true" but frees about 6 GB
        tool-cache: true
        
        # all of these default to true, but feel free to set to
        # "false" if necessary for your workflow
        android: true
        dotnet: true
        haskell: true
        large-packages: true
        swap-storage: true

zzy19868, 2023-08-27:
> https://github.com/BenjaminX/NanoPiR4SOpenWrt/blob/main/.github/workflows/build-openwrt.yml
>
> ```
> - name: Free Disk Space (Ubuntu)
>   uses: jlumbroso/free-disk-space@main
>   with:
>     # this might [...]

解决问题，多谢！

xuexijiaoben, 2023-09-19:
> https://github.com/BenjaminX/NanoPiR4SOpenWrt/blob/main/.github/workflows/build-openwrt.yml
>
> ```
> - name: Free Disk Space (Ubuntu)
>   uses: jlumbroso/free-disk-space@main
>   with:
>     # this might [...]

我加了这个代码也失败，服了

terada46, 2024-08-30:
> https://github.com/BenjaminX/NanoPiR4SOpenWrt/blob/main/.github/workflows/build-openwrt.yml
>
> ```
> - name: Free Disk Space (Ubuntu)
>   uses: jlumbroso/free-disk-space@main
>   with:
>     # this might [...]

解决了问题，感谢🙏

coolsnowwolf, 2026-09-05:
该 issue 是 GitHub Actions/在线编译器运行环境磁盘空间不足，不是 LEDE 包或构建系统回归。评论已给出清理 runner 工具链/缓存的方案，后续也确认通过释放磁盘解决；仓库源码无需修改，关闭为 completed。

closed 2026-09-05 by coolsnowwolf