coolsnowwolf/lede #11481: uboot-rockchip 出错

xiangfeidexiaohuo, 2023-08-25:
### (1) 具体问题

A：rk3399/3568 uboot-rockchip编译失败

### (2) 路由器型号和固件版本

A：r4s、r5s、h68k等

### (3) 详细日志

A：

2023-08-24T23:06:34.8545562Z if aarch64-openwrt-linux-musl-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I.. -I. -I. -I../lib -I.. -I. -I./../libelf -I./../libebl -I./../libdw -I./../libdwelf -I./../debuginfod -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Werror -Wunused -Wextra -Wstack-usage=262144 -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/build_dir/target-aarch64_generic_musl/elfutils-0.188=elfutils-0.188 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral -c -o relocate.os -fPIC -DPIC -DSHARED -DSYMBOL_VERSIONING -MT relocate.os -MD -MP \
2023-08-24T23:06:34.8547834Z   -MF ".deps/relocate.Tpo" `test -f 'relocate.c' || echo './'`relocate.c; \
2023-08-24T23:06:34.8548223Z then cat ".deps/relocate.Tpo" >> ".deps/relocate.Po"; \
2023-08-24T23:06:34.8548599Z      rm -f ".deps/relocate.Tpo"; \
2023-08-24T23:06:34.8548942Z else rm -f ".deps/relocate.Tpo"; exit 1; \
2023-08-24T23:06:34.8549196Z fi
2023-08-24T23:06:35.0901246Z if aarch64-openwrt-linux-musl-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I.. -I. -I. -I../lib -I.. -I. -I./../libelf -I./../libebl -I./../libdw -I./../libdwelf -I./../debuginfod -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Werror -Wunused -Wextra -Wstack-usage=262144 -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/build_dir/target-aarch64_generic_musl/elfutils-0.188=elfutils-0.188 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral -c -o dwfl_module_build_id.os -fPIC -DPIC -DSHARED -DSYMBOL_VERSIONING -MT dwfl_module_build_id.os -MD -MP \
2023-08-24T23:06:35.0907710Z   -MF ".deps/dwfl_module_build_id.Tpo" `test -f 'dwfl_module_build_id.c' || echo './'`dwfl_module_build_id.c; \
2023-08-24T23:06:35.0908223Z then cat ".deps/dwfl_module_build_id.Tpo" >> ".deps/dwfl_module_build_id.Po"; \
2023-08-24T23:06:35.0908670Z      rm -f ".deps/dwfl_module_build_id.Tpo"; \
2023-08-24T23:06:35.0909211Z else rm -f ".deps/dwfl_module_build_id.Tpo"; exit 1; \
2023-08-24T23:06:35.0909550Z fi
2023-08-24T23:06:35.1692604Z if aarch64-openwrt-linux-musl-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I.. -I. -I. -I../lib -I.. -I. -I./../libelf -I./../libebl -I./../libdw -I./../libdwelf -I./../debuginfod -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Werror -Wunused -Wextra -Wstack-usage=262144 -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/build_dir/target-aarch64_generic_musl/elfutils-0.188=elfutils-0.188 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral -c -o dwfl_module_report_build_id.os -fPIC -DPIC -DSHARED -DSYMBOL_VERSIONING -MT dwfl_module_report_build_id.os -MD -MP \
2023-08-24T23:06:35.1695594Z   -MF ".deps/dwfl_module_report_build_id.Tpo" `test -f 'dwfl_module_report_build_id.c' || echo './'`dwfl_module_report_build_id.c; \
2023-08-24T23:06:35.1696219Z then cat ".deps/dwfl_module_report_build_id.Tpo" >> ".deps/dwfl_module_report_build_id.Po"; \
2023-08-24T23:06:35.1696777Z      rm -f ".deps/dwfl_module_report_build_id.Tpo"; \
2023-08-24T23:06:35.1697298Z else rm -f ".deps/dwfl_module_report_build_id.Tpo"; exit 1; \
2023-08-24T23:06:35.1697637Z fi
2023-08-24T23:06:35.2380001Z if aarch64-openwrt-linux-musl-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I.. -I. -I. -I../lib -I.. -I. -I./../libelf -I./../libebl -I./../libdw -I./../libdwelf -I./../debuginfod -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Werror -Wunused -Wextra -Wstack-usage=262144 -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/build_dir/target-aarch64_generic_musl/elfutils-0.188=elfutils-0.188 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral -c -o derelocate.os -fPIC -DPIC -DSHARED -DSYMBOL_VERSIONING -MT derelocate.os -MD -MP \
2023-08-24T23:06:35.2383210Z   -MF ".deps/derelocate.Tpo" `test -f 'derelocate.c' || echo './'`derelocate.c; \
2023-08-24T23:06:35.2383816Z then cat ".deps/derelocate.Tpo" >> ".deps/derelocate.Po"; \
2023-08-24T23:06:35.2384606Z      rm -f ".deps/derelocate.Tpo"; \
2023-08-24T23:06:35.2385205Z else rm -f ".deps/derelocate.Tpo"; exit 1; \
2023-08-24T23:06:35.2385884Z fi
2023-08-24T23:06:35.3845257Z binman: [Errno 2] No such file or directory: '/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/target-aarch64_generic_musl/image/rk3399_bl31_v1.35.elf'
2023-08-24T23:06:35.3893177Z if aarch64-openwrt-linux-musl-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I.. -I. -I. -I../lib -I.. -I. -I./../libelf -I./../libebl -I./../libdw -I./../libdwelf -I./../debuginfod -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Werror -Wunused -Wextra -Wstack-usage=262144 -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/build_dir/target-aarch64_generic_musl/elfutils-0.188=elfutils-0.188 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral -c -o offline.os -fPIC -DPIC -DSHARED -DSYMBOL_VERSIONING -MT offline.os -MD -MP \
2023-08-24T23:06:35.3896073Z   -MF ".deps/offline.Tpo" `test -f 'offline.c' || echo './'`offline.c; \
[... cut ...]
2023-08-24T22:32:47.0976177Z make[3]: Leaving directory '/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/package/boot/uboot-rockchip'
2023-08-24T22:32:47.0985394Z time: package/boot/uboot-rockchip/nanopi-r5s-rk3568/compile#82.56#10.34#95.66
2023-08-24T22:32:47.0990829Z     ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r5s-rk3568).
2023-08-24T22:32:47.0997514Z make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1

应该是这个5eda7bd5919d395da728c4110ba4f8be277c07bb提交的问题

BenjaminX, 2023-08-25:
[1_build.txt](https://github.com/coolsnowwolf/lede/files/12435685/1_build.txt)

closed 2023-08-25 by coolsnowwolf

smallprogram, 2023-08-26:
解决了？

xiangfeidexiaohuo, 2023-08-26:
> 解决了？

没有 目前最新源码一样是报错，应该可能还有提交，没整出来