coolsnowwolf/lede #11493: aliyundrive-webdav报错

binge8, 2023-08-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：aliyundrive-webdav报错

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：看截图
[image]

woni928, 2023-08-29:
一样，怎么又炸？那位同学知道怎么解决吗？

Jason6111, 2023-08-29:
我恢复到2.2.1了

woni928, 2023-08-29:
> 我恢复到2.2.1了

云编译上具体怎么操作可以恢复到2.2.1?

Jason6111, 2023-08-29:
> > 我恢复到2.2.1了
> 
> 云编译上具体怎么操作可以恢复到2.2.1?

''
curl -o ./feeds/packages/multimedia/aliyundrive-webdav/Makefile https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/trunk/other/ali/Makefile
''  
试试看

Jason6111, 2023-08-29:
> > 我恢复到2.2.1了
> 
> 云编译上具体怎么操作可以恢复到2.2.1?
`
curl -o ./feeds/packages/multimedia/aliyundrive-webdav/Makefile https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/aliyun/Makefile
`试试

woni928, 2023-08-29:
> > > 我恢复到2.2.1了
> > 
> > 
> > 云编译上具体怎么操作可以恢复到2.2.1?
> > `curl -o ./feeds/packages/multimedia/aliyundrive-webdav/Makefile https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/aliyun/Makefile`试试

能告诉一下以后遇到这种类似的问题，怎么处理？https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/aliyun/Makefile  这种链接怎么去找？
谢谢！

woni928, 2023-08-29:
> > > > > 我恢复到2.2.1了
> > > > 
> > > > 
> > > > 云编译上具体怎么操作可以恢复到2.2.1?
> > > > `curl -o ./feeds/packages/multimedia/aliyundrive-webdav/Makefile https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/aliyun/Makefile`试试
> > 
> > 
> > 能告诉一下以后遇到这种类似的问题，怎么处理？https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/aliyun/Makefile 这种链接怎么去找？ 谢谢！
> 
> 去找历史记录 然后放到自己的一个地方让他curl

去哪里找历史记录？小白一个，很多操作束手无策，还请指点一二，感谢！

Jason6111, 2023-08-29:
> 去哪里找历史记录？小白一个，很多操作束手无策，还请指点一二，感谢！

[image]

woni928, 2023-08-29:
> > > > > > > 我恢复到2.2.1了
> > > > > > 
> > > > > > 
> > > > > > 云编译上具体怎么操作可以恢复到2.2.1?
> > > > > > `curl -o ./feeds/packages/multimedia/aliyundrive-webdav/Makefile https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/aliyun/Makefile`试试
> > > > 
> > > > 
> > > > 能告诉一下以后遇到这种类似的问题，怎么处理？https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/aliyun/Makefile 这种链接怎么去找？ 谢谢！
> > > 
> > > 
> > > 去找历史记录 然后放到自己的一个地方让他curl
> > 
> > 
> > 去哪里找历史记录？小白一个，很多操作束手无策，还请指点一二，感谢！
> 
> [image] [image] [image] 在找对应的文件

非常感谢兄弟的图文教程，再次感谢！

liwenjie119, 2023-09-02:
好几天了，lean也没修复

Jason6111, 2023-09-02:
> 好几天了，lean也没修复，今天看编译又出错了

这个其实跟lede没啥太大关系是你本地的rustc版本低于它要求的1.70.0自己更新下就可以

lrgcx, 2023-09-02:
> > 好几天了，lean也没修复，今天看编译又出错了
> 
> 这个其实跟lede没啥太大关系是你本地的rustc版本低于它要求的1.70.0自己更新下就可以

小白诚恳请教怎么更新？先谢过！

Jason6111, 2023-09-02:
> > > 好几天了，lean也没修复，今天看编译又出错了
> > 
> > 
> > 这个其实跟lede没啥太大关系是你本地的rustc版本低于它要求的1.70.0自己更新下就可以
> 
> 小白诚恳请教怎么更新？先谢过！

尝试`curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- -y
source $HOME/.cargo/env`

lrgcx, 2023-09-03:
提示  error: error: Found argument 'source' which wasn't expected, or isn't valid in this context

USAGE:
    rustup-init [OPTIONS]

For more information try --help

r1172464137, 2023-09-03:
feeds/packages/lang下的rust打patch升级，patch在官方package里有

Jason6111, 2023-09-03:
> 提示 error: error: Found argument 'source' which wasn't expected, or isn't valid in this context
> 
> USAGE: rustup-init [OPTIONS]
> 
> For more information try --help

diy2里尝试添加'sed -i 's/1.68.1/1.72.0/g' feetd/packages/lang/rust
/Makefile
sed -i 's/ccb051df5701d4c588e3d0558f83e73e7eea0a9b165dab3e39dd2db8a6a25d03/ea9d61bbb51d76b6ea681156f69f0e0596b59722f04414b01c6e100b4b5be3a1/g' feetd/packages/lang/rust
/Makefile'

lrgcx, 2023-09-03:
太菜了，不会弄。🙏

liwenjie119, 2023-09-03:
> > 提示 error: error: Found argument 'source' which wasn't expected, or isn't valid in this context
> > USAGE: rustup-init [OPTIONS]
> > For more information try --help
> 
> diy2里尝试添加`sed -i 's/1.68.1/1.72.0/g' feeds/packages/lang/rust/Makefile sed -i 's/ccb051df5701d4c588e3d0558f83e73e7eea0a9b165dab3e39dd2db8a6a25d03/ea9d61bbb51d76b6ea681156f69f0e0596b59722f04414b01c6e100b4b5be3a1/g' feeds/packages/lang/rust/Makefile`

不太对，这样的话patch报错，我看你提commit了，但是patch没动
```
. /home/user/1/lede/include/shell.sh; /home/user/1/lede/staging_dir/host/bin/libdeflate-gzip -dc /home/user/1/lede/dl/rustc-1.72.0-src.tar.gz | tar -C /home/user/1/lede/build_dir/hostpkg/rustc-1.72.0-src/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/user/1/lede/build_dir/hostpkg/rustc-1.72.0-src

Applying ./patches/0001-Update-xz2-and-use-it-static.patch using plaintext:
patching file Cargo.lock
Hunk #1 FAILED at 2490.
Hunk #2 FAILED at 6341.
2 out of 2 hunks FAILED -- saving rejects to file Cargo.lock.rej
patching file src/bootstrap/Cargo.lock
Hunk #1 succeeded at 425 (offset 74 lines).
Hunk #2 succeeded at 894 (offset 109 lines).
patching file src/bootstrap/Cargo.toml
Hunk #1 FAILED at 42.
1 out of 1 hunk FAILED -- saving rejects to file src/bootstrap/Cargo.toml.rej
Patch failed!  Please fix ./patches/0001-Update-xz2-and-use-it-static.patch!
make[3]: *** [Makefile:109: /home/user/1/lede/build_dir/hostpkg/rustc-1.72.0-src/.prepared21a72118757e86c1ffda0be071226730_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/user/1/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#4.60#10.94#589.57
    ERROR: package/feeds/packages/rust [host] failed to build.
```

Jason6111, 2023-09-03:
> 不太对，这样的话patch报错，我看你提commit了，但是patch没动
>
> ```
> . /home/user/1/lede/include/shell.sh; /home/user/1/lede/staging_dir/host/bin/libdeflate-gzip -dc /home/user/1/lede/dl/rustc-1.72.0-src.tar.gz | tar -C [...]

感谢尝试 还敢再试下嘛`curl -o ./feeds/packages/lang/rust/Makefile https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/rust/Makefile
curl -o ./feeds/packages/lang/rust/patches/0001-Update-xz2-and-use-it-static.patch https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/rust/0001-Update-xz2-and-use-it-static.patch`

liwenjie119, 2023-09-03:
> 感谢尝试 还敢再试下嘛`curl -o ./feeds/packages/lang/rust/Makefile https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/rust/Makefile curl -o [...]

目前卡在这个报错了：
```
make[5]: Leaving directory '/home/user/1/lede/build_dir/target-x86_64_musl/libselinux-3.3/include'
make[5]: Entering directory '/home/user/1/lede/build_dir/target-x86_64_musl/libselinux-3.3/src'
test -d /home/user/1/lede/build_dir/target-x86_64_musl/libselinux-3.3/ipkg-install/usr/lib || install -m 755 -d /home/user/1/lede/build_dir/target-x86_64_musl/libselinux-3.3/ipkg-install/usr/lib
install: error while loading shared libraries: libpcre.so.1: cannot open shared object file: No such file or directory
make[5]: *** [Makefile:176: install] Error 127
make[5]: Leaving directory '/home/user/1/lede/build_dir/target-x86_64_musl/libselinux-3.3/src'
make[4]: *** [Makefile:48: install] Error 1
make[4]: Leaving directory '/home/user/1/lede/build_dir/target-x86_64_musl/libselinux-3.3'
make[3]: *** [Makefile:163: /home/user/1/lede/build_dir/target-x86_64_musl/libselinux-3.3/.built] Error 2
make[3]: Leaving directory '/home/user/1/lede/package/libs/libselinux'
time: package/libs/libselinux/compile#0.17#0.00#0.16
    ERROR: package/libs/libselinux failed to build.
make[2]: *** [package/Makefile:116: package/libs/libselinux/compile] Error 1
```
但是 好像和rust没关系？因为我最后搜索了一下，在lede\bin\packages\x86_64\packages内找到了最新时间的aliyundrive-webdav_2.3.2-1_x86_64.ipk，luci-app-aliyundrive-webdav_2.2.1-1_all.ipk luci-i18n-aliyundrive-webdav-zh-cn_2.2.1-1_all.ipk，应该是能正常编译了。不过我没时间继续测试了，一会儿要出门
[image]

Jason6111, 2023-09-03:
已经发起提交等待合并

binge8, 2023-09-03:
已测试，还是有点问题
make[3]: Entering directory '/home/runner/work/op/op/openwrt/feeds/packages/lang/rust'
[12062](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12063)
( cd /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/host/rust-x86_64-unknown-linux-musl/rustc-1.72.0-src ; python3 x.py --config ./config.toml dist build-manifest cargo llvm-tools rustc rust-std rust-src ; )
[12063](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12064)
Building bootstrap
[12064](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12065)
      Updating crates.io index
[12065](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12066)
  error: failed to select a version for `xz2`.
[12066](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12067)
      ... required by package `bootstrap v0.0.0 (/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/host/rust-x86_64-unknown-linux-musl/rustc-1.72.0-src/src/bootstrap)`
[12067](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12068)
  versions that meet the requirements `^0.1` (locked to 0.1.6) are: 0.1.6
[12068](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12069)
  
[12069](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12070)
  the package `bootstrap` depends on `xz2`, with features: `static` but `xz2` does not have these features.
[12070](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12071)
  
[12071](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12072)
  
[12072](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12073)
  failed to select a version for `xz2` which could resolve this conflict
[12073](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12074)
  failed to run: /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/host/rust-x86_64-unknown-linux-musl/rustc-1.72.0-src/build/x86_64-unknown-linux-gnu/stage0/bin/cargo build --manifest-path /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/host/rust-x86_64-unknown-linux-musl/rustc-1.72.0-src/src/bootstrap/Cargo.toml
[12074](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12075)
  Build completed unsuccessfully in 0:00:00
[12075](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12076)
  make[3]: *** [Makefile:104: /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/host/rust-x86_64-unknown-linux-musl/rustc-1.72.0-src/.built] Error 1
[12076](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12077)
  make[3]: Leaving directory '/home/runner/work/op/op/openwrt/feeds/packages/lang/rust'
[12077](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12078)
  time: package/feeds/packages/rust/host-compile#0.26#0.11#0.52
[12078](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12079)
      ERROR: package/feeds/packages/rust [host] failed to build.
[12079](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12080)
  make[2]: *** [package/Makefile:114: package/feeds/packages/rust/host/compile] Error 1
[12080](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12081)
  make[2]: Leaving directory '/home/runner/work/op/op/openwrt'
[12081](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12082)
  make[1]: *** [package/Makefile:110: /home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
[12082](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12083)
  make[1]: Leaving directory '/home/runner/work/op/op/openwrt'
[12083](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12084)
  make: *** [/home/runner/work/op/op/openwrt/include/toplevel.mk:231: world] Error 2

>

binge8, 2023-09-03:
> 已经发起提交等待合并

已测试，还是有点问题
make[3]: Entering directory '/home/runner/work/op/op/openwrt/feeds/packages/lang/rust'
[12062](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12063)
( cd /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/host/rust-x86_64-unknown-linux-musl/rustc-1.72.0-src ; python3 x.py --config ./config.toml dist build-manifest cargo llvm-tools rustc rust-std rust-src ; )
[12063](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12064)
Building bootstrap
[12064](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12065)
      Updating crates.io index
[12065](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12066)
  error: failed to select a version for `xz2`.
[12066](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12067)
      ... required by package `bootstrap v0.0.0 (/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/host/rust-x86_64-unknown-linux-musl/rustc-1.72.0-src/src/bootstrap)`
[12067](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12068)
  versions that meet the requirements `^0.1` (locked to 0.1.6) are: 0.1.6
[12068](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12069)
  
[12069](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12070)
  the package `bootstrap` depends on `xz2`, with features: `static` but `xz2` does not have these features.
[12070](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12071)
  
[12071](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12072)
  
[12072](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12073)
  failed to select a version for `xz2` which could resolve this conflict
[12073](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12074)
  failed to run: /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/host/rust-x86_64-unknown-linux-musl/rustc-1.72.0-src/build/x86_64-unknown-linux-gnu/stage0/bin/cargo build --manifest-path /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/host/rust-x86_64-unknown-linux-musl/rustc-1.72.0-src/src/bootstrap/Cargo.toml
[12074](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12075)
  Build completed unsuccessfully in 0:00:00
[12075](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12076)
  make[3]: *** [Makefile:104: /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/host/rust-x86_64-unknown-linux-musl/rustc-1.72.0-src/.built] Error 1
[12076](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12077)
  make[3]: Leaving directory '/home/runner/work/op/op/openwrt/feeds/packages/lang/rust'
[12077](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12078)
  time: package/feeds/packages/rust/host-compile#0.26#0.11#0.52
[12078](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12079)
      ERROR: package/feeds/packages/rust [host] failed to build.
[12079](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12080)
  make[2]: *** [package/Makefile:114: package/feeds/packages/rust/host/compile] Error 1
[12080](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12081)
  make[2]: Leaving directory '/home/runner/work/op/op/openwrt'
[12081](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12082)
  make[1]: *** [package/Makefile:110: /home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
[12082](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12083)
  make[1]: Leaving directory '/home/runner/work/op/op/openwrt'
[12083](https://github.com/binge8/op/actions/runs/6064285344/job/16452558605#step:9:12084)
  make: *** [/home/runner/work/op/op/openwrt/include/toplevel.mk:231: world] Error 2

Jason6111, 2023-09-03:
本地编译？建议make clean 云编译是通过的
[image]
[image]

binge8, 2023-09-04:
> 本地编译？建议make clean 云编译是通过的 [image] [image]

就是云编译，每次都最新环境
[image]

Jason6111, 2023-09-04:
> > 本地编译？建议make clean 云编译是通过的 [image] [image]
> 
> 就是云编译，每次都最新环境 [image]

你的项目地址在哪

binge8, 2023-09-04:
> > > 本地编译？建议make clean 云编译是通过的 [image] [image]
> > 
> > 
> > 就是云编译，每次都最新环境 [image]
> 
> 你的项目地址在哪

https://github.com/coolsnowwolf/lede/issues/11493#issuecomment-1704466522   我不确定你能不能看见日志，好像要授权？

Jason6111, 2023-09-04:
> > > > 本地编译？建议make clean 云编译是通过的 [image] [image]
> > > 
> > > 
> > > 就是云编译，每次都最新环境 [image]
> > 
> > 
> > 你的项目地址在哪
> 
> https://github.com/binge8/op/actions/runs/6064360857/job/16452727799 我不确定你能不能看见日志，好像要授权？

可以看见 我修改了下再试试 我没用到py3所以没出这个问题

binge8, 2023-09-04:
> > > > > 本地编译？建议make clean 云编译是通过的 [image] [image]
> > > > 
> > > > 
> > > > 就是云编译，每次都最新环境 [image]
> > > 
> > > 
> > > 你的项目地址在哪
> > 
> > 
> > https://github.com/binge8/op/actions/runs/6064360857/job/16452727799 我不确定你能不能看见日志，好像要授权？
> 
> 可以看见 我修改了下再试试 我没用到py3所以没出这个问题

好的，小白鼠现在开始测试

binge8, 2023-09-04:
> > > > > 本地编译？建议make clean 云编译是通过的 [image] [image]
> > > > 
> > > > 
> > > > 就是云编译，每次都最新环境 [image]
> > > 
> > > 
> > > 你的项目地址在哪
> > 
> > 
> > https://github.com/binge8/op/actions/runs/6064360857/job/16452727799 我不确定你能不能看见日志，好像要授权？
> 
> 可以看见 我修改了下再试试 我没用到py3所以没出这个问题

[image]
hs还是不行

WYC-2020, 2023-09-04:
有多少错误不是乱升级导致的，很多测都不测 直接就提，一合并各种翻车

closed 2023-09-06 by binge8