coolsnowwolf/lede #11496: aliyundrive-webdav编译不了

ChaingTsung, 2023-08-30:
### 日志如下：
```log
...
Downloaded signal-hook-tokio v0.3.1
  Downloaded psl-types v2.0.11
  Downloaded base64 v0.13.1
  Downloaded crossbeam-utils v0.8.16
  Downloaded crossbeam-channel v0.5.8
  Downloaded ipnet v2.8.0
  Downloaded cookie v0.16.2
  Downloaded byteorder v1.4.3
  Downloaded path-slash v0.2.1
  Downloaded checked_int_cast v1.0.0
  Downloaded serde v1.0.188
  Downloaded slab v0.4.9
  Downloaded pin-project-lite v0.2.13
  Downloaded parking v2.1.0
  Downloaded tower-service v0.3.2
  Downloaded time-core v0.1.1
  Downloaded dirs v5.0.1
  Downloaded hyper-rustls v0.24.1
  Downloaded anstyle-query v1.0.0
  Downloaded scopeguard v1.2.0
  Downloaded async-trait v0.1.73
  Downloaded memoffset v0.9.0
  Downloaded ppv-lite86 v0.2.17
  Downloaded reqwest-retry v0.2.2
  Downloaded dirs-sys v0.4.1
  Downloaded clap_lex v0.5.1
  Downloaded mio v0.8.8
  Downloaded http v0.2.9
  Downloaded triomphe v0.1.9
  Downloaded parking_lot_core v0.9.8
  Downloaded rand_core v0.6.4
  Downloaded quote v1.0.33
  Downloaded adler v1.0.2
  Downloaded proc-macro2 v1.0.66
  Downloaded indicatif v0.17.6
  Downloaded async-lock v2.8.0
  Downloaded tokio-macros v2.1.0
  Downloaded serde_urlencoded v0.7.1
  Downloaded parking_lot v0.12.1
  Downloaded cpufeatures v0.2.9
  Downloaded polling v2.8.0
  Downloaded semver v1.0.18
  Downloaded retry-policies v0.1.2
  Downloaded signal-hook v0.3.17
  Downloaded clap v4.4.1
  Downloaded futures-sink v0.3.28
  Downloaded cfg-if v1.0.0
  Downloaded strsim v0.10.0
  Downloaded signal-hook-mio v0.2.3
  Downloaded flate2 v1.0.27
  Downloaded spin v0.5.2
  Downloaded htmlescape v0.3.1
  Downloaded task-local-extensions v0.1.4
  Downloaded pin-project v1.1.3
  Downloaded colorchoice v1.0.0
  Downloaded atty v0.2.14
  Downloaded tokio-socks v0.5.1
  Downloaded qr2term v0.3.1
  Downloaded crypto-common v0.1.6
  Downloaded sct v0.7.0
  Downloaded qrcode v0.12.0
  Downloaded digest v0.10.7
  Downloaded tinyvec v1.6.0
  Downloaded tracing-log v0.1.3
  Downloaded openssl-sys v0.9.92
  Downloaded sha1 v0.10.5
  Downloaded scheduled-thread-pool v0.2.7
  Downloaded tinyvec_macros v0.1.1
  Downloaded moka v0.11.3
  Downloaded signal-hook-registry v1.4.1
  Downloaded iana-time-zone v0.1.57
  Downloaded smallvec v1.11.0
  Downloaded headers v0.3.8
  Downloaded concurrent-queue v2.2.0
  Downloaded tracing-attributes v0.1.26
  Downloaded autocfg v1.1.0
  Downloaded either v1.9.0
  Downloaded tokio-rustls v0.24.1
  Downloaded rustls-pemfile v1.0.3
  Downloaded cookie_store v0.16.2
  Downloaded ryu v1.0.15
  Downloaded time-macros v0.2.14
  Downloaded hashbrown v0.12.3
  Downloaded deranged v0.3.8
  Downloaded rand v0.8.5
  Downloaded bitflags v1.3.2
  Downloaded self_update v0.37.0
  Downloaded tls-listener v0.7.0
  Downloaded block-buffer v0.10.4
  Downloaded anstyle-parse v0.2.1
  Downloaded hashbrown v0.14.0
  Downloaded anstream v0.5.0
  Downloaded tokio-native-tls v0.3.1
  Downloaded publicsuffix v2.2.3
  Downloaded crc32fast v1.3.2
  Downloaded console v0.15.7
  Downloaded clap_derive v4.4.0
  Downloaded socket2 v0.4.9
  Downloaded async-io v1.13.0
  Downloaded cc v1.0.83
  Downloaded futures-util v0.3.28
  Downloaded dashmap v5.5.3
  Downloaded mime v0.3.17
  Downloaded pin-project-internal v1.1.3
  Downloaded tracing-core v0.1.31
  Downloaded sharded-slab v0.1.4
  Downloaded regex-automata v0.1.10
  Downloaded bitflags v2.4.0
  Downloaded anyhow v1.0.75
  Downloaded portable-atomic v1.4.3
  Downloaded h2 v0.3.21
  Downloaded tar v0.4.40
  Downloaded serde_derive v1.0.188
  Downloaded socket2 v0.5.3
  Downloaded bytes v1.4.0
  Downloaded tracing v0.1.37
  Downloaded crossbeam-epoch v0.9.15
  Downloaded reqwest v0.11.20
  Downloaded dav-server v0.5.7
  Downloaded hyper v0.14.27
  Downloaded quick-xml v0.23.1
  Downloaded regex v1.9.4
  Downloaded idna v0.4.0
  Downloaded idna v0.3.0
  Downloaded idna v0.2.3
  Downloaded openssl v0.10.57
  Downloaded crossterm v0.25.0
  Downloaded async-compression v0.4.2
  Downloaded serde_json v1.0.105
  Downloaded rustls-webpki v0.101.4
  Downloaded regex-syntax v0.6.29
  Downloaded clap_builder v4.4.1
  Downloaded regex-syntax v0.7.5
  Downloaded tracing-subscriber v0.3.17
  Downloaded aho-corasick v1.0.5
  Downloaded libc v0.2.147
  Downloaded tagptr v0.2.0
  Downloaded chrono v0.4.27
  Downloaded rustls v0.21.7
  Downloaded regex-automata v0.3.7
  Downloaded rustix v0.37.23
  Downloaded tempfile v3.8.0
  Downloaded rustix v0.38.10
  Downloaded syn v2.0.29
  Downloaded tokio v1.32.0
  Downloaded time v0.3.28
  Downloaded unicode-normalization v0.1.22
  Downloaded linux-raw-sys v0.4.5
  Downloaded urlencoding v2.1.3
  Downloaded xattr v1.0.1
  Downloaded uuid v1.4.1
  Downloaded terminal_size v0.2.6
  Downloaded xmltree v0.10.3
  Downloaded webpki-roots v0.25.2
  Downloaded untrusted v0.7.1
  Downloaded try-lock v0.2.4
  Downloaded linux-raw-sys v0.3.8
  Downloaded encoding_rs v0.8.33
  Downloaded openssl-src v111.27.0+1.1.1v
  Downloaded thiserror-impl v1.0.47
  Downloaded ring v0.16.20
  Downloaded want v0.3.1
  Downloaded unicode-bidi v0.3.13
  Downloaded unicode-ident v1.0.11
  Downloaded vcpkg v0.2.15
  Downloaded version_check v0.9.4
  Downloaded utf8parse v0.2.1
  Downloaded zip v0.6.6
  Downloaded waker-fn v1.1.0
  Downloaded xml-rs v0.8.16
  Downloaded unicode-width v0.1.10
  Downloaded url v2.4.1
  Downloaded unicase v2.7.0
  Downloaded typenum v1.16.0
  Downloaded tokio-util v0.7.8
  Downloaded thread_local v1.1.7
  Downloaded thiserror v1.0.47
  Downloaded 230 crates (28.0 MB) in 7.33s (largest was `openssl-src` at 5.1 MB)
error: failed to compile `aliyundrive-webdav v2.3.2 (/home/yoki/src/lede/build_dir/target-aarch64_cortex-a53_musl/aliyundrive-webdav-2.3.2)`, intermediate artifacts can be found at `/home/yoki/src/lede/build_dir/target-aarch64_cortex-a53_musl/aliyundrive-webdav-2.3.2/target`

Caused by:
  package `clap_lex v0.5.1` cannot be built because it requires rustc 1.70.0 or newer, while the currently active rustc version is 1.68.1
  Either upgrade to rustc 1.70.0 or newer, or use
  cargo update -p clap_lex@0.5.1 --precise ver
  where `ver` is the latest version of `clap_lex` supporting rustc 1.68.1
make[3]: *** [Makefile:57: /home/yoki/src/lede/build_dir/target-aarch64_cortex-a53_musl/aliyundrive-webdav-2.3.2/.built] Error 101
make[3]: Leaving directory '/home/yoki/src/lede/feeds/packages/multimedia/aliyundrive-webdav'
time: package/feeds/packages/aliyundrive-webdav/compile#123.46#7.45#223.52
    ERROR: package/feeds/packages/aliyundrive-webdav failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/aliyundrive-webdav/compile] Error 1
make[2]: Leaving directory '/home/yoki/src/lede'
make[1]: *** [package/Makefile:110: /home/yoki/src/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yoki/src/lede'
make: *** [/home/yoki/src/lede/include/toplevel.mk:231：world] 错误 2
```

yanxin152133, 2023-08-31:
Caused by:
  package `clap_lex v0.5.1` cannot be built because it requires rustc 1.70.0 or newer, while the currently active rustc version is 1.68.1

yanxin152133, 2023-08-31:
https://github.com/coolsnowwolf/lede/issues/11493

ChaingTsung, 2023-08-31:
> [第 11493 章](https://github.com/coolsnowwolf/lede/issues/11493)

ok

closed 2023-08-31 by ChaingTsung