coolsnowwolf/lede #11506: 通过action编译，如果在编译前修改默认wan口为eth3

Pacio, 2023-09-03:
diy-part1.sh里面应该加入哪句修改命令

ecrasy, 2023-09-04:
1.  获取board id
cat /tmp/sysinfo/board_name

2. 保存记住你的board id

3. 根据board id编写相应的设置
如果 board id 有效，不是长得像default string那种字符
编辑 target/linux/你的板子platform/etc/board.d/02_network
添加你的板子case到文件中
如果你的board id 无效，就是长得像default string那种字符
编辑 package/base-files/files/etc/board.d/99-default_network
将默认设置修改为你想要的样式

Pacio, 2023-09-04:
> 1. 获取board id
>    cat /tmp/sysinfo/board_name
> 2. 保存记住你的board id
> 3. 根据board id编写相应的设置
>    如果 board id 有效，不是长得像default string那种字符
>    编辑 target/linux/你的板子platform/etc/board.d/02_network [...]

3Q，我的id是default string~
那我插入以下命令可以吗？
sed -i 's/eth1/eth3/g' package/base-files/files/etc/board.d/99-default_network

closed 2023-09-11 by Pacio