coolsnowwolf/lede #11507: 是不是aliyundrive-webdav引起的编译失败？

yuzhe1129, 2023-09-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
试过单单勾SSR 没事，但是加上webdav 就失败，请问这个是什么原因啊？

### (2) 路由器型号和固件版本

jdcloud_re-sp-01b

### (3) 详细日志

echo "uci set luci.languages.zh_cn='简体中文 (Simplified Chinese)'; uci commit luci" > /home/c/lede/build_dir/target-mipsel_24kc_musl/luci-base/.pkgdir/luci-i18n-base-zh-cn/etc/uci-defaults/luci-i18n-base-zh-cn
install -d -m0755 /home/c/lede/build_dir/target-mipsel_24kc_musl/luci-base/.pkgdir/luci-i18n-base-zh-cn/usr/lib/lua/luci/i18n
po2lmo /home/c/lede/feeds/luci/modules/luci-base/po/zh-cn/base.po /home/c/lede/build_dir/target-mipsel_24kc_musl/luci-base/.pkgdir/luci-i18n-base-zh-cn/usr/lib/lua/luci/i18n/base.zh-cn.lmo;
touch /home/c/lede/build_dir/target-mipsel_24kc_musl/luci-base/.pkgdir/luci-i18n-base-zh-cn.installed
mkdir -p /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-mipsel_24kc_musl/luci-base/.pkgdir/luci-i18n-base-zh-cn/. /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-i18n-base-zh-cn_installed
echo "luci-i18n-base-zh-cn" >> /home/c/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-base.default.install
make[3]: Leaving directory '/home/c/lede/feeds/luci/modules/luci-base'
time: package/feeds/luci/luci-base/compile#2.83#1.72#6.81
make[3]: Entering directory '/home/c/lede/feeds/luci/modules/luci-mod-admin-full'
make[3]: Entering directory '/home/c/lede/feeds/luci/libs/luci-lib-ipkg'
make[3]: Entering directory '/home/c/lede/feeds/luci/modules/luci-newapi'
mkdir -p /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp                                                                                                 
SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-mipsel_24kc_musl/luci-lib-ipkg/.pkgdir/luci-lib-ipkg/. /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
mkdir -p /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-mipsel_24kc_musl/luci-newapi/.pkgdir/luci-newapi/. /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-lib-ipkg_installed
touch /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-newapi_installed
echo "luci-newapi" >> /home/c/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-newapi.default.install
echo "luci-lib-ipkg" >> /home/c/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-lib-ipkg.default.install
make[3]: Leaving directory '/home/c/lede/feeds/luci/modules/luci-newapi'
time: package/feeds/luci/luci-newapi/compile#0.26#0.10#0.55
make[3]: Leaving directory '/home/c/lede/feeds/luci/libs/luci-lib-ipkg'
time: package/feeds/luci/luci-lib-ipkg/compile#0.23#0.12#0.56
mkdir -p /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp                                                                                                 
SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-mipsel_24kc_musl/luci-mod-admin-full/.pkgdir/luci-mod-admin-full/. /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-mod-admin-full_installed
echo "luci-mod-admin-full" >> /home/c/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-mod-admin-full.default.install
make[3]: Leaving directory '/home/c/lede/feeds/luci/modules/luci-mod-admin-full'
time: package/feeds/luci/luci-mod-admin-full/compile#0.32#0.18#0.69
make[3]: Entering directory '/home/c/lede/feeds/luci/collections/luci'
mkdir -p /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp                                                                                                 
SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-mipsel_24kc_musl/luci/.pkgdir/luci/. /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci_installed
echo "luci" >> /home/c/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci.default.install
make[3]: Leaving directory '/home/c/lede/feeds/luci/collections/luci'
time: package/feeds/luci/luci/compile#0.24#0.11#0.34
make[3]: Entering directory '/home/c/lede/package/lean/default-settings'ltas                                                                                                
mkdir -p /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamps                                                                                                
SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-mipsel_24kc_musl/default-settings-3/.pkgdir/default-settings/. /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/c/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.default-settings_installed
echo "default-settings" >> /home/c/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/default-settings.default.install
make[3]: Leaving directory '/home/c/lede/package/lean/default-settings'
time: package/lean/default-settings/compile#0.14#0.05#0.20
error: failed to compile `aliyundrive-webdav v2.3.2 (/home/c/lede/build_dir/target-mipsel_24kc_musl/aliyundrive-webdav-2.3.2)`, intermediate artifacts can be found at `/home/c/lede/build_dir/target-mipsel_24kc_musl/aliyundrive-webdav-2.3.2/target`

Caused by:
  package `clap_builder v4.4.2` cannot be built because it requires rustc 1.70.0 or newer, while the currently active rustc version is 1.68.1
  Either upgrade to rustc 1.70.0 or newer, or use
  cargo update -p clap_builder@4.4.2 --precise ver
  where `ver` is the latest version of `clap_builder` supporting rustc 1.68.1
make[3]: *** [Makefile:57: /home/c/lede/build_dir/target-mipsel_24kc_musl/aliyundrive-webdav-2.3.2/.built] Error 101
make[3]: Leaving directory '/home/c/lede/feeds/packages/multimedia/aliyundrive-webdav'
time: package/feeds/packages/aliyundrive-webdav/compile#1.90#1.02#23.27
    ERROR: package/feeds/packages/aliyundrive-webdav failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/aliyundrive-webdav/compile] Error 1
make[2]: Leaving directory '/home/c/lede'
make[1]: *** [package/Makefile:110: /home/c/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/c/lede'
make: *** [/home/c/lede/include/toplevel.mk:231：world] 错误 2
c@ubuntu:~/lede$

binge8, 2023-09-03:
都不搜搜吗 https://github.com/coolsnowwolf/lede/issues/11493#issue-1870744629

asdfyounghuie, 2023-09-05:
error: failed to compile `aliyundrive-webdav v2.3.2 (/home/yh/lede/build_dir/target-aarch64_generic_musl/aliyundrive-webdav-2.3.2)`, intermediate artifacts can be found at `/home/yh/lede/build_dir/target-aarch64_generic_musl/aliyundrive-webdav-2.3.2/target`

Caused by:
  package `clap_builder v4.4.2` cannot be built because it requires rustc 1.70.0 or newer, while the currently active rustc version is 1.68.1
  Either upgrade to rustc 1.70.0 or newer, or use
  cargo update -p clap_builder@4.4.2 --precise ver
  where `ver` is the latest version of `clap_builder` supporting rustc 1.68.1
make[3]: *** [Makefile:57: /home/yh/lede/build_dir/target-aarch64_generic_musl/aliyundrive-webdav-2.3.2/.built] Error 101
make[3]: Leaving directory '/home/yh/lede/feeds/packages/multimedia/aliyundrive-webdav'
time: package/feeds/packages/aliyundrive-webdav/compile#194.25#10.26#211.40
    ERROR: package/feeds/packages/aliyundrive-webdav failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/aliyundrive-webdav/compile] Error 1
make[2]: Leaving directory '/home/yh/lede'
make[1]: *** [package/Makefile:110: /home/yh/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yh/lede'
make: *** [/home/yh/lede/include/toplevel.mk:231: world] Error 2

gngpp, 2023-09-12:
这应该去问Rustacean

gngpp, 2023-09-12:
rustup update一下就好了，版本过低了

closed 2023-11-13 by aiamadeus