coolsnowwolf/lede #11511: 8125b驱动有问题,编译出来的op无法识别

hanlin0323, 2023-09-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
最新8125b驱动有问题,编译出来的op包识别不了8125b网卡
### (1) 具体问题

A：最新8125b驱动有问题,编译出来的op包识别不了8125b网卡

### (2) 路由器型号和固件版本

A：esxi6.7直通8125b网卡到op

### (3) 详细日志

A：

hcym, 2023-09-04:
我的正常

WYC-2020, 2023-09-04:
贴日志给他们看，要不你就用老版本

hanlin0323, 2023-09-04:
[   49.877568] vmxnet3 0000:1b:00.0 eth0: intr type 3, mode 0, 3 vectors allocated
[   49.878540] vmxnet3 0000:1b:00.0 eth0: NIC Link is Up 10000 Mbps
[   49.879057] 8021q: adding VLAN 0 to HW filter on device eth0
[   49.879580] br-lan: port 1(eth0) entered blocking state
[   49.879898] br-lan: port 1(eth0) entered disabled state
[   49.880399] device eth0 entered promiscuous mode
[   49.908117] br-lan: port 1(eth0) entered blocking state
[   49.908484] br-lan: port 1(eth0) entered forwarding state
[   49.908940] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   49.917597] eth1: 0xffffc90000540000, c4:83:4f:16:0b:c9, IRQ 18
[   49.978445] br-lan: port 2(eth1) entered blocking state
[   49.978784] br-lan: port 2(eth1) entered disabled state
[   49.979193] device eth1 entered promiscuous mode
[   49.983428] eth2: 0xffffc90000580000, c4:83:4f:16:0b:ca, IRQ 19
[   50.029224] br-lan: port 3(eth2) entered blocking state
[   50.029556] br-lan: port 3(eth2) entered disabled state
[   50.029950] device eth2 entered promiscuous mode
[   50.034548] eth3: 0xffffc900005a0000, c4:83:4f:16:0b:cb, IRQ 16
[   53.058413] r8125: eth2: link up
[   53.058671] br-lan: port 3(eth2) entered blocking state
[   53.058995] br-lan: port 3(eth2) entered forwarding state
[   53.188370] r8125: eth3: link up
旧版本的,能正常识别

hanlin0323, 2023-09-04:
> 我的正常

你的驱动是大雕最新的9.0.11版本吗

hcym, 2023-09-04:
对，就是lean的，其实去年11月的版本就好了，后面一直正常，就是前几天版本不能用官方2305编译，现在也好了

hanlin0323, 2023-09-04:
有其他人试过了 大雕昨天更新的8125b驱动 版本 这个驱动有问题 之前是没事的

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2023年09月04日 21:32 |
| 收件人 | coolsnowwolf/lede ***@***.***> |
| 抄送人 | 黑土 ***@***.***>,
Author ***@***.***> |
| 主题 | Re: [coolsnowwolf/lede] 8125b驱动有问题,编译出来的op无法识别 (Issue #11511) |

对，就是lean的，其实去年11月的版本就好了，后面一直正常，就是前几天版本不能用官方2305编译，现在也好了

hcym, 2023-09-05:
昨天下午编译的，pn41板载8125
不管了，目前为止正常

closed 2023-09-14 by coolsnowwolf