coolsnowwolf/lede #11512: 本地编译aliyundrive-webdav插件报错求解答

hztbhtbh, 2023-09-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 有类似的issue    #11493    本人新手看不太懂，求解答

## 2. 详细叙述

### (1) 具体问题

A：本地编译aliyundrive-webdav插件报错

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：看截图
[image]

hztbhtbh, 2023-09-05:
我翻译了报错，就是不知道怎么操作才行，
造成原因：
   无法构建包“clap v4.4.2”，因为它需要 rustc 1.70.0 或更高版本，而当前有效的 rustc 版本是 1.68.1
   升级到 rustc 1.70.0 或更高版本，或者使用
   货物更新-p clap@4.4.2 --precise 版本
   其中“ver”是支持 rustc 1.68.1 的最新版本“clap”

newcardriver, 2023-09-05:
你是在服务器环境编译的吗？可以尝试升级一下rust的版本，
rustup update 
[image]

hztbhtbh, 2023-09-05:
> 你是在服务器环境编译的吗？可以尝试升级一下rust的版本， rustup update [image]

我是在本地虚拟机安装ubuntu18系统上编译的

lxg2001, 2023-09-05:
> > 你是在服务器环境编译的吗？可以尝试升级一下rust的版本， rustup update [image]
> 
> 我是本地win10台式电脑安装虚拟机vmware上用ubuntu18编译环境编译的
> 不知道输入什么命令能升级？

看这里：https://stackoverflow.com/questions/37928591/is-there-a-command-to-update-cargo-to-the-latest-official-release

r1172464137, 2023-09-05:
```
PKG_BUILD_DEPENDS:=rust/host

RUST_PKG_FEATURES:=rustls-tls atomic64 native-tls native-tls-vendored

include $(INCLUDE_DIR)/package.mk
include ../../lang/rust/rust-package.mk
```
有没有一种可能这个软件包的编译跟本机自带的rust没有关系

hztbhtbh, 2023-09-05:
谢谢你们还是看不懂，不知道有没有简单的方法，比如怎么替换阿里云webdav之前2.2.1的版本呢？我就是输入了下面两行命令后就编译失败了
./scripts/feeds update -a       
./scripts/feeds install -a
请各位大神帮忙解答，感激不尽，谢谢

hztbhtbh, 2023-09-07:
谢谢各位大神的指导，我还是想编译的时候给编译进去，这样需要复位的时候就不需要重新的去安装这个插件了。

lrgcx, 2023-09-08:
不知道有什么办法完美解决，我一直在找。

hztbhtbh, 2023-09-09:
只能等大佬修复了，没有其它办法。

Jason6111, 2023-09-11:
./scripts/feeds install -a后用这个  
```
curl -o ./feeds/packages/lang/rust/rust-values.mk https://raw.githubusercontent.com/Jason6111/packages/patch-1/lang/rust/rust-values.mk
curl -o ./feeds/packages/lang/rust/Makefile https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/rust/Makefile
curl -o ./feeds/packages/lang/rust/patches/0001-Update-xz2-and-use-it-static.patch https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/rust/0001-Update-xz2-and-use-it-static.patch
```  
早解决了给出方案了 但有人不成功就取消合并了

lrgcx, 2023-09-12:
编译成功，不过怎么还是2.2.1？
Generating index for package ./vsftpd-alt_3.0.5-1_x86_64.ipk
Generating index for package ./wireguard-tools_1.0.20210914-2_x86_64.ipk
Generating index for package ./wol_0.7.1-3_x86_64.ipk
Generating index for package ./zlib_1.2.13-9_x86_64.ipk
Generating index for package ./luci-app-accesscontrol_1-11_all.ipk
Generating index for package ./luci-app-adbyby-plus_2.0-76_all.ipk
Generating index for package ./luci-app-aliyundrive-webdav_2.2.1-1_all.ipk
Generating index for package ./luci-app-arpbind_1-3_all.ipk
Generating index for package ./luci-app-autoreboot_1.0-8_all.ipk
Generating index for package ./luci-app-ddns_2.4.9-9_all.ipk
Generating index for package ./luci-app-diskman_v0.2.11-3_all.ipk
Generating index for package ./luci-app-filetransfer_1-2_all.ipk

Jason6111, 2023-09-12:
> 编译成功，不过怎么还是2.2.1？ Generating index for package ./vsftpd-alt_3.0.5-1_x86_64.ipk Generating index for package ./wireguard-tools_1.0.20210914-2_x86_64.ipk Generating index for package [...]

之前是不是用过替换ali成2.2.1版本啊 改回来 或删了重来

lrgcx, 2023-09-12:
没有，今天重装的ubuntu20，按照你步骤重新开始的。

Jason6111, 2023-09-12:
> 没有，今天重装的ubuntu20，按照你步骤重新开始的。

cat ./feeds/packages/blob/master/multimedia/aliyundrive-webdav/Makefile 看下是不是2.3.2

lrgcx, 2023-09-12:
我只是用大雕的源码编译测试，只勾选了webdav，其他都没动。没有blob这个目录。

lrgcx, 2023-09-12:
[image]

gngpp, 2023-09-12:
rustup update

lrgcx, 2023-09-12:
编译时我看到已经是1.72.0版本了

lrgcx, 2023-09-12:
[image]

lrgcx, 2023-09-12:
> > 没有，今天重装的ubuntu20，按照你步骤重新开始的。
> 
> cat ./feeds/packages/blob/master/multimedia/aliyundrive-webdav/Makefile 看下是不是2.3.2

刚才把我想要安装插件全部加上多线程编译了一次，在刚编译完的洁面那里还是显示2.2.1，但是刻盘装在软路由上显示的是2.3.2。总算解决问题了，谢谢🙏
[image]

Jason6111, 2023-09-12:
> > > 没有，今天重装的ubuntu20，按照你步骤重新开始的。
> > 
> > 
> > cat ./feeds/packages/blob/master/multimedia/aliyundrive-webdav/Makefile 看下是不是2.3.2
> 
> 刚才把我想要安装插件全部加上多线程编译了一次，在刚编译完的洁面那里还是显示2.2.1，但是刻盘装在软路由上显示的是2.3.2。总算解决问题了，谢谢🙏 [image]

嗯 那就好

Wilson-coder, 2023-09-13:
> ./scripts/feeds install -a后用这个
>
> ```
> curl -o ./feeds/packages/lang/rust/rust-values.mk https://raw.githubusercontent.com/Jason6111/packages/patch-1/lang/rust/rust-values.mk
> curl -o [...]

修改了这几个文件之后，已经顺利把Rust升级到了1.72，完整的编译也跑完了。

Jason6111, 2023-09-13:
> 修改了这几个文件之后，已经顺利把Rust升级到了1.72，完整的编译也跑完了。

能帮到你 很高兴～

binge8, 2023-09-13:
看样子本地编译都成功了，有云编译的尝试过吗

WYC-2020, 2023-09-13:
别挣扎，我这边亲测gcc13尿崩
[image]
至于 其他gcc 不清楚，看这个错误应该是什么gcc兼容问题
然后解决方法就是不用下载的llvm用编译的，然后就可以了，亲测，至于其他gcc你们可以试试，升级的话也简单直接把op那边拿过来覆盖就行了

kid424, 2023-09-13:
> ./scripts/feeds install -a后用这个
>
> ```
> curl -o ./feeds/packages/lang/rust/rust-values.mk https://raw.githubusercontent.com/Jason6111/packages/patch-1/lang/rust/rust-values.mk
> curl -o [...]

请问这个方案是必须要完全重新编译？我在之前编译过的目录里执行，编译rust的时候报错了
make[3]: Entering directory '/data/lede/feeds/packages/lang/rust'
. /data/lede/include/shell.sh; /data/lede/staging_dir/host/bin/libdeflate-gzip -dc /data/lede/dl/rustc-1.72.0-src.tar.gz | tar -C /data/lede/build_dir/target-x86_64_musl/host/rust-x86_64-unknown-linux-musl/rustc-1.72.0-src/.. -xf -
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
Makefile:103: recipe for target '/data/lede/build_dir/target-x86_64_musl/host/rust-x86_64-unknown-linux-musl/rustc-1.72.0-src/.prepareda02dbafe26f0c3f162da86553e501a2c_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/data/lede/build_dir/target-x86_64_musl/host/rust-x86_64-unknown-linux-musl/rustc-1.72.0-src/.prepareda02dbafe26f0c3f162da86553e501a2c_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory '/data/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#5.44#15.30#65.01
    ERROR: package/feeds/packages/rust [host] failed to build.
package/Makefile:114: recipe for target 'package/feeds/packages/rust/host/compile' failed
make[2]: *** [package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/data/lede'
package/Makefile:110: recipe for target '/data/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/data/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/data/lede'
/data/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

试了下单独执行/data/lede/staging_dir/host/bin/libdeflate-gzip -dc /data/lede/dl/rustc-1.72.0-src.tar.gz | tar -C /data/lede/build_dir/target-x86_64_musl/host/rust-x86_64-unknown-linux-musl/rustc-1.72.0-src/.. -xf -也是报一样的错，但是把libdeflate-gzip替换成gzip就可以解压成功。不知道这个错有没有什么办法手动改一下

hztbhtbh, 2023-09-13:
我是编译报错后，执行一下下面命令
`make distclean      #清除所有的编译文件和已经下载的软件包`
然后执行楼上的命令，在重新编译就没有报错了
`./scripts/feeds update -a           #更新源
./scripts/feeds install -a             #安装源
curl -o ./feeds/packages/lang/rust/rust-values.mk https://raw.githubusercontent.com/Jason6111/packages/patch-1/lang/rust/rust-values.mk
curl -o ./feeds/packages/lang/rust/Makefile https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/rust/Makefile
curl -o ./feeds/packages/lang/rust/patches/0001-Update-xz2-and-use-it-static.patch https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/rust/0001-Update-xz2-and-use-it-static.patch`
编译出来后的aliyundrive-webdav就是2.3.2版本了，在此感谢楼上兄弟帮助。

kid424, 2023-09-13:
> 我是编译报错后，执行一下下面命令 `make distclean #清除所有的编译文件和已经下载的软件包` 然后执行楼上的命令，在重新编译就没有报错了 `./scripts/feeds update -a` `./scripts/feeds install -a ` `curl -o ./feeds/packages/lang/rust/rust-values.mk https://raw.githubusercontent.com/Jason6111/packages/patch-1/lang/rust/rust-values.mk` `curl -o ./feeds/packages/lang/rust/Makefile https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/rust/Makefile` `curl -o ./feeds/packages/lang/rust/patches/0001-Update-xz2-and-use-it-static.patch https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/rust/0001-Update-xz2-and-use-it-static.patch`
> 
> 编译出来后的aliyundrive-webdav就是2.3.2版本了，在此感谢楼上兄弟帮助。

感谢回复，回头我clean下再试试

No06, 2023-09-16:
mips还不能编译，因为ring现在不支持 [https://github.com/briansmith/ring/pull/1277](url)

gngpp, 2023-09-16:
默认使用rustls，依赖ring，切换openssl features就行了

binge8, 2023-09-23:
[L大已修复](https://github.com/coolsnowwolf/packages/commit/d4ad9b75be0913abf675f9cad3444ec6fa7cb5a6)，但是增加了编译时间，插件多点6小时就不够用了，除了缩减插件，大哥大姐们还有别的好办法吗

gngpp, 2023-09-23:
> [L大已修复](https://github.com/coolsnowwolf/packages/commit/d4ad9b75be0913abf675f9cad3444ec6fa7cb5a6)，但是增加了编译时间，插件多点6小时就不够用了，除了缩减插件，大哥大姐们还有别的好办法吗

切回上游的Makefile就行了，使用预编译的二进制，rust编译很久的，根本吃不消，

binge8, 2023-09-24:
> > [L大已修复](https://github.com/coolsnowwolf/packages/commit/d4ad9b75be0913abf675f9cad3444ec6fa7cb5a6)，但是增加了编译时间，插件多点6小时就不够用了，除了缩减插件，大哥大姐们还有别的好办法吗
> 
> 切回上游的Makefile就行了，使用预编译的二进制，rust编译很久的，根本吃不消，

是啊，我看了云编译多出半小时左右

closed 2023-11-13 by aiamadeus