coolsnowwolf/lede #11521: MTK系列编译报错

VIKINGYFY, 2023-09-09:
这个[提交](https://github.com/coolsnowwolf/lede/commit/77abd0b9756786a019ad9c512849841b659d91f7)之后，编译报错

Error: ../dts/mt7986a-glinet-gl-mt6000.dts:284.1-6 Label or path wifi not found
FATAL ERROR: Syntax error parsing input tree
make[5]: *** [Makefile:37: /home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/image-mt7986a-glinet-gl-mt6000.dtb] Error 1
make[5]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/target/linux/mediatek/image'
make[4]: *** [Makefile:17: install] Error 2
make[4]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/target/linux/mediatek'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/target/linux'
time: target/linux/install#86.12#7.09#95.51
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt'
make[1]: *** [target/Makefile:24: /home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt'
make: *** [/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

cercky, 2023-09-09:
最近更新的这版报错。同样的情况。

13049554242, 2023-09-09:
我也是

closed 2023-09-09 by coolsnowwolf