coolsnowwolf/lede #11526: 升级openssl后pgyvpn编译失败

GreatMichaelLee, 2023-09-15:
ssl库版本还是依赖1.1的

touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.pgyvpn_installed
mkdir -p /home/builder/lede_x86/bin/targets/x86/64/packages /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/pgyvpn-3.1.0/ipkg-x86_64/pgyvpn/CONTROL /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/pgyvpn-3.1.0/ipkg-x86_64/pgyvpn/etc/init.d
install -d -m0755 /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/pgyvpn-3.1.0/ipkg-x86_64/pgyvpn/etc/config
install -d -m0755 /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/pgyvpn-3.1.0/ipkg-x86_64/pgyvpn/etc/hotplug.d/iface
install -d -m0755 /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/pgyvpn-3.1.0/ipkg-x86_64/pgyvpn/usr/sbin
install -d -m0755 /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/pgyvpn-3.1.0/ipkg-x86_64/pgyvpn/usr/share/pgyvpn
cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/pgyvpn-3.1.0/{pgyvpn_oraysl,pgyvpnsvr} /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/pgyvpn-3.1.0/ipkg-x86_64/pgyvpn/usr/sbin
install -m0755 ./files/etc/init.d/* /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/pgyvpn-3.1.0/ipkg-x86_64/pgyvpn/etc/init.d
install -m0600 ./files/etc/config/* /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/pgyvpn-3.1.0/ipkg-x86_64/pgyvpn/etc/config
install -m0755 ./files/etc/hotplug.d/iface/* /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/pgyvpn-3.1.0/ipkg-x86_64/pgyvpn/etc/hotplug.d/iface
install -m0755 ./files/usr/share/pgyvpn/* /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/pgyvpn-3.1.0/ipkg-x86_64/pgyvpn/usr/share/pgyvpn
find /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/pgyvpn-3.1.0/ipkg-x86_64/pgyvpn -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package pgyvpn is missing dependencies for the following libraries:
libcrypto.so.1.1
libssl.so.1.1
make[3]: *** [Makefile:124: /home/builder/lede_x86/bin/targets/x86/64/packages/pgyvpn_3.1.0-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/pgyvpn'
time: package/feeds/packages/pgyvpn/compile#0.34#0.14#2.15
    ERROR: package/feeds/packages/pgyvpn failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/pgyvpn/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231: world] Error 2

aiamadeus, 2023-09-15:
暂时不会支持 3.0，你可以选择 revert openssl 的更改。

closed 2023-10-01 by coolsnowwolf

sKyissKy, 2024-10-28:
问题又复现了，这个怎么搞，有大佬能在issue留个手工解决办法？