coolsnowwolf/lede #11529: 当编译目标的CPU架构为 MIPS32R2 时，openssl 3.0.10 编译报错失败

blickers, 2023-09-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
更新后的 openssl 3.0.10 编译失败，系统是 64位 ubuntu 20.04，编译的目标是优酷路由宝 YK-L1，该路由器 SoC 是联发科 mt7620。看编译输出信息大概是 crypto/aes/aes-mips.S 报很多非法操作符，附上编译输出，请分析并修正。
### (1) 具体问题
大体报错信息是这样的：
`mipsel-openwrt-linux-musl-gcc  -Icrypto -I. -Iinclude -Iproviders/common/include -Iproviders/implementations/include  -DAES_ASM -DOPENSSL_BN_ASM_MONT -DSHA1_ASM -DSHA256_ASM -fPIC -pthread -mabi=32 -Wa,--noexecstack -Wall -O3 -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/mnt/openwrt/yk-l1/lean-20230915/build_dir/target-mipsel_24kc_musl/openssl-3.0.10=openssl-3.0.10 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -O3 -DPIC -fpic -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/mnt/openwrt/yk-l1/lean-20230915/build_dir/target-mipsel_24kc_musl/openssl-3.0.10=openssl-3.0.10 -Wformat -Werror=format-security -fstack-protector -O3 -fpic -znow -zrelro -DOPENSSL_USE_NODELETE -DOPENSSL_PIC -DOPENSSLDIR="\"/etc/ssl\"" -DENGINESDIR="\"/usr/lib/engines-3\"" -DMODULESDIR="\"/usr/lib/ossl-modules\"" -DOPENSSL_BUILDING_OPENSSL -DNDEBUG -D_FORTIFY_SOURCE=1 -DPIC -DOPENSSL_PREFER_CHACHA_OVER_GCM -I/mnt/openwrt/yk-l1/lean-20230915/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/mnt/openwrt/yk-l1/lean-20230915/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/mnt/openwrt/yk-l1/lean-20230915/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include -c -o crypto/aes/libcrypto-lib-aes-mips.o crypto/aes/aes-mips.S
crypto/aes/aes-mips.S: Assembler messages:
crypto/aes/aes-mips.S:13: Error: invalid operands `lw $12,0($6)'
crypto/aes/aes-mips.S:14: Error: invalid operands `lw $13,4($6)'
crypto/aes/aes-mips.S:15: Error: invalid operands `lw $14,8($6)'
crypto/aes/aes-mips.S:16: Error: invalid operands `lw $15,12($6)'
crypto/aes/aes-mips.S:17: Error: invalid operands `lw $30,240($6)'
crypto/aes/aes-mips.S:20: Error: invalid operands `xor $8,$12'
crypto/aes/aes-mips.S:21: Error: invalid operands `xor $9,$13'
crypto/aes/aes-mips.S:22: Error: invalid operands `xor $10,$14'
crypto/aes/aes-mips.S:23: Error: invalid operands `xor $11,$15'
crypto/aes/aes-mips.S:25: Error: invalid operands `subu $30,1'
crypto/aes/aes-mips.S:114: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ext $12,$9,8,8'
crypto/aes/aes-mips.S:116: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ext $13,$10,8,8'
crypto/aes/aes-mips.S:117: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ext $14,$11,8,8'`

### (2) 路由器型号和固件版本

优酷路由宝 YK-L1，SoC为联发科 MT7620

### (3) 详细日志
见附件
[buildlog.log](https://github.com/coolsnowwolf/lede/files/12621113/buildlog.log)

aiamadeus, 2023-09-16:
[image]
实测没有问题，你要用我改的这个 commit 7494eb16185a176de226f55e842cadf94f1c5a11

XXXstyle, 2023-09-16:
大佬，请问日志出现这样的提示，会有什么影响吗  “daemon.notice procd: /etc/rc.d/S13openssl: Skipping devcrypto: /usr/lib/engines-3/devcrypto.so not found.”

XXXstyle, 2023-09-16:
> [image] 实测没有问题，你要用我改的这个 commit [7494eb1](https://github.com/coolsnowwolf/lede/commit/7494eb16185a176de226f55e842cadf94f1c5a11)

大佬，请问日志出现这样的提示，会有什么影响吗 “daemon.notice procd: /etc/rc.d/S13openssl: Skipping devcrypto: /usr/lib/engines-3/devcrypto.so not found.”