coolsnowwolf/lede #11530: 5.10内核的代码P&W R619AC 128M无法启动

qpzr, 2023-09-15:
不能启动正常启动，WEB刷新半天都没反应，所以没有日志
BIN和UBI格式都试过了，翻车
从PB-BOOT再试两次也是一样翻车
无奈，只能翻5.4内核的固件刷回去正常上网，赶紧上来开个问题
请大神看看是什么个情况..........

coolsnowwolf, 2023-09-16:
TTL log ?

建议用 Mleaf 的 uboot ，带自动修复ubi

dengxin6, 2023-09-17:
之前发过issue，被关了

确实启动不了，5.4.252开始就不行。最新[qcom-ipq4019-r619ac.dtsi]还是不行，间隔几秒钟反复重启

coolsnowwolf, 2023-09-18:
没有 TTL log 就只能关闭了

coolsnowwolf, 2023-09-19:
可以提供 TTL 启动日志么？

qpzr, 2023-09-19:
> 可以提供 TTL 启动日志么？

没有编程器没有ttl线，
只能作死等了两个小时刚编译好最新源码的固件，直接刷。结果如图还是不行
[image]

[image]

光猫桥接的拨号主路由，又要跑楼下去拔线带回房间，进pb-boot刷回5.4的固件了

ZeroHoo, 2023-09-20:
opboot1.1.3 刷入5.10后路由器一直重启，只能刷回5.4的

WRMing1008, 2023-10-10:
我arch17也是一样，同时编译的几个不同插件的有些能正常刷进正常启动，有些不能启动，还有一个问题就WIFI不能满速，只能跑55%左右。之前编译5.4内核的正常

closed 2023-10-10 by aiamadeus

qpzr, 2023-10-15:
提前编译好固件刷进去，下楼坐等一死。再从pb-boot里刷成opboot1.1.2居然就正常了，一脸懵。21年2月21日潘多拉固件里自带的pb-boot竟然会翻车，想不明白了。

aiamadeus, 2023-10-16:
@qpzr 跟不死没关系，我前几天出院才修的

4thanks, 2023-10-22:
@aiamadeus 能帮忙看一下星际宝盒cm520的吗？也是ipq40xx的。云编译后得到openwrt-ipq40xx-generic-8dev_habanero-dvk-squashfs-sysupgrade.bin，无论是固件升级还是opboot刷入都失败，直接升级提升`不支持所上传的映像文件格式，请选择适合当前平台的通用映像文件。`，opboot刷入就一直没反应，手动重启还是原来的固件。

aiamadeus, 2023-10-22:
@4thanks 你就不能重新选 config 再编译。。。

4thanks, 2023-10-22:
> @4thanks 你就不能重新选 config 再编译。。。

不太懂，能帮忙指出哪里出来问题吗？
之前都是抄别人的GitHub action云编译，改一下型号编译后能正常升级刷入，现在用相同config编译出来的是`openwrt-ipq40xx-generic-8dev_habanero-dvk-squashfs-sysupgrade.bin`，以前编译出来的是`openwrt-ipq40xx-mobipromo_cm520-79f-squashfs-nand-sysupgrade.bin `，是带型号的`mobipromo_cm520-79f`，现在编译出来是`generic-8dev_habanero-dvk`，我想可能这也是问题。

qpzr, 2023-10-22:
> 不太懂，能帮忙指出哪里出来问题吗？ 之前都是抄别人的GitHub [...]

都是去抄作业，你去抄人家5.10内核成功的. config的时候，  难道没有发现跟以前5.4的时候多了generic这个？
以前是这样 CONFIG_TARGET_ipq40xx_DEVICE_
现在你改成这样 CONFIG_TARGET_ipq40xx_generic_DEVICE_
名字就不会错乱了  这不就行了么

4thanks, 2023-10-23:
> 都是去抄作业，你去抄人家5.10内核成功的. config的时候， 难道没有发现跟以前5.4的时候多了generic这个？ 以前是这样 CONFIG_TARGET_ipq40xx_DEVICE_ 现在你改成这样 CONFIG_TARGET_ipq40xx_generic_DEVICE_ 名字就不会错乱了 这不就行了么

哈哈哈 被自己蠢笑了，明明已经知道原因了，还在找问题。

commit 7deff887 (2023-10-10): ipq40xx: fixes k5.10 boot issue

Fixes: #11530