coolsnowwolf/lede #11534: 全新安装ubunt22.03编译报错

hzmax0416, 2023-09-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
全新安装ubunt22.03编译报错

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency samba4-libs for samba4-server
 * pkg_hash_fetch_best_installation_candidate: Packages for samba4-server found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for autosamba:
 * 	samba4-libs
 * opkg_install_cmd: Cannot install package autosamba.
 * check_conflicts_for: The following packages conflict with samba4-server:
 * check_conflicts_for: 	samba36-server * 
 * opkg_install_cmd: Cannot install package samba4-server.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/hzmax/lede'
make[1]: *** [package/Makefile:111: /home/hzmax/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/hzmax/lede'
make: *** [/home/hzmax/lede/include/toplevel.mk:231: world] Error 2

hcym, 2023-09-17:
少了什么吧，

XXXstyle, 2023-09-17:
去了samba36-server 再编译试试

hzmax0416, 2023-09-18:
> 去了samba36-server重新编译尝试

OK了 谢谢

coolsnowwolf, 2023-09-18:
rm -rf bin

WYC-2020, 2023-09-20:
明显的冲突而已别什么都选samba4 和36冲突的

closed 2023-09-22 by aiamadeus