coolsnowwolf/lede #11546: web管理后台无法打开

redjumper, 2023-09-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
*  [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

使用最新编译版本，在x86 lxc容器下，web管理后台无法打开

### (2) 路由器型号和固件版本

x86 lxc容器，OP是最新编译版本r6184-407fe3d68

### (3) 详细日志

系统日志如下

> daemon.err uhttpd[5343]: Failed to initialize ustream-ssl
daemon.info procd: Instance uhttpd::instance1 s in a crash loop 6 crashes, 0 seconds since last crash

Zane-E, 2023-09-25:
一样情况   回退到openssl: bump to 3.0.10前正常

smdx, 2023-09-27:
昨天编译的固件今天更新了遇到相同的问题。

leochien0102, 2023-10-01:
openssl回滚到v1.1.1v问题解决

paranoiagu, 2023-10-12:
> openssl回滚到v1.1.1v问题解决

请问怎么会滚 openssl ？

paranoiagu, 2023-10-12:
无需替换 openssl，修改 /etc/config/uhttpd 文件，注释掉 2 行 listen_https，即解决。

lionlinzq, 2023-12-09:
> 件，注释

你的方法奏效了，感谢

lionlinzq, 2023-12-09:
> 无需替换 openssl，修改 /etc/config/uhttpd 文件，注释掉 2 行 listen_https，即解决。

你的方法奏效了，感谢

leux0, 2024-01-02:
关于WEB管理后台无法打开的问题解决

```bash
错误显示：daemon.err uhttpd[5343]: Failed to initialize ustream-ssl

方法一：无需替换 openssl，修改 /etc/config/uhttpd 文件中注释掉2行 listen_https再重启即可解决
方法二：问题是由于LEDE将openssl升级到3.0导致的，回滚到v1.1.1可解决问题

# 删除自带的3.0，使用持续更新的istore的openssl来顶替
cd lede/package/libs/ && rm -rf openssl/
cp -r istoreos/package/libs/openssl/ openssl/

# 或者也可以通过git回退到openssl未升级3.0时
git log --pretty=oneline				# 查看历史提交的ID
7494eb16185a176de226f55e842cadf94f1c5a11 openssl: bump to 3.0.10
a2a0e5b9f6ce420398487ea29a85b0b0c49c9087 ipq-wifi: bump to latest git HEAD

git checkout a2a0e5b9f6ce420398487ea29a85b0b0c49c9087	# 回退到未升级3.0时

```

closed 2024-04-16 by redjumper