coolsnowwolf/lede #11550: 如何导入替换成自己config插件？

KridgeDsouza, 2023-09-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

yanxin152133, 2023-10-01:
比如这个frp
[image]

heiha10, 2023-10-28:
> 比如这个frp [image]
打开lede文件夹，直接复制git 那行代码就行了，然后 再运行编译流程那两行，之后在配置就行了

coolsnowwolf, 2026-09-05:
LEDE 使用仓库根目录的 `.config` 作为编译配置，不需要替换“config 插件”文件。可运行 `make menuconfig` 选择 x86/64 目标和软件包，或将已有配置复制为 `.config` 后执行 `make defconfig` 校正依赖；也可用 `make savedefconfig` 导出精简配置。自定义包应放在 `package/` 或独立 feed，再通过 `./scripts/feeds update/install` 引入。该 issue 没有具体错误，属于配置使用咨询，关闭。

closed 2026-09-05 by coolsnowwolf