coolsnowwolf/lede #11558: armv8报错

binge8, 2023-09-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：云编译armv8最新源码报错

### (2) 路由器型号和固件版本

A：armv8

### (3) 详细日志

A：看截图
[image]

breakings, 2023-09-30:
遇到相同报错

fengou, 2023-10-02:
google上找到解决办法了：https://www.spinics.net/lists/stable-commits/msg318335.html

201821143044, 2023-10-02:
> google上找到解决办法了：https://www.spinics.net/lists/stable-commits/msg318335.html

小白没看懂，请问具体怎么解决呢？

fengou, 2023-10-03:
去到你本地编译的路径下，比如我的是/home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/drivers/interconnect，找到一个叫core.c的文件打开，在里面这个位置添加一行代码“#include <linux/sched/mm.h>”，如下所示：

// SPDX-License-Identifier: GPL-2.0
/*
 * Interconnect framework core driver
 *
 * Copyright (c) 2017-2019, Linaro Ltd.
 * Author: Georgi Djakov <georgi.djakov@linaro.org>
 */

#include <linux/debugfs.h>
#include <linux/device.h>
#include <linux/idr.h>
#include <linux/init.h>
#include <linux/interconnect.h>
#include <linux/interconnect-provider.h>
#include <linux/list.h>
#include <linux/sched/mm.h>
#include <linux/module.h>
#include <linux/mutex.h>
#include <linux/slab.h>
#include <linux/of.h>
#include <linux/overflow.h>

hcym, 2023-10-03:
学习一下，谢谢

closed 2023-10-03 by coolsnowwolf