coolsnowwolf/lede #11559: 拉取最新源码编译报错

bi4tms, 2023-09-30:
拉取最新源码编译报错，提示curl组件无法下载，具体如下：
make[3]: Entering directory '/home/r2s.6.1/lede/feeds/packages/net/curl'
mkdir -p /home/r2s.6.1/lede/dl
SHELL= flock /home/r2s.6.1/lede/tmp/.curl-8.3.0.tar.xz.flock -c '  	/home/r2s.6.1/lede/scripts/download.pl "/home/r2s.6.1/lede/dl" "curl-8.3.0.tar.xz" "376D627767D6C4F05105AB6D497B0D9ABA7111770DD9D995225478209C37EA63" "" "https://github.com/curl/curl/releases/download/curl-8_3_0/" "https://dl.uxnr.de/mirror/curl/" "https://curl.askapache.com/download/" "https://curl.se/download/"    '
+ curl -f --connect-timeout 20 --retry 5 --location https://github.com/curl/curl/releases/download/curl-8_3_0/curl-8.3.0.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 2579k  100 2579k    0     0  1723k      0  0:00:01  0:00:01 --:--:-- 10.5M
Hash of the downloaded file does not match (file: 376d627767d6c4f05105ab6d497b0d9aba7111770dd9d995225478209c37ea63, requested: 376D627767D6C4F05105AB6D497B0D9ABA7111770DD9D995225478209C37EA63) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location https://dl.uxnr.de/mirror/curl/curl-8.3.0.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://curl.askapache.com/download/curl-8.3.0.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   252  100   252    0     0    322      0 --:--:-- --:--:-- --:--:--   322
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://curl.se/download/curl-8.3.0.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 2579k  100 2579k    0     0  3060k      0 --:--:-- --:--:-- --:--:-- 3056k
Hash of the downloaded file does not match (file: 376d627767d6c4f05105ab6d497b0d9aba7111770dd9d995225478209c37ea63, requested: 376D627767D6C4F05105AB6D497B0D9ABA7111770DD9D995225478209C37EA63) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/curl-8.3.0.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 2579k  100 2579k    0     0  3524k      0 --:--:-- --:--:-- --:--:-- 3519k
Hash of the downloaded file does not match (file: 376d627767d6c4f05105ab6d497b0d9aba7111770dd9d995225478209c37ea63, requested: 376D627767D6C4F05105AB6D497B0D9ABA7111770DD9D995225478209C37EA63) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.openwrt.org/curl-8.3.0.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 2579k  100 2579k    0     0   779k      0  0:00:03  0:00:03 --:--:--  779k
Hash of the downloaded file does not match (file: 376d627767d6c4f05105ab6d497b0d9aba7111770dd9d995225478209c37ea63, requested: 376D627767D6C4F05105AB6D497B0D9ABA7111770DD9D995225478209C37EA63) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/curl-8.3.0.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:185: /home/r2s.6.1/lede/dl/curl-8.3.0.tar.xz] Error 2
make[3]: Leaving directory '/home/r2s.6.1/lede/feeds/packages/net/curl'
time: package/feeds/packages/curl/compile#0.59#0.21#10.11
    ERROR: package/feeds/packages/curl failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/curl/compile] Error 1
make[2]: Leaving directory '/home/r2s.6.1/lede'
make[1]: *** [package/Makefile:110: /home/r2s.6.1/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/r2s.6.1/lede'
make: *** [/home/r2s.6.1/lede/include/toplevel.mk:231：world] 错误 2

Pacio, 2023-10-01:
我也是这样的报错

xiaojing110, 2023-10-01:
找到这个文件目录
feeds/package/net   修改哈希值大写字母为小写就可以了

gsblay, 2023-10-01:
[](url)

Ericj2011, 2023-10-01:
> 找到这个文件目录 feeds/package/net 修改哈希值大写字母为小写就可以了

可否具体说明怎样修改，字面上不明白

binge8, 2023-10-01:
> > 找到这个文件目录 feeds/package/net 修改哈希值大写字母为小写就可以了
> 
> 可否具体说明怎样修改，字面上不明白

已经修复了，正常编译就可以了

closed 2023-10-01 by bi4tms