coolsnowwolf/lede #11562: 360打包提示error

zpj7377327, 2023-10-02:
/反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：打包失败

### (2) 路由器型号和固件版本

A：360 T7

### (3) 详细日志

A：make[5]: *** [scripts/Makefile.build:289: /media/zpj/8926fa12-724d-4e84-a68a-0676d01db3ab/zpj/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2023-09-18-2afc7285/mac80211.o] Error 1
make[4]: *** [Makefile:1907: /media/zpj/8926fa12-724d-4e84-a68a-0676d01db3ab/zpj/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2023-09-18-2afc7285] Error 2
make[4]: Leaving directory '/media/zpj/8926fa12-724d-4e84-a68a-0676d01db3ab/zpj/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.133'
make[3]: *** [Makefile:585: /media/zpj/8926fa12-724d-4e84-a68a-0676d01db3ab/zpj/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2023-09-18-2afc7285/.built] Error 2
make[3]: Leaving directory '/media/zpj/8926fa12-724d-4e84-a68a-0676d01db3ab/zpj/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#0.96#0.20#1.38
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/media/zpj/8926fa12-724d-4e84-a68a-0676d01db3ab/zpj/lede'
make[1]: *** [package/Makefile:110: /media/zpj/8926fa12-724d-4e84-a68a-0676d01db3ab/zpj/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/media/zpj/8926fa12-724d-4e84-a68a-0676d01db3ab/zpj/lede'
make: *** [/media/zpj/8926fa12-724d-4e84-a68a-0676d01db3ab/zpj/lede/include/toplevel.mk:231：world] 错误 2

yanxin152133, 2023-10-02:
https://github.com/coolsnowwolf/lede/issues/8965

zpj7377327, 2023-10-02:
> https://github.com/coolsnowwolf/lede/issues/8965

好的 我试试

zpj7377327, 2023-10-02:
还是没解决@yanxin152133

yanxin152133, 2023-10-02:
我也不会，我自己编译的报错也是才解决好

closed 2023-10-08 by coolsnowwolf