coolsnowwolf/lede #11563: drivers/interconnect/core.c: In function 'icc_init':编译报错

fengou, 2023-10-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志
fengou@ubuntu20a:~/lede$ 
fengou@ubuntu20a:~/lede$ make V=s -j1
make[2]: Entering directory '/home/fengou/lede/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/fengou/lede/scripts/config'
make[1]: Entering directory '/home/fengou/lede'
make[2]: Entering directory '/home/fengou/lede'
make[3]: Entering directory '/home/fengou/lede/target/linux'
make[4]: Entering directory '/home/fengou/lede/target/linux/armvirt'
rm -f /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/localversion
/home/fengou/lede/scripts/kconfig.pl  + + /home/fengou/lede/target/linux/generic/config-5.15 /home/fengou/lede/target/linux/armvirt/config-5.15 /home/fengou/lede/target/linux/armvirt/64/config-5.15 > /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/fengou/lede/.config >> /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.target
/home/fengou/lede/scripts/package-metadata.pl kconfig /home/fengou/lede/tmp/.packageinfo /home/fengou/lede/.config 5.15 > /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.override
/home/fengou/lede/scripts/kconfig.pl 'm+' '+' /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.target /dev/null /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.override > /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.set
mv /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.set /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.old
grep -v INITRAMFS /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.old > /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.set
rm -rf /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/modules
cmp -s /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.set /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.prev || { cp /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.set /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config; cp /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.set /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.prev; }
export MAKEFLAGS= ; [ -d /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/user_headers ] || make -C /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133 KCFLAGS="-fmacro-prefix-map=/home/fengou/lede/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/fengou/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Sep 30 09:44:03 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/fengou/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.133 CC="aarch64-openwrt-linux-musl-gcc"  INSTALL_HDR_PATH=/home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/user_headers headers_install
grep '=[ym]' /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.config.set | LC_ALL=C sort | /home/fengou/lede/staging_dir/host/bin/mkhash md5 > /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.vermagic
touch /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.configured
rm -f /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/vmlinux /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/System.map
make -C /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133 KCFLAGS="-fmacro-prefix-map=/home/fengou/lede/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/fengou/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Sep 30 09:44:03 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/fengou/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.133 CC="aarch64-openwrt-linux-musl-gcc" olddefconfig
make[5]: Entering directory '/home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133'
net/sched/Kconfig:45: warning: menuconfig statement without prompt
#
# No change to .config
#
make[5]: Leaving directory '/home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133'
make -C /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133 KCFLAGS="-fmacro-prefix-map=/home/fengou/lede/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/fengou/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Sep 30 09:44:03 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/fengou/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.133 CC="aarch64-openwrt-linux-musl-gcc" all modules
make[5]: Entering directory '/home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  CC      drivers/interconnect/core.o
drivers/interconnect/core.c: In function 'icc_init':
drivers/interconnect/core.c:1148:9: error: implicit declaration of function 'fs_reclaim_acquire' [-Werror=implicit-function-declaration]
 1148 |         fs_reclaim_acquire(GFP_KERNEL);
      |         ^~~~~~~~~~~~~~~~~~
drivers/interconnect/core.c:1150:9: error: implicit declaration of function 'fs_reclaim_release' [-Werror=implicit-function-declaration]
 1150 |         fs_reclaim_release(GFP_KERNEL);
      |         ^~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[7]: *** [scripts/Makefile.build:289: drivers/interconnect/core.o] Error 1
make[6]: *** [scripts/Makefile.build:552: drivers/interconnect] Error 2
make[5]: *** [Makefile:1907: drivers] Error 2
make[5]: Leaving directory '/home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133'
make[4]: *** [Makefile:20: /home/fengou/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.133/.modules] Error 2
make[4]: Leaving directory '/home/fengou/lede/target/linux/armvirt'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/fengou/lede/target/linux'
time: target/linux/compile#3.65#0.77#4.46
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/fengou/lede'
make[1]: *** [target/Makefile:23: /home/fengou/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/fengou/lede'
make: *** [/home/fengou/lede/include/toplevel.mk:231: world] Error 2
fengou@ubuntu20a:~/lede$

binge8, 2023-10-02:
[https://github.com/coolsnowwolf/lede/issues/11558#issue-1920313827](url)

fengou, 2023-10-02:
问题已经有了

closed 2023-10-02 by fengou