coolsnowwolf/lede #11565: frp 编译失败

coldrook, 2023-10-02:
x86软路由
frp编译失败 qtls需要go版本1.21
日志如下：
make[3]: Entering directory '/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/feeds/packages/net/frp'
rm -f /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/frp-0.47.0/.built
touch /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/frp-0.47.0/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/tmp/go-build" GO_MOD_CACHE_DIR="/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/fatedier/frp" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/fatedier/frp/cmd/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/frp-0.47.0" GO_BUILD_DIR="/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/frp-0.47.0/.go_work/build" GO_BUILD_BIN_DIR="/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/frp-0.47.0/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/frp-0.47.0=frp-0.47.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/frp-0.47.0=frp-0.47.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro" GOPATH="/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/frp-0.47.0/.go_work/build" GOCACHE="/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/tmp/go-build" GOMODCACHE="/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env bash ../../lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1696067043' -linkmode external -extldflags '-L/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags "-s -w  -buildid '1696067043' -linkmode external -extldflags '-L/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
Finding targets

Building targets
github.com/quic-go/quic-go/internal/qtls
# github.com/quic-go/quic-go/internal/qtls
../../../../../dl/go-mod-cache/github.com/quic-go/quic-go@v0.32.0/internal/qtls/go121.go:5:13: cannot use "The version of quic-go you're using can't be built on Go 1.21 yet. For more details, please see https://github.com/quic-go/quic-go/wiki/quic-go-and-Go-versions." (untyped string constant "The version of quic-go you're using can't be built on Go 1.21 yet. F...) as int value in variable declaration

make[3]: *** [Makefile:67: /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/frp-0.47.0/.built] Error 1
make[3]: Leaving directory '/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/feeds/packages/net/frp'
time: package/feeds/packages/frp/compile#0.70#0.46#1.08
    ERROR: package/feeds/packages/frp failed to build.

yanxin152133, 2023-10-02:
[image]
那你说这个是什么意思

swxk521, 2023-10-02:
golang升级到1.21了，不支持的就都编译不了，等大佬搞定吧。
https://github.com/coolsnowwolf/packages/commit/f937504996886608d3803b63cc6288790bc6a19e

Beginner-Go, 2023-10-02:
Fixed.

authon, 2023-10-04:
> Fixed.

V2ray-plugin也有这个问题

AstralHope, 2023-10-04:
> > Fixed.
> 
> V2ray-plugin也有这个问题

Building targets
github.com/quic-go/quic-go/internal/qtls
# github.com/quic-go/quic-go/internal/qtls
/lede/lede/dl/go-mod-cache/github.com/quic-go/quic-go@v0.32.0/internal/qtls/go121.go:5:13: cannot use "The version of quic-go you're using can't be built on Go 1.21 yet. For more details, please see https://github.com/quic-go/quic-go/wiki/quic-go-and-Go-versions." (untyped string constant "The version of quic-go you're using can't be built on Go 1.21 yet. F...) as int value in variable declaration

make[3]: *** [Makefile:48: /lede/lede/build_dir/target-x86_64_musl/brook-20230606/.built] Error 1

这个问题普遍性有点广

closed 2023-10-05 by coldrook