coolsnowwolf/lede #11568: 编译失败，错误提示：ERROR: Feature 'openssl-linked' was enabled, but the pre-condition '!features.securetransport && !features.schannel && libs.openssl' failed.

linkinuul, 2023-10-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败，错误提示：ERROR: Feature 'openssl-linked' was enabled, but the pre-condition '!features.securetransport && !features.schannel && libs.openssl' failed.

### (2) 路由器型号和固件版本

A：newwifi D2  R23.10.1

### (3) 详细日志

A：Configure summary:

Building on: linux-g++ (x86_64, CPU features: mmx sse sse2)
Building for: devices/linux-generic-g++ (mips, CPU features: <none>)
Target compiler: gcc 8.4.0
Configuration: cross_compile enable_new_dtags largefile ltcg shared shared release c++11 c++14 c++17 c++1z no-gui reduce_exports stl no-widgets
Build options:
  Mode ................................... release
  Optimize release build for size ........ no
  Building shared libraries .............. yes
  Using C standard ....................... C11
  Using C++ standard ..................... C++17
  Using ccache ........................... no
  Using new DTAGS ........................ yes
  Relocatable ............................ yes
  Using precompiled headers .............. no
  Using LTCG ............................. yes
  Target compiler supports:
    DSP .................................. no
    DSPr2 ................................ no
  Build parts ............................ libs
Qt modules and options:
  Qt Concurrent .......................... no
  Qt D-Bus ............................... no
  Qt D-Bus directly linked to libdbus .... no
  Qt Gui ................................. no
  Qt Network ............................. yes
  Qt Sql ................................. yes
  Qt Testlib ............................. no
  Qt Widgets ............................. no
  Qt Xml ................................. yes
Support enabled for:
  Using pkg-config ....................... yes
  udev ................................... no
  Using system zlib ...................... yes
  Zstandard support ...................... no
Qt Core:
  DoubleConversion ....................... yes
    Using system DoubleConversion ........ no
  GLib ................................... no
  iconv .................................. no
  ICU .................................... no
  Built-in copy of the MIME database ..... yes
  Tracing backend ........................ <none>
  Logging backends:
    journald ............................. no
    syslog ............................... no
    slog2 ................................ no
  PCRE2 .................................. yes
    Using system PCRE2 ................... yes
Qt Network:
[... 135 lines cut ...]
make[4]: Leaving directory '/home/linkin/openwrt/lede/build_dir/target-mipsel_24kc_musl/luci-lib-nixio/src'
find /home/linkin/openwrt/lede/build_dir/target-mipsel_24kc_musl/luci-lib-nixio/ipkg-mipsel_24kc/luci-lib-nixio -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="mipsel-openwrt-linux-musl-"  NO_RENAME=1 ; NM="mipsel-openwrt-linux-musl-nm" STRIP="/home/linkin/openwrt/lede/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/linkin/openwrt/lede/scripts/strip-kmod.sh" PATCHELF="/home/linkin/openwrt/lede/staging_dir/host/bin/patchelf" /home/linkin/openwrt/lede/scripts/rstrip.sh /home/linkin/openwrt/lede/build_dir/target-mipsel_24kc_musl/luci-lib-nixio/ipkg-mipsel_24kc/luci-lib-nixio
rstrip.sh: /home/linkin/openwrt/lede/build_dir/target-mipsel_24kc_musl/luci-lib-nixio/ipkg-mipsel_24kc/luci-lib-nixio/usr/lib/lua/nixio.so: shared object
(cd /home/linkin/openwrt/lede/build_dir/target-mipsel_24kc_musl/luci-lib-nixio/ipkg-mipsel_24kc/luci-lib-nixio/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/linkin/openwrt/lede/bin/packages/mipsel_24kc/luci
/home/linkin/openwrt/lede/staging_dir/host/bin/fakeroot /home/linkin/openwrt/lede/staging_dir/host/bin/bash /home/linkin/openwrt/lede/scripts/ipkg-build -m "" /home/linkin/openwrt/lede/build_dir/target-mipsel_24kc_musl/luci-lib-nixio/ipkg-mipsel_24kc/luci-lib-nixio /home/linkin/openwrt/lede/bin/packages/mipsel_24kc/luci
Packaged contents of /home/linkin/openwrt/lede/build_dir/target-mipsel_24kc_musl/luci-lib-nixio/ipkg-mipsel_24kc/luci-lib-nixio into /home/linkin/openwrt/lede/bin/packages/mipsel_24kc/luci/luci-lib-nixio_git-23.269.07510-663e299-1_mipsel_24kc.ipk
echo "luci-lib-nixio" >> /home/linkin/openwrt/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-lib-nixio.default.install
make[3]: Leaving directory '/home/linkin/openwrt/lede/feeds/luci/libs/luci-lib-nixio'
time: package/feeds/luci/luci-lib-nixio/compile#2.93#1.64#4.44
mkdir -p /home/linkin/openwrt/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/linkin/openwrt/lede/tmp/.root-copy.flock -c 'cp -fpR /home/linkin/openwrt/lede/build_dir/target-mipsel_24kc_musl/luci-app-firewall/.pkgdir/luci-app-firewall/. /home/linkin/openwrt/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/linkin/openwrt/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-app-firewall_installed
echo "luci-app-firewall" >> /home/linkin/openwrt/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-app-firewall.default.install
mkdir -p /home/linkin/openwrt/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/linkin/openwrt/lede/tmp/.root-copy.flock -c 'cp -fpR /home/linkin/openwrt/lede/build_dir/target-mipsel_24kc_musl/luci-app-firewall/.pkgdir/luci-i18n-firewall-zh-cn/. /home/linkin/openwrt/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/linkin/openwrt/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-i18n-firewall-zh-cn_installed
echo "luci-i18n-firewall-zh-cn" >> /home/linkin/openwrt/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-app-firewall.default.install
make[3]: Leaving directory '/home/linkin/openwrt/lede/feeds/luci/applications/luci-app-firewall'
time: package/feeds/luci/luci-app-firewall/compile#1.51#2.74#4.04
make[2]: Leaving directory '/home/linkin/openwrt/lede'
make[1]: *** [package/Makefile:110: /home/linkin/openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/linkin/openwrt/lede'
make: *** [/home/linkin/openwrt/lede/include/toplevel.mk:231：world] 错误 2

alterem, 2023-10-08:
同

suizhenmin, 2023-10-09:
同样的编译错误

Rhystar, 2023-12-03:
同样的错误

QZQ-QSQ, 2024-01-19:
我也是这样。

alterem, 2024-01-22:
> 我也是这样。

删掉项目后，重新拉代码编译

gforce1206, 2024-10-03:
发现是qbittorrent的锅，要make clean后取消它的编译

coolsnowwolf, 2026-09-05:
已核对当前源码。原日志是在 qtbase 强制 `-openssl-linked` 时找不到 OpenSSL；现行 `feeds/packages/libs/qtbase/Makefile` 已让 `qt5-network` 显式依赖 `+libopenssl`，而 qBittorrent 也显式依赖 `+libopenssl` 与 `+qt5-network`，因此正常依赖解析会把 OpenSSL 放入 staging，避免该配置前置条件失败。原始版本为 2023 年的 Qt 5.15.8/旧依赖树，后续评论也确认通过取消 qBittorrent 或 clean 后重新编译解决。当前 master 无需额外补丁，关闭为 completed。

closed 2026-09-05 by coolsnowwolf