coolsnowwolf/lede #11571: zstd编译错误

coldrook, 2023-10-05:
x86软路由 6.1.52内核

passwallpackages更新后v2ray和book的qtls的go版本问题修复后重新拉取编译，出现zstd编译错误

2023-10-05T09:16:17.9319292Z make[3]: Entering directory '/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/feeds/packages/utils/zstd'
2023-10-05T09:16:18.1157330Z rm -f /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/.configured_*
2023-10-05T09:16:18.1188224Z rm -f /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/target-x86_64_musl/stamp/.zstd_installed
2023-10-05T09:16:18.1240152Z /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/bin/sed -e "s|@CC@|'gcc',|" -e "s|@CXX@|'g++',|" -e "s|@PKGCONFIG@|/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/bin/pkg-config|" -e "s|@CMAKE@|/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/bin/cmake|" -e "s|@PYTHON@|/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/bin/python3|" -e "s|@CFLAGS@|'-O2', '-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/include', '-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/hostpkg/include', '-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/target-x86_64_musl/host/include', '-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/include', '-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/hostpkg/include', '-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/target-x86_64_musl/host/include',|" -e "s|@CXXFLAGS@|'-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/include', '-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/hostpkg/include', '-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/target-x86_64_musl/host/include',|" -e "s|@LDFLAGS@|'-L/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/lib', '-L/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/hostpkg/lib', '-L/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/target-x86_64_musl/host/lib',|" -e "s|@PREFIX@||" < /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/lib/meson/openwrt-native.txt.in > /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/openwrt-native.txt
2023-10-05T09:16:18.1279405Z /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/bin/sed -e "s|@CC@|'x86_64-openwrt-linux-musl-gcc',|" -e "s|@CXX@|'x86_64-openwrt-linux-musl-g++',|" -e "s|@AR@|x86_64-openwrt-linux-musl-gcc-ar|" -e "s|@STRIP@|x86_64-openwrt-linux-musl-strip|" -e "s|@NM@|x86_64-openwrt-linux-musl-gcc-nm|" -e "s|@PKGCONFIG@|/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/bin/pkg-config|" -e "s|@CMAKE@|/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/bin/cmake|" -e "s|@PYTHON@|/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/bin/python3|" -e "s|@CFLAGS@|'-pipe', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-fmacro-prefix-map=/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5=zstd-1.5.5', '-Wformat', '-Werror=format-security', '-fstack-protector', '-D_FORTIFY_SOURCE=1', '-Wl,-z,now', '-Wl,-z,relro', '-O3', '-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include', '-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify', '-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include',|" -e "s|@CXXFLAGS@|'-pipe', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-fmacro-prefix-map=/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5=zstd-1.5.5', '-Wformat', '-Werror=format-security', '-fstack-protector', '-D_FORTIFY_SOURCE=1', '-Wl,-z,now', '-Wl,-z,relro', '-O3', '-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include', '-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify', '-I/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include',|" -e "s|@LDFLAGS@|'-L/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib', '-L/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib', '-znow', '-zrelro',|" -e "s|@ARCH@|"x86_64"|" -e "s|@CPU@|"generic"|" -e "s|@ENDIAN@|little|" < /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/lib/meson/openwrt-cross.txt.in > /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/openwrt-cross.txt
2023-10-05T09:16:18.1311542Z /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/bin/python3 /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/bin/meson.py  --buildtype plain --native-file /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/openwrt-native.txt --cross-file /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/openwrt-cross.txt -Dlegacy_level=7 -Ddebug_level=0 -Dbacktrace=false -Dstatic_runtime=false -Dbin_programs=true -Dbin_tests=false -Dbin_contrib=false -Dmulti_thread=enabled -Dzlib=disabled -Dlzma=disabled -Dlz4=disabled -Db_lto=true /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/build/meson/openwrt-build /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/build/meson/openwrt-build/..
2023-10-05T09:16:19.0470716Z The Meson build system
2023-10-05T09:16:19.0471388Z Version: 0.61.5
2023-10-05T09:16:19.0472950Z Source dir: /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/build/meson
2023-10-05T09:16:19.0473981Z Build dir: /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/build/meson/openwrt-build
2023-10-05T09:16:19.0474742Z Build type: cross build
2023-10-05T09:16:19.0476747Z Program GetZstdLibraryVersion.py found: YES (/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/bin/python3 /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/build/meson/GetZstdLibraryVersion.py)
2023-10-05T09:16:19.0477694Z 
2023-10-05T09:16:19.0478590Z ../../../../build_dir/target-x86_64_musl/zstd-1.5.5/build/meson/meson.build:11:0: ERROR: Value "false" (of type "string") for combo option "Display a stack backtrace when execution generates a runtime exception" is not one of the choices. Possible choices are (as string): "enabled", "disabled", "auto".
2023-10-05T09:16:19.0479183Z 
2023-10-05T09:16:19.0479936Z A full log can be found at /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/build/meson/openwrt-build/meson-logs/meson-log.txt
2023-10-05T09:16:19.0912858Z make[3]: *** [Makefile:103: /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
2023-10-05T09:16:19.0913850Z make[3]: Leaving directory '/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/feeds/packages/utils/zstd'
2023-10-05T09:16:19.0914875Z time: package/feeds/packages/zstd/compile#1.08#0.08#1.16
2023-10-05T09:16:19.0922814Z     ERROR: package/feeds/packages/zstd failed to build.

binge8, 2023-10-05:
在等待合并了 https://github.com/coolsnowwolf/packages/pull/720#issuecomment-1748722683

coldrook, 2023-10-05:
看comment6小时前合并了，但4小时前自动重新拉取编译依然出错，还望大佬看看，是否还存在问题

2023-10-05T21:56:10.7822199Z /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/bin/python3 /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/host/bin/meson.py  --buildtype plain --native-file /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/openwrt-native.txt --cross-file /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/openwrt-cross.txt -Dlegacy_level=7 -Ddebug_level=0 -Dzlib=disabled \  /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/build/meson/openwrt-build /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/build/meson/openwrt-build/..
2023-10-05T21:56:11.6770150Z usage: meson [-h]
2023-10-05T21:56:11.6770900Z              {setup,configure,dist,install,introspect,init,test,wrap,subprojects,help,rewrite,compile,devenv}
2023-10-05T21:56:11.6771529Z              ...
2023-10-05T21:56:11.6772745Z meson: error: unrecognized arguments: /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/build/meson/openwrt-build/..
2023-10-05T21:56:11.7200589Z make[3]: *** [Makefile:97: /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/build_dir/target-x86_64_musl/zstd-1.5.5/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 2
2023-10-05T21:56:11.7201443Z make[3]: Leaving directory '/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/feeds/packages/utils/zstd'
2023-10-05T21:56:11.7206017Z time: package/feeds/packages/zstd/compile#1.07#0.07#1.15
2023-10-05T21:56:11.7212048Z     ERROR: package/feeds/packages/zstd failed to build.
2023-10-05T21:56:11.7216626Z make[2]: *** [package/Makefile:116: package/feeds/packages/zstd/compile] Error 1
2023-10-05T21:56:11.7224069Z make[2]: Leaving directory '/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt'
2023-10-05T21:56:11.7233523Z make[1]: *** [package/Makefile:110: /home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2023-10-05T21:56:11.7238428Z make[1]: Leaving directory '/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt'
2023-10-05T21:56:11.7250343Z make: *** [/home/runner/work/haiibo-opauto-actions/haiibo-opauto-actions/openwrt/include/toplevel.mk:231: world] Error 2
2023-10-05T21:56:11.7402999Z ##[error]Process completed with exit code 2.

allok1985, 2023-10-06:
make[3] -C feeds/packages/libs/glib2 host-compile
 make[3] -C feeds/packages/utils/zstd compile
    ERROR: package/feeds/packages/zstd failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1
刚刚云编译还是报错

coldrook, 2023-10-06:
感谢，编译成功

closed 2023-10-06 by coldrook