coolsnowwolf/lede #11576: mt76：基于5.4内核编译报错，其他内核正常

ywt114, 2023-10-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败，错误提示：ERROR: package/kernel/mt76 failed to build.

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：2023-10-06T22:32:31.7406214Z make[4]: Entering directory '/home/runner/work/kegin/kegin/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.257'
2023-10-06T22:32:32.6165841Z   CC [M]  /home/runner/work/kegin/kegin/openwrt/build_dir/target-x86_64_musl/linux-x86_64/mt76-2022-12-22-5b509e80/mt7921/mac.o
2023-10-06T22:32:33.7588787Z   CC [M]  /home/runner/work/kegin/kegin/openwrt/build_dir/target-x86_64_musl/linux-x86_64/mt76-2022-12-22-5b509e80/mt7921/mcu.o
2023-10-06T22:32:34.3523455Z /home/runner/work/kegin/kegin/openwrt/build_dir/target-x86_64_musl/linux-x86_64/mt76-2022-12-22-5b509e80/mt7921/mcu.c:16:52: error: expected ')' before 'bool'
2023-10-06T22:32:34.3524291Z    16 | module_param_named(disable_clc, mt7921_disable_clc, bool, 0644);
2023-10-06T22:32:34.3524907Z       |                                                    ^~~~~
2023-10-06T22:32:34.3525238Z       |                                                    )
2023-10-06T22:32:34.3525879Z /home/runner/work/kegin/kegin/openwrt/build_dir/target-x86_64_musl/linux-x86_64/mt76-2022-12-22-5b509e80/mt7921/mcu.c:17:30: error: expected ')' before string constant
2023-10-06T22:32:34.3526319Z    17 | MODULE_PARM_DESC(disable_clc, "disable CLC support");
2023-10-06T22:32:34.3527122Z       |                              ^~~~~~~~~~~~~~~~~~~~~~
2023-10-06T22:32:34.3527367Z       |                              )
2023-10-06T22:32:34.4261814Z make[6]: *** [scripts/Makefile.build:262: /home/runner/work/kegin/kegin/openwrt/build_dir/target-x86_64_musl/linux-x86_64/mt76-2022-12-22-5b509e80/mt7921/mcu.o] Error 1
2023-10-06T22:32:34.4262995Z make[5]: *** [scripts/Makefile.build:497: /home/runner/work/kegin/kegin/openwrt/build_dir/target-x86_64_musl/linux-x86_64/mt76-2022-12-22-5b509e80/mt7921] Error 2
2023-10-06T22:32:34.4266162Z make[4]: *** [Makefile:1750: /home/runner/work/kegin/kegin/openwrt/build_dir/target-x86_64_musl/linux-x86_64/mt76-2022-12-22-5b509e80] Error 2
2023-10-06T22:32:34.4267176Z make[4]: Leaving directory '/home/runner/work/kegin/kegin/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.257'
2023-10-06T22:32:34.4274217Z make[3]: *** [Makefile:585: /home/runner/work/kegin/kegin/openwrt/build_dir/target-x86_64_musl/linux-x86_64/mt76-2022-12-22-5b509e80/.built] Error 2
2023-10-06T22:32:34.4275049Z make[3]: Leaving directory '/home/runner/work/kegin/kegin/openwrt/package/kernel/mt76'
2023-10-06T22:32:34.4275949Z time: package/kernel/mt76/compile#2.51#0.60#3.11
2023-10-06T22:32:34.4285199Z     ERROR: package/kernel/mt76 failed to build.
2023-10-06T22:32:34.4289559Z make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
2023-10-06T22:32:34.4300443Z make[2]: Leaving directory '/home/runner/work/kegin/kegin/openwrt'
2023-10-06T22:32:34.4306619Z make[1]: *** [package/Makefile:110: /home/runner/work/kegin/kegin/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2023-10-06T22:32:34.4320463Z make[1]: Leaving directory '/home/runner/work/kegin/kegin/openwrt'
2023-10-06T22:32:34.4327425Z make: *** [/home/runner/work/kegin/kegin/openwrt/include/toplevel.mk:231: world] Error 2
2023-10-06T22:32:34.4435655Z ##[error]Process completed with exit code 2.

qqsir-dev, 2023-10-07:
6.1也报错
make[2] -C package/kernel/linux compile
    ERROR: package/kernel/linux failed to build.
make: *** [/home/runner/work/AutoLede/AutoLede/workspace/openwrt/include/toplevel.mk:231: package/compile] Error 1
make -r package/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make[2]: Entering directory '/home/runner/work/AutoLede/AutoLede/workspace/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/runner/work/AutoLede/AutoLede/workspace/openwrt/scripts/config'
make[1]: Entering directory '/home/runner/work/AutoLede/AutoLede/workspace/openwrt'
make[2]: Entering directory '/home/runner/work/AutoLede/AutoLede/workspace/openwrt/package/libs/libjson-c'
make[2]: Leaving directory '/home/runner/work/AutoLede/AutoLede/workspace/openwrt/package/libs/libjson-c'
time: package/libs/libjson-c/host-compile#0.21#0.08#0.28
make[2]: Entering directory '/home/runner/work/AutoLede/AutoLede/workspace/openwrt/package/libs/libubox'
make[2]: Leaving directory '/home/runner/work/AutoLede/AutoLede/workspace/openwrt/package/libs/libubox'
time: package/libs/libubox/host-compile#0.21#0.09#0.30
make[2]: Entering directory '/home/runner/work/AutoLede/AutoLede/workspace/openwrt/package/system/opkg'
make[2]: Leaving directory '/home/runner/work/AutoLede/AutoLede/workspace/openwrt/package/system/opkg'
time: package/system/opkg/host-compile#0.38#0.12#0.47
make[2]: Entering directory '/home/runner/work/AutoLede/AutoLede/workspace/openwrt/package/libs/toolchain'
touch /home/runner/work/AutoLede/AutoLede/workspace/openwrt/build_dir/target-x86_64_musl/toolchain/.prepared_212b4d59d5382e47fae6657b24946e32_6664517399ebbbc92a37c5bb081b5c53_check
mkdir -p /home/runner/work/AutoLede/AutoLede/workspace/openwrt/build_dir/target-x86_64_musl/toolchain
touch /home/runner/work/AutoLede/AutoLede/workspace/openwrt/build_dir/target-x86_64_musl/toolchain/.prepared_212b4d59d5382e47fae6657b24946e32_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/runner/work/AutoLede/AutoLede/workspace/openwrt/build_dir/target-x86_64_musl/toolchain/.configured_*
rm -f /home/runner/work/AutoLede/AutoLede/workspace/openwrt/staging_dir/target-x86_64_musl/stamp/.toolchain_installed
(cd /home/runner/work/AutoLede/AutoLede/workspace/openwrt/build_dir/target-x86_64_musl/toolchain/./; if [ -x ./configure ]; then find /home/runner/work/AutoLede/AutoLede/workspace/openwrt/build_dir/target-x86_64_musl/toolchain/ -name config.guess | xargs -r chmod u+w; find /home/runner/work/AutoLede/AutoLede/workspace/openwrt/build_dir/target-x86_64_musl/toolchain/ -name config.guess | xargs -r -n1 cp --remove-destination /home/runner/work/AutoLede/AutoLede/workspace/openwrt/scripts/config.guess; find /home/runner/work/AutoLede/AutoLede/workspace/openwrt/build_dir/target-x86_64_musl/toolchain/ -name config.sub | xargs -r chmod u+w; find /home/runner/work/AutoLede/AutoLede/workspace/openwrt/build_dir/target-x86_64_musl/toolchain/ -name config.sub | xargs -r -n1 cp --remove-destination /home/runner/work/AutoLede/AutoLede/workspace/openwrt/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/AutoLede/AutoLede/workspace/openwrt/build_dir/target-x86_64_musl/toolchain=toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/AutoLede/AutoLede/workspace/openwrt/build_dir/target-x86_64_musl/toolchain=toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/AutoLede/AutoLede/workspace/openwrt/build_dir/target-x86_64_musl/toolchain=toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/runner/work/AutoLede/AutoLede/workspace/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/runner/work/AutoLede/AutoLede/workspace/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/runner/work/AutoLede/AutoLede/workspace/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/home/runner/work/AutoLede/AutoLede/workspace/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/runner/work/AutoLede/AutoLede/workspace/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /home/runner/work/AutoLede/AutoLede/workspace/openwrt/build_dir/target-x86_64_musl/toolchain/.configured_68b329da9893e34099c7d8ad5cb9c940
[... cut ...]
make[2]: Leaving directory '/home/runner/work/AutoLede/AutoLede/workspace/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#4.36#0.51#4.87
    ERROR: package/kernel/linux failed to build.
make[1]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[1]: Leaving directory '/home/runner/work/AutoLede/AutoLede/workspace/openwrt'
make: *** [/home/runner/work/AutoLede/AutoLede/workspace/openwrt/include/toplevel.mk:231: package/compile] Error 2
Error: Process completed with exit code 2.

ywt114, 2023-10-07:
> 6.1也报错 make[2] -C package/kernel/linux compile ERROR: package/kernel/linux failed to build. make: *** [/home/runner/work/AutoLede/AutoLede/workspace/openwrt/include/toplevel.mk:231: package/compile] [...]

你的和我的不是同一个问题

ywt114, 2023-10-07:
我的问题出自于https://github.com/openwrt/mt76/tree/master/mt7921上周的更新，猜测是关于mcu.c文件的更新造成的
[image]

closed 2023-10-08 by coolsnowwolf

ywt114, 2023-10-11:
还没有解决 是啥原因呢