coolsnowwolf/lede #11582: 更新完lede，367T7编译错误

ItownTech, 2023-10-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译报错

### (2) 路由器型号和固件版本

A：360T7.     R23.10.1

### (3) 详细日志

A：
Applying /home/pi/lede/target/linux/mediatek/patches-5.15/846-v5.18-i2c-mediatek-Add-i2c-compatible-for-Mediatek-MT8168.patch using plaintext: 
patching file drivers/i2c/busses/i2c-mt65xx.c
Hunk #1 succeeded at 442 with fuzz 1 (offset 53 lines).
Hunk #2 FAILED at 461.
1 out of 2 hunks FAILED -- saving rejects to file drivers/i2c/busses/i2c-mt65xx.c.rej
Patch failed!  Please fix /home/pi/lede/target/linux/mediatek/patches-5.15/846-v5.18-i2c-mediatek-Add-i2c-compatible-for-Mediatek-MT8168.patch!
make[4]: *** [Makefile:27: /home/pi/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.133/.prepared_feba8ead21cc894d0f3418b6de495040] Error 1
make[4]: Leaving directory '/home/pi/lede/target/linux/mediatek'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/pi/lede/target/linux'
time: target/linux/compile#6.09#3.02#7.46
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/pi/lede'
make[1]: *** [target/Makefile:23: /home/pi/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/pi/lede'
make: *** [/home/pi/lede/include/toplevel.mk:231: world] Error 2

aiamadeus, 2023-10-12:
[image]
。。。浪费我时间

closed 2023-10-12 by aiamadeus

ItownTech, 2023-10-18:
> +1

我反复测试了好几次，要重新拉库才行，直接更新就会报这个错。不知道是哪里出现的问题