coolsnowwolf/lede #11585: k2p ykl1 编译出错

mitsukileung, 2023-10-11:
Building targets
runtime/cgo
# runtime/cgo
gcc_mipsx.S: Assembler messages:
gcc_mipsx.S:27: Error: invalid operands `sw $18,12($29)'
gcc_mipsx.S:28: Error: invalid operands `sw $19,16($29)'
gcc_mipsx.S:29: Error: invalid operands `sw $20,20($29)'
gcc_mipsx.S:30: Error: invalid operands `sw $21,24($29)'
gcc_mipsx.S:31: Error: invalid operands `sw $22,28($29)'
gcc_mipsx.S:32: Error: invalid operands `sw $23,32($29)'
gcc_mipsx.S:33: Error: invalid operands `sw $30,36($29)'
gcc_mipsx.S:46: Error: invalid operands `jalr $20'
gcc_mipsx.S:50: Error: invalid operands `lw $18,12($29)'
gcc_mipsx.S:51: Error: invalid operands `lw $19,16($29)'
gcc_mipsx.S:52: Error: invalid operands `lw $20,20($29)'
gcc_mipsx.S:53: Error: invalid operands `lw $21,24($29)'
gcc_mipsx.S:54: Error: invalid operands `lw $22,28($29)'
gcc_mipsx.S:55: Error: invalid operands `lw $23,32($29)'
gcc_mipsx.S:56: Error: invalid operands `lw $30,36($29)'
gcc_mipsx.S:65: Error: invalid operands `lw $31,0($29)'

make[3]: *** [Makefile:67: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.51.3/.built] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/frp'
time: package/feeds/packages/frp/compile#0.81#0.27#0.81
    ERROR: package/feeds/packages/frp failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/frp/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

X86编译没有这问题

yanxin152133, 2023-10-11:
frp换个老版本的试试

mitsukileung, 2023-10-11:
> frp换个老版本的试试

好像和FRP版本没有多大关系，9月16日编译ykl1的时候也可以用 0.5.13 编译成功

[image]

[image]

yanxin152133, 2023-10-11:
> > frp换个老版本的试试
> 
> 好像和FRP版本没有多大关系，9月16日编译ykl1的时候也可以用 0.5.13 编译成功
> 
> 
> [image]
> 
> [image]
> 
> 
我之前用的0.37

yanxin152133, 2023-10-11:
> > frp换个老版本的试试
> 
> 好像和FRP版本没有多大关系，9月16日编译ykl1的时候也可以用 0.5.13 编译成功
> 
> 
> [image]
> 
> [image]
> 
> 

我前段时间也碰到0.47版本的别人编译不了我能编译的情况

mitsukileung, 2023-10-11:
> > > frp换个老版本的试试
> > 
> > 
> > 好像和FRP版本没有多大关系，9月16日编译ykl1的时候也可以用 0.5.13 编译成功
> > [image]
> > [image]
> 
> 我前段时间也碰到0.47版本的别人编译不了我能编译的情况

配置文件什么的都没有改过，刚刚又试了下5.10内核，还是一样的问题

yanxin152133, 2023-10-13:
> > > > frp换个老版本的试试
> > > 
> > > 
> > > 好像和FRP版本没有多大关系，9月16日编译ykl1的时候也可以用 0.5.13 编译成功
> > > [image]
> > > [image]
> > 
> > 
> > 我前段时间也碰到0.47版本的别人编译不了我能编译的情况
> 
> 配置文件什么的都没有改过，刚刚又试了下5.10内核，还是一样的问题

试了下0.44版本可以正常编译。
[image]
[image]

mitsukileung, 2023-10-13:
> > > > > frp换个老版本的试试
> > > > 
> > > > 
> > > > 好像和FRP版本没有多大关系，9月16日编译ykl1的时候也可以用 0.5.13 编译成功
> > > > [image]
> > > > [image]
> > > 
> > > 
> > > 我前段时间也碰到0.47版本的别人编译不了我能编译的情况
> > 
> > 
> > 配置文件什么的都没有改过，刚刚又试了下5.10内核，还是一样的问题
> 
> 试了下0.44版本可以正常编译。 [image] [image]

X86 编译一直都是正常呀，就是MTK编译出错

liwenjie119, 2023-10-21:
[image]
同样的问题，已经有好一段时间了，之前以为是frp的问题，但是去看frp官方mips有release，说明源码应该没问题，不知道是不是gcc冲突

luzheng123, 2023-10-29:
我也遇到完全一样的问题，找不到原因，期待有大佬

liwenjie119, 2023-11-06:
仍然卡在这里，不清楚具体原因

mitsukileung, 2023-11-07:
只有等大佬们了

closed 2023-11-26 by aiamadeus