coolsnowwolf/lede #11586: 蒲公英vpn在openwrt的库中貌似已被删除

TritiumXs, 2023-10-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在make download的时候，下载蒲公英的源码包时出现404 Not Found的问题。而后经与https://mirror2.openwrt.org/sources/比对，发现库中并没有对应的源码包。

### (2) 路由器型号和固件版本

A：Thunder OneCloud

### (3) 详细日志

A：

SHELL= flock /home/tritium/lede/tmp/.pgyvpn_oraysl.flock -c '   /home/tritium/lede/scripts/download.pl "/home/tritium/lede/dl" "pgyvpn_oraysl" "skip" "" "https://mirrors.oray.com/orayos/packages/pgyvpn/arm_cortex-a5_vfpv4-musl-unknown/3.1.0/bin"    '
+ wget --tries=5 --timeout=20 --output-document=- https://mirrors.oray.com/orayos/packages/pgyvpn/arm_cortex-a5_vfpv4-musl-unknown/3.1.0/bin/pgyvpn_oraysl
--2023-10-11 10:03:32--  https://mirrors.oray.com/orayos/packages/pgyvpn/arm_cortex-a5_vfpv4-musl-unknown/3.1.0/bin/pgyvpn_oraysl
Resolving mirrors.oray.com (mirrors.oray.com)... 198.18.1.18
Connecting to mirrors.oray.com (mirrors.oray.com)|198.18.1.18|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-10-11 10:03:32 ERROR 404: Not Found.

Download failed.
+ wget --tries=5 --timeout=20 --output-document=- https://sources.cdn.openwrt.org/pgyvpn_oraysl
--2023-10-11 10:03:32--  https://sources.cdn.openwrt.org/pgyvpn_oraysl
Resolving sources.cdn.openwrt.org (sources.cdn.openwrt.org)... 198.18.0.24
Connecting to sources.cdn.openwrt.org (sources.cdn.openwrt.org)|198.18.0.24|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-10-11 10:03:32 ERROR 404: Not Found.

Download failed.
+ wget --tries=5 --timeout=20 --output-document=- https://sources.openwrt.org/pgyvpn_oraysl
--2023-10-11 10:03:32--  https://sources.openwrt.org/pgyvpn_oraysl
Resolving sources.openwrt.org (sources.openwrt.org)... 198.18.0.41
Connecting to sources.openwrt.org (sources.openwrt.org)|198.18.0.41|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-10-11 10:03:34 ERROR 404: Not Found.

Download failed.
+ wget --tries=5 --timeout=20 --output-document=- https://mirror2.openwrt.org/sources/pgyvpn_oraysl
--2023-10-11 10:03:34--  https://mirror2.openwrt.org/sources/pgyvpn_oraysl
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 198.18.0.42
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|198.18.0.42|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-10-11 10:03:35 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
make[2]: *** [Makefile:73: /home/tritium/lede/dl/pgyvpn_oraysl] Error 2
make[2]: Leaving directory '/home/tritium/lede/feeds/packages/net/pgyvpn'
time: package/feeds/packages/pgyvpn/download#0.11#0.03#3.06
    ERROR: package/feeds/packages/pgyvpn failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/pgyvpn/download] Error 1
make[1]: Leaving directory '/home/tritium/lede'
make[1]: Entering directory '/home/tritium/lede'
make[2]: Entering directory '/home/tritium/lede/target/linux'
make[3]: Entering directory '/home/tritium/lede/target/linux/meson'
make[3]: Nothing to be done for 'download'.
make[3]: Leaving directory '/home/tritium/lede/target/linux/meson'
make[2]: Leaving directory '/home/tritium/lede/target/linux'
time: target/linux/download#0.03#0.01#0.04
make[1]: Leaving directory '/home/tritium/lede'

qingtian110, 2023-10-11:
那是他们官方的下载链接炸了  https://mirrors.oray.com ，这跟 openwrt mirror 毫无关系。openwrt 官方不可能会收录这种奇奇怪怪的软件。

aiamadeus, 2023-10-12:
这是 openwrt 所没有的 target，你可以进 dl 下载
https://mirrors.oray.com/orayos/packages/pgyvpn/arm_cortex-a7-musl-unknown/3.1.0/bin/pgyvpn_oraysl
https://mirrors.oray.com/orayos/packages/pgyvpn/arm_cortex-a7-musl-unknown/3.1.0/bin/pgyvpnsvr

closed 2023-11-17 by TritiumXs