coolsnowwolf/lede #11587: daemon.err uhttpd[2042]: expr: syntax error

rongcenter, 2023-10-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
daemon.err uhttpd[2042]: expr: syntax error

### (2) 路由器型号和固件版本

X86 最新固件

### (3) 详细日志
[image]

Thu Oct 12 00:03:20 2023 daemon.err miniupnpd[3519]: could not open lease file: /var/upnp.leases
Thu Oct 12 00:03:20 2023 daemon.notice miniupnpd[3519]: HTTP listening on port 5000
Thu Oct 12 00:03:20 2023 daemon.notice miniupnpd[3519]: Listening for NAT-PMP/PCP traffic on port 5351
Thu Oct 12 00:03:27 2023 daemon.notice procd: /etc/rc.d/S96adbyby: udhcpc: started, v1.36.0
Thu Oct 12 00:03:27 2023 daemon.notice procd: /etc/rc.d/S96adbyby: udhcpc: broadcasting discover
Thu Oct 12 00:03:30 2023 daemon.notice procd: /etc/rc.d/S96adbyby: udhcpc: no lease, failing
Thu Oct 12 00:03:30 2023 daemon.info procd: - init complete -
Thu Oct 12 00:03:32 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:33 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:41 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:42 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:43 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:44 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:45 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:46 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:47 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:48 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:49 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:50 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:51 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:52 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:53 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:54 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:55 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:56 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:57 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:58 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:03:59 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:00 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:01 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:02 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:03 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:04 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:05 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:06 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:07 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:08 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:09 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:10 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:11 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:12 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:13 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:14 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:15 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:16 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:17 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:18 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:19 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:20 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:21 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:22 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:23 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:24 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:25 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:26 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:27 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:28 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:30 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:31 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:32 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:33 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:34 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:35 2023 daemon.err uhttpd[2042]: expr: syntax error
Thu Oct 12 00:04:36 2023 daemon.err uhttpd[2042]: expr: syntax error

wkq2004, 2023-10-17:
我也是同样的情况。

CYaNu, 2023-10-21:
改回原来的源码就行 打的补丁不行了
local cpu_usage = (luci.sys.exec("expr 100 - $(top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}')") or "6") .. "%"
位置在
package/lean/autocore/files/x86/index.htm

closed 2023-10-22 by rongcenter

rongcenter, 2023-10-22:
> 改回原来的源码就行 打的补丁不行了 local cpu_usage = (luci.sys.exec("expr 100 - $(top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}')") or "6") .. "%" 位置在 package/lean/autocore/files/x86/index.htm

发现的时候已经改过了，谢谢