coolsnowwolf/lede #11592: adguardhome编译报错，go: updates to go.mod needed; to update it: go mod tidy

fxxccp, 2023-10-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [*] 没有类似的issue

## 2. 详细叙述

###详细日志

 make[3] -C package/adguardhome compile
    ERROR: package/adguardhome failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/workdir/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/workdir/openwrt/scripts/config'
 make[1] world
 make[2] target/compile
 make[3] -C target/linux compile
 make[2] buildinfo
 make[3] diffconfig buildversion feedsversion
 make[4] scripts/config/conf
 make[2] package/cleanup
 make[2] package/compile
 make[3] -C package/libs/libjson-c host-compile
 make[3] -C package/libs/libubox host-compile
 make[3] -C package/system/opkg host-compile
 make[3] -C feeds/packages/lang/golang/golang host-compile
 make[3] -C feeds/packages/libs/libffi host-compile
 make[3] -C package/utils/bzip2 host-compile
 make[3] -C feeds/packages/lang/python/python3 host-compile
 make[3] -C feeds/packages/lang/node host-compile
 make[3] -C feeds/packages/lang/node-yarn host-compile
 make[3] -C package/libs/toolchain compile
 make[3] -C package/system/ca-certificates compile
 make[3] -C package/adguardhome compile
    ERROR: package/adguardhome failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/workdir/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/workdir/openwrt/scripts/config'
make[1]: Entering directory '/workdir/openwrt'
make[2]: Entering directory '/workdir/openwrt'
make[3]: Entering directory '/workdir/openwrt/target/linux'
make[4]: Entering directory '/workdir/openwrt/target/linux/x86'
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/localversion
/workdir/openwrt/scripts/kconfig.pl  + + /workdir/openwrt/target/linux/generic/config-5.15 /workdir/openwrt/target/linux/x86/config-5.15 /workdir/openwrt/target/linux/x86/64/config-5.15 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /workdir/openwrt/.config >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.target
/workdir/openwrt/scripts/package-metadata.pl kconfig /workdir/openwrt/tmp/.packageinfo /workdir/openwrt/.config 5.15 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.override
/workdir/openwrt/scripts/kconfig.pl 'm+' '+' /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.target /dev/null /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.override > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.set
mv /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.old
grep -v INITRAMFS /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.old > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.set
rm -rf /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/modules
cmp -s /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.prev || { cp /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config; cp /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.prev; }
export MAKEFLAGS= ; [ -d /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/user_headers ] || make -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Oct 11 04:16:25 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.133 CC="x86_64-openwrt-linux-musl-gcc"  INSTALL_HDR_PATH=/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/user_headers headers_install
grep '=[ym]' /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.config.set | LC_ALL=C sort | /workdir/openwrt/staging_dir/host/bin/mkhash md5 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.vermagic
touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/.configured
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/vmlinux /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133/System.map
make -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Oct 11 04:16:25 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.133 CC="x86_64-openwrt-linux-musl-gcc" olddefconfig
make[5]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.133'
net/sched/Kconfig:45: warning: menuconfig statement without prompt
#
[... 134 lines cut ...]
( pushd /workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.39 ; NODE_OPTIONS=--openssl-legacy-provider make js-deps js-build ; popd ; 	CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/workdir/openwrt/tmp/go-build" GO_MOD_CACHE_DIR="/workdir/openwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/AdguardTeam/AdGuardHome" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/AdguardTeam/AdGuardHome" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.39" GO_BUILD_DIR="/workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.39/.go_work/build" GO_BUILD_BIN_DIR="/workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.39/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/workdir/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.39=adguardhome-0.107.39 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/i [... cut ...]
make[3]: Leaving directory '/workdir/openwrt/package/adguardhome'
time: package/adguardhome/compile#0.38#0.18#0.53
    ERROR: package/adguardhome failed to build.
make[2]: *** [package/Makefile:116: package/adguardhome/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

closed 2023-10-14 by fxxccp

fairok, 2024-01-29:
楼主是怎么解决的啊？

ppyohhh, 2024-02-03:
> 楼主是怎么解决的啊？ [...]

> 楼主是怎么解决的啊？ [...]

我也是提示这个go.mod  问题  等了好几个版本的都提示 所幸我给adg删掉了....

fairok, 2024-02-04:
> 我也是提示这个go.mod 问题 等了好几个版本的都提示 所幸我给adg删掉了....

最近几个月的删掉adg可以编译就没问题，只是感觉更麻烦了