coolsnowwolf/lede #11600: make download是arm-trusted-firmware-rockchip-vendor下载失败

0YUI0, 2023-10-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
在make download的时候出现arm-trusted-firmware-rockchip-vendor下载失败
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
arm-trusted-firmware-rockchip-vendor-2023-07-26-b4558da0.tar.xz: Download from https://github.com/rockchip-linux/rkbin.git failed
### (1) 具体问题

A：dl文件夹中有atf的源码包，但是在链接中是没有的，不知道是否会影响编译

### (2) 路由器型号和固件版本

A：友善r2s

### (3) 详细日志

A：arm-trusted-firmware-rockchip-vendor-2023-07-26-b4558da0.tar.xz: Download from https://github.com/rockchip-linux/rkbin.git failed

aiamadeus, 2023-10-15:
看了下好像没有 force push，应该是你网络问题

0YUI0, 2023-10-15:
但是我是全局魔法，打开那个链接也是没有问题的，就下载回出错，用的clash，下载其他文件都没问题，就这个会失败

yanxin152133, 2023-10-16:
所以影响你编译了吗？

yanxin152133, 2023-10-16:
发日志发一行是为国家省流量嘛

0YUI0, 2023-10-16:
build失败了，就是这个的问题😭

closed 2023-10-19 by coolsnowwolf