coolsnowwolf/lede #11602: 拉取最新源码提示缺少依赖pcre2/hos

bi4tms, 2023-10-16:
拉取最新源码编译，提示缺少pcre2/hos，具体日志如下：
Create index file './feeds/packages.index' 
Create index file './feeds/luci.index' 
Create index file './feeds/routing.index' 
Create index file './feeds/telephony.index' 
Create index file './feeds/passwall_packages.index' 
Create index file './feeds/passwall.index' 
WARNING: Makefile 'package/feeds/packages/glib2/Makefile' has a build dependency on 'pcre2/host', which does not exist
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed passwall_packages.
Installing all packages from feed passwall.
请问是否需要处理，还是忽略即可

hcym, 2023-10-17:
我的编译没影响，好像是官方master带来的

bikioxo, 2023-10-17:
同提示这个，不知道有没有影响

closed 2023-10-17 by bi4tms