coolsnowwolf/lede #11604: 求助,云编译空间不够的问题

younger911, 2023-10-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
求助,云编译空间不够的问题:使用P3TERX-Actions-OpenWrt大神的云编译,No space left on device 
地址https://github.com/younger911/P3TERX-Actions-OpenWrt  
config如下:
https://github.com/younger911/P3TERX-Actions-OpenWrt/blob/main/.config
请问各位大神,该如何解决

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue,搜索了相关issue,不能解决问题

## 2. 详细叙述

### (1) 具体问题

A：
问题如下:[build](https://github.com/younger911/P3TERX-Actions-OpenWrt/actions/runs/6547266241/job/17779403228)
System.IO.IOException: No space left on device : '/home/runner/runners/2.310.2/_diag/Worker_20231017-123717-utc.log'
   at System.IO.RandomAccess.WriteAtOffset(SafeFileHandle handle, ReadOnlySpan`1 buffer, Int64 fileOffset)
   at System.IO.Strategies.BufferedFileStreamStrategy.FlushWrite()
   at System.IO.StreamWriter.Flush(Boolean flushStream, Boolean flushEncoder)
   at System.Diagnostics.TextWriterTraceListener.Flush()
   at GitHub.Runner.Common.HostTraceListener.WriteHeader(String source, TraceEventType eventType, Int32 id)
   at GitHub.Runner.Common.HostTraceListener.TraceEvent(TraceEventCache eventCache, String source, TraceEventType eventType, Int32 id, String message)
   at System.Diagnostics.TraceSource.TraceEvent(TraceEventType eventType, Int32 id, String message)
   at GitHub.Runner.Worker.Worker.RunAsync(String pipeIn, String pipeOut)
   at GitHub.Runner.Worker.Program.MainAsync(IHostContext context, String[] args)
System.IO.IOException: No space left on device : '/home/runner/runners/2.310.2/_diag/Worker_20231017-123717-utc.log'
   at System.IO.RandomAccess.WriteAtOffset(SafeFileHandle handle, ReadOnlySpan`1 buffer, Int64 fileOffset)
   at System.IO.Strategies.BufferedFileStreamStrategy.FlushWrite()
   at System.IO.StreamWriter.Flush(Boolean flushStream, Boolean flushEncoder)
   at System.Diagnostics.TextWriterTraceListener.Flush()
   at GitHub.Runner.Common.HostTraceListener.WriteHeader(String source, TraceEventType eventType, Int32 id)
   at GitHub.Runner.Common.HostTraceListener.TraceEvent(TraceEventCache eventCache, String source, TraceEventType eventType, Int32 id, String message)
   at System.Diagnostics.TraceSource.TraceEvent(TraceEventType eventType, Int32 id, String message)
   at GitHub.Runner.Common.Tracing.Error(Exception exception)
   at GitHub.Runner.Worker.Program.MainAsync(IHostContext context, String[] args)
Unhandled exception. System.IO.IOException: No space left on device : '/home/runner/runners/2.310.2/_diag/Worker_20231017-123717-utc.log'
   at System.IO.RandomAccess.WriteAtOffset(SafeFileHandle handle, ReadOnlySpan`1 buffer, Int64 fileOffset)
   at System.IO.Strategies.BufferedFileStreamStrategy.FlushWrite()
   at System.IO.StreamWriter.Flush(Boolean flushStream, Boolean flushEncoder)
   at System.Diagnostics.TextWriterTraceListener.Flush()
   at System.Diagnostics.TraceSource.Flush()
   at GitHub.Runner.Common.TraceManager.Dispose(Boolean disposing)
   at GitHub.Runner.Common.TraceManager.Dispose()
   at GitHub.Runner.Common.HostContext.Dispose(Boolean disposing)
   at GitHub.Runner.Common.HostContext.Dispose()
   at GitHub.Runner.Worker.Program.Main(String[] args)

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

lyin888, 2023-10-18:
- name: 释放磁盘空间
      uses: jlumbroso/free-disk-space@main
      with:
        # this might remove tools that are actually needed,
        # if set to "true" but frees about 6 GB
        tool-cache: true
        # all of these default to true, but feel free to set to
        # "false" if necessary for your workflow
        android: true
        dotnet: true
        haskell: true
        large-packages: true
        swap-storage: true

lyin888, 2023-10-18:
steps:
    - name: Checkout
      uses: actions/checkout@main

添加在这段代码下面

yanxin152133, 2023-10-18:
https://github.com/coolsnowwolf/lede/pull/7796

younger911, 2023-10-18:
我去试试，非常感谢大神的帮助

younger911, 2023-10-19:
可以了,现在能够完成编译,感谢大佬

closed 2023-10-19 by younger911

yaoyue12548, 2024-10-27:
> ```
> - name: 释放磁盘空间
>   uses: jlumbroso/free-disk-space@main
>   with:
>     # this might remove tools that are actually needed,
>     # if set to "true" but frees about 6 GB
>     tool-cache: true
>     # all [...]