coolsnowwolf/lede #11605: 求助,云编译6小时不够用的问题

binge8, 2023-10-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

大杂烩插件比较多，用了那个缓存，还是6小时不够用，除了精简插件，还有别的好办法吗

binge8, 2023-10-18:
> #7796 听说可以加速,不过我没试过,你看看吧

就是用了这个缓存之后还不够

liwenjie119, 2023-10-18:
试试添加这个
timeout-minutes: 1000
https://stackoverflow.com/questions/59073731/set-default-timeout-on-github-action-pipeline

binge8, 2023-10-19:
> 试试添加这个 timeout-minutes: 1000 https://stackoverflow.com/questions/59073731/set-default-timeout-on-github-action-pipeline

[image]
这么加的没效果，还是不能突破6小时

VIKINGYFY, 2023-10-21:
好奇什么情况会用到大杂烩

xuexijiaoben, 2023-10-22:
我也是不

Gabrielxzx, 2023-10-22:
> 我也是不够，感觉插件比flippy的少了一些，时间还是不够 有时候5个半小时就编译完了，多数情况是超时，只能多试几次， 不清楚怎么弄

是不是插件包含了网易云？node版本的网易云会耗费大量时间

minlang112, 2023-10-24:
> > 我也是不够，感觉插件比flippy的少了一些，时间还是不够 有时候5个半小时就编译完了，多数情况是超时，只能多试几次， 不清楚怎么弄
> 
> 是不是插件包含了网易云？依赖包含node的话会耗费大量时间

依赖包含node相当相当的耗时……当年用3代I5要跑好几个小时

clsddy, 2023-11-17:
> > > 我也是不够，感觉插件比flippy的少了一些，时间还是不够 有时候5个半小时就编译完了，多数情况是超时，只能多试几次， 不清楚怎么弄
> > 
> > 
> > 是不是插件包含了网易云？依赖包含node的话会耗费大量时间
> 
> > > > 我也是不够，感觉插件比flippy的少了一些，时间还是不够 有时候5个半小时就编译完了，多数情况是超时，只能多试几次， 不清楚怎么弄
> > > 
> > > 
> > > 是不是插件包含了网易云？依赖包含node的话会耗费大量时间
> > 
> > 
> > 依赖包含node相当相当的耗时……当年用3代I5要跑好几个小时
> 
> 大神，都有什么软件依赖会用到node呀，小白不懂，下面是我编译包含的一些软件，网易云用的不是node版本的，
> 
> CONFIG_PACKAGE_luci-app-adbyby-plus=y CONFIG_PACKAGE_luci-app-adguardhome=y CONFIG_PACKAGE_luci-app-advanced=y CONFIG_PACKAGE_luci-app-aliddns=y CONFIG_PACKAGE_luci-app-alist=y CONFIG_PACKAGE_luci-app-aliyundrive-webdav=y CONFIG_PACKAGE_luci-app-amlogic=y CONFIG_PACKAGE_luci-app-argone-config=y CONFIG_PACKAGE_luci-app-bypass=y CONFIG_PACKAGE_luci-app-bypass_INCLUDE_Hysteria=y CONFIG_PACKAGE_luci-app-bypass_INCLUDE_Kcptun=y CONFIG_PACKAGE_luci-app-bypass_INCLUDE_NaiveProxy=y CONFIG_PACKAGE_luci-app-bypass_INCLUDE_Socks5_Proxy=y CONFIG_PACKAGE_luci-app-bypass_INCLUDE_V2ray_plugin=y CONFIG_PACKAGE_luci-app-cloudflarespeedtest=y CONFIG_PACKAGE_luci-app-cpufreq=y CONFIG_PACKAGE_luci-app-ddnsto=y CONFIG_PACKAGE_luci-app-design-config=y CONFIG_PACKAGE_luci-app-diskman=y CONFIG_PACKAGE_luci-app-diskman_INCLUDE_btrfs_progs=y CONFIG_PACKAGE_luci-app-diskman_INCLUDE_lsblk=y CONFIG_PACKAGE_luci-app-dockerman=y CONFIG_PACKAGE_luci-app-filebrowser=y CONFIG_PACKAGE_luci-app-hd-idle=y CONFIG_PACKAGE_luci-app-istorex=y CONFIG_PACKAGE_luci-app-lucky=y CONFIG_PACKAGE_luci-app-minidlna=y CONFIG_PACKAGE_luci-app-mosdns=y CONFIG_PACKAGE_luci-app-mwan3=y CONFIG_PACKAGE_luci-app-nfs=y CONFIG_PACKAGE_luci-app-openclash=y CONFIG_PACKAGE_luci-app-passwall=y CONFIG_PACKAGE_luci-app-passwall2=y CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_Brook=y CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_Hysteria=y CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_IPv6_Nat=y CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_NaiveProxy=y CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_tuic_client=y CONFIG_PACKAGE_luci-app-passwall2_Iptables_Transparent_Proxy=y CONFIG_PACKAGE_luci-app-passwall2_Nftables_Transparent_Proxy=y CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Brook=y CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Hysteria=y CONFIG_PACKAGE_luci-app-passwall_INCLUDE_NaiveProxy=y CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Trojan_GO=y CONFIG_PACKAGE_luci-app-passwall_INCLUDE_V2ray_Geodata=y CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Xray_Plugin=y CONFIG_PACKAGE_luci-app-passwall_INCLUDE_tuic_client=y CONFIG_PACKAGE_luci-app-passwall_Iptables_Transparent_Proxy=y CONFIG_PACKAGE_luci-app-passwall_Nftables_Transparent_Proxy=y CONFIG_PACKAGE_luci-app-pushbot=y CONFIG_PACKAGE_luci-app-qbittorrent=y CONFIG_PACKAGE_luci-app-qos=y CONFIG_PACKAGE_luci-app-quickstart=y CONFIG_PACKAGE_luci-app-ramfree=y CONFIG_PACKAGE_luci-app-samba4=y CONFIG_PACKAGE_luci-app-smartdns=y CONFIG_PACKAGE_luci-app-socat=y CONFIG_PACKAGE_luci-app-sqm=y CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Hysteria=y CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_IPT2Socks=y CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Kcptun=y CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_NaiveProxy=y CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Redsocks2=y CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadow_TLS=y CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks_V2ray_Plugin=y CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Trojan=y CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Tuic_Client=y CONFIG_PACKAGE_luci-app-store=y CONFIG_PACKAGE_luci-app-syncdial=y CONFIG_PACKAGE_luci-app-ttyd=y CONFIG_PACKAGE_luci-app-udpxy=y CONFIG_PACKAGE_luci-app-unblockmusic=y CONFIG_PACKAGE_luci-app-vssr=y CONFIG_PACKAGE_luci-app-vssr_INCLUDE_Kcptun=y CONFIG_PACKAGE_luci-app-wireguard=y CONFIG_PACKAGE_luci-app-wrtbwmon=y CONFIG_PACKAGE_luci-app-zerotier=y

一个一个增加编译测试

Link-HXX, 2023-11-30:
> 以加速,不过我没试过,你

fdvfdvfdv

coolsnowwolf, 2026-09-05:
这是云编译资源/配置咨询，不是 LEDE 源码缺陷。大规模插件会显著增加首次工具链和包编译时间，缓存只能复用未变化的步骤；可拆分目标、减少不需要的包、固定 feeds 版本并使用自建 runner/本地编译。GitHub Actions 的单 job 时间上限由工作流平台决定，仓库无法通过补丁延长。无可提交代码修复，关闭该 issue。

closed 2026-09-05 by coolsnowwolf