coolsnowwolf/lede #11609: 选择了kmod-video-core就会出错

392377870, 2023-10-20:
选择了kmod-video-core就会出错

 路由器型号和固件版本

king3399 

### (3) 详细日志

removed '/home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-usb-net-rtl8152_6.1.56-1_aarch64_generic.ipk'
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-usb-net-rtl8152/lib/modules/6.1.56/r8152.ko: relocatable
Packaged contents of /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-usb-net-rtl8152 into /home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-usb-net-rtl8152_6.1.56-1_aarch64_generic.ipk
removed '/home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-backlight_6.1.56-1_aarch64_generic.ipk'
NOTICE: module '/home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.56/drivers/video/backlight/backlight.ko' is built-in.
Packaged contents of /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-backlight into /home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-backlight_6.1.56-1_aarch64_generic.ipk
removed '/home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-fb_6.1.56-1_aarch64_generic.ipk'
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb/lib/modules/6.1.56/font.ko: relocatable
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb/lib/modules/6.1.56/fb.ko: relocatable
Packaged contents of /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb into /home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-fb_6.1.56-1_aarch64_generic.ipk
removed '/home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-fb-cfb-fillrect_6.1.56-1_aarch64_generic.ipk'
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb-cfb-fillrect/lib/modules/6.1.56/cfbfillrect.ko: relocatable
Packaged contents of /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb-cfb-fillrect into /home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-fb-cfb-fillrect_6.1.56-1_aarch64_generic.ipk
removed '/home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-fb-cfb-copyarea_6.1.56-1_aarch64_generic.ipk'
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb-cfb-copyarea/lib/modules/6.1.56/cfbcopyarea.ko: relocatable
Packaged contents of /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb-cfb-copyarea into /home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-fb-cfb-copyarea_6.1.56-1_aarch64_generic.ipk
removed '/home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-fb-cfb-imgblt_6.1.56-1_aarch64_generic.ipk'
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb-cfb-imgblt/lib/modules/6.1.56/cfbimgblt.ko: relocatable
Packaged contents of /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb-cfb-imgblt into /home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-fb-cfb-imgblt_6.1.56-1_aarch64_generic.ipk
removed '/home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-fb-sys-fops_6.1.56-1_aarch64_generic.ipk'
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb-sys-fops/lib/modules/6.1.56/fb_sys_fops.ko: relocatable
Packaged contents of /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb-sys-fops into /home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-fb-sys-fops_6.1.56-1_aarch64_generic.ipk
removed '/home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-fb-sys-ram_6.1.56-1_aarch64_generic.ipk'
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb-sys-ram/lib/modules/6.1.56/sysimgblt.ko: relocatable
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb-sys-ram/lib/modules/6.1.56/sysfillrect.ko: relocatable
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb-sys-ram/lib/modules/6.1.56/syscopyarea.ko: relocatable
Packaged contents of /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-fb-sys-ram into /home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-fb-sys-ram_6.1.56-1_aarch64_generic.ipk
removed '/home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-multimedia-input_6.1.56-1_aarch64_generic.ipk'
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-multimedia-input/lib/modules/6.1.56/rc-core.ko: relocatable
Packaged contents of /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-multimedia-input into /home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-multimedia-input_6.1.56-1_aarch64_generic.ipk
removed '/home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-drm_6.1.56-1_aarch64_generic.ipk'
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-drm/lib/modules/6.1.56/drm_panel_orientation_quirks.ko: relocatable
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-drm/lib/modules/6.1.56/drm.ko: relocatable
Packaged contents of /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-drm into /home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-drm_6.1.56-1_aarch64_generic.ipk
removed '/home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-drm-kms-helper_6.1.56-1_aarch64_generic.ipk'
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-drm-kms-helper/lib/modules/6.1.56/drm_kms_helper.ko: relocatable
Packaged contents of /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-drm-kms-helper into /home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-drm-kms-helper_6.1.56-1_aarch64_generic.ipk
removed '/home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-drm-display-helper_6.1.56-1_aarch64_generic.ipk'
rstrip.sh: /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-drm-display-helper/lib/modules/6.1.56/drm_display_helper.ko: relocatable
Packaged contents of /home/openwrt-19810729/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-drm-display-helper into /home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-drm-display-helper_6.1.56-1_aarch64_generic.ipk
Package kmod-video-core is missing dependencies for the following libraries:
mc.ko
make[3]: *** [modules/video.mk:514: /home/openwrt-19810729/lede/bin/targets/rockchip/armv8/packages/kmod-video-core_6.1.56-1_aarch64_generic.ipk] Error 1
make[3]: Leaving directory '/home/openwrt-19810729/lede/package/kernel/linux'
time: package/kernel/linux/compile#18.73#11.32#30.89
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/openwrt-19810729/lede'
make[1]: *** [package/Makefile:110: /home/openwrt-19810729/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt-19810729/lede'
make: *** [/home/openwrt-19810729/lede/include/toplevel.mk:231：world] 错误 2

coolsnowwolf, 2023-10-21:
fixed

closed 2023-10-21 by coolsnowwolf

xbf321, 2023-12-14:
现在还报错，看来没有修复

xbf321, 2023-12-14:
NOTICE: module '/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.62/drivers/usb/host/ehci-hcd.ko' is built-in.
Packaged contents of /openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-ehci into /openwrt/lede/bin/targets/x86/64/packages/kmod-usb-ehci_6.1.62-1_x86_64.ipk
removed '/openwrt/lede/bin/targets/x86/64/packages/kmod-usb2_6.1.62-1_x86_64.ipk'
rstrip.sh: /openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb2/lib/modules/6.1.62/ehci-platform.ko: relocatable
rstrip.sh: /openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb2/lib/modules/6.1.62/ehci-fsl.ko: relocatable
rstrip.sh: /openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb2/lib/modules/6.1.62/fsl-mph-dr-of.ko: relocatable
Packaged contents of /openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb2 into /openwrt/lede/bin/targets/x86/64/packages/kmod-usb2_6.1.62-1_x86_64.ipk
removed '/openwrt/lede/bin/targets/x86/64/packages/kmod-usb-audio_6.1.62-1_x86_64.ipk'
Package kmod-usb-audio is missing dependencies for the following libraries:
mc.ko
make[3]: *** [modules/usb.mk:562: /openwrt/lede/bin/targets/x86/64/packages/kmod-usb-audio_6.1.62-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/openwrt/lede/package/kernel/linux'
time: package/kernel/linux/compile#29.56#11.15#47.42
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/openwrt/lede'
make[1]: *** [package/Makefile:110: /openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/openwrt/lede'
make: *** [/openwrt/lede/include/toplevel.mk:231: world] Error 2